anestisb/android-prepare-vendor #3: System partition de-optimization failed

rom1v, 2016-04-18:
Hi,

Thanks for your work.

However, execution fails:
~~~
# ./execute-all.sh -d bullhead -b MHC19Q -i bullhead-mhc19q-factory-22ba5e5a.tgz -o out
[*] Setting output base to 'out/bullhead/mhc19q'
[*] Extracting 'bullhead-mhc19q-factory-22ba5e5a.tgz'
[*] Unzipping 'image-bullhead-mhc19q.zip'
[*] Copying files from system parition ...
[*] Copying files from vendor partition ...
[*] Preparing environment for 'arm' ABI
[!] Boot classes extraction failed
[-] System partition de-optimization failed
~~~

anestisb, 2016-04-18:
The usual suspect for the repair script is that your system doesn't have/expose correctly Java 8 bins in $PATH env. You can verify this by running the oat2dex jar with the following command from repo root dir. If you get an error message you need to apply Java 8 system-wide or modify the scripts to match your env.

```
anestisb@nemesis:[android-prepare-vendor]: java -jar hostTools/Java/oat2dex.jar
Easy oat2dex 0.86
Usage:
 java -jar oat2dex.jar [options] <action>
[options]
 Api level (for raw odex): -a <integer>
 Output folder: -o <folder path>
 Print detail : -v
<action>
 Get dex from boot.oat: boot <boot.oat>
 Get dex from oat/odex: <oat/odex file> <boot-class-folder>
 Get raw odex from oat: odex <oat file>
 Get raw odex smali   : smali <oat/odex file>
 Deodex framework     : devfw [empty or path of /system/framework/]
anestisb@nemesis:[android-prepare-vendor]: java -version
java version "1.8.0_77"
Java(TM) SE Runtime Environment (build 1.8.0_77-b03)
Java HotSpot(TM) 64-Bit Server VM (build 25.77-b03, mixed mode)
```

If the repair script fails for other reasons, consider running the script in debug mode by editing (uncomment the ```set -x``` line at the top area) the ```scripts/system-img-repair.sh``` file. If you still can't figure out the problem paste the log here in case I can help any further.

rom1v, 2016-04-18:
Thank you, the default on my system was Java 7.

:+1:

closed 2016-04-18 by rom1v