anestisb/android-prepare-vendor #4: GoogleMediaServices included?

rom1v, 2016-04-18:
Hi,

This is not an issue, but rather a question.

The execution printed the following lines:

~~~.
…
[*] '/priv-app/PrebuiltGmsCore/PrebuiltGmsCore.apk' is multi-dex - adjusting recursive archive adds
[*] '/priv-app/GoogleServicesFramework/GoogleServicesFramework.apk' not pre-optimized with sanity checks passed - copying without changes
[*] '/app/Music2/Music2.apk' is multi-dex - adjusting recursive archive adds
[*] '/app/Photos/Photos.apk' is multi-dex - adjusting recursive archive adds
[-] '/app/PrebuiltExchange3Google/PrebuiltExchange3Google.apk' de-optimization failed consider manual inspection - skipping archive
[*] '/app/Maps/Maps.apk' is multi-dex - adjusting recursive archive adds
…
~~~

Does this mean that Google proprietary stuff will be included?

These apks are copied to `out/bullhead/mhc19q/factory_imgs_repaired_data/system/priv-app`, but not to `out/bullhead/mhc19q/vendor/lge/bullhead`. so I think the answer is no, but I prefer asking. I don't want google stuff in my build.

A more general question: is the proprietary stuff extracted for the AOSP build the minimal set to make the device work correctly?

thestinger, 2016-04-18:
It doesn't include Play Services. You could mean a lot of different things by 'minimal set' though. Ideally, Google would stop putting proprietary blobs other than Play outside the vendor image and this would just regenerate a *complete* vendor image. That's pretty close to what it does on the Nexus 9, but they screwed up the 5X and 6P.

rom1v, 2016-04-19:
OK, thank you :+1:

closed 2016-04-19 by rom1v