anestisb/android-prepare-vendor #5: Results on android-6.0.1_r24 bullhead

rom1v, 2016-04-18:
Hi again,

I generated `vendor` using your tool, and built `android-6.0.1_r24` for Nexus 5X.

The build was OK.

On boot, it prints *There's an internal problem with your device. Contact your manufacturer for details.*
This is a known problem that is easily [resolved](http://forum.xda-developers.com/nexus-9/development/fix-build-prop-variety-fix-aka-contact-t3133347) by using the same fingerprint in `/system/build.prop` and `/vendor/build.prop` but maybe it could be automated.

I have another problem, an alert *Exchange Services stopped*.

~~~
04-18 22:53:42.440  5051  5065 I ActivityManager: Start proc 8078:com.android.exchange/u0a32 for service com.android.exchange/.service.EasService
04-18 22:53:42.476  8078  8078 W System  : ClassLoader referenced unknown path: /system/app/Exchange2/lib/arm64
04-18 22:53:42.482  8078  8078 D AndroidRuntime: Shutting down VM
04-18 22:53:42.483  8078  8078 E AndroidRuntime: FATAL EXCEPTION: main
04-18 22:53:42.483  8078  8078 E AndroidRuntime: Process: com.android.exchange, PID: 8078
04-18 22:53:42.483  8078  8078 E AndroidRuntime: java.lang.RuntimeException: Unable to instantiate application com.android.exchange.Exchange: java.lang.ClassNotFoundException: Didn't find class "com.android.exchange.Exchange" on path: DexPathList[[zip file "/system/app/Exchange2/Exchange2.apk"],nativeLibraryDirectories=[/system/app/Exchange2/lib/arm64, /vendor/lib64, /system/lib64]]
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at android.app.LoadedApk.makeApplication(LoadedApk.java:578)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:4685)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at android.app.ActivityThread.-wrap1(ActivityThread.java)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1405)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:102)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:148)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:5422)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: Caused by: java.lang.ClassNotFoundException: Didn't find class "com.android.exchange.Exchange" on path: DexPathList[[zip file "/system/app/Exchange2/Exchange2.apk"],nativeLibraryDirectories=[/system/app/Exchange2/lib/arm64, /vendor/lib64, /system/lib64]]
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at java.lang.ClassLoader.loadClass(ClassLoader.java:511)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at java.lang.ClassLoader.loadClass(ClassLoader.java:469)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at android.app.Instrumentation.newApplication(Instrumentation.java:981)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	at android.app.LoadedApk.makeApplication(LoadedApk.java:573)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	... 9 more
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	Suppressed: java.io.IOException: No original dex files found for dex location /system/app/Exchange2/Exchange2.apk
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at dalvik.system.DexFile.openDexFileNative(Native Method)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at dalvik.system.DexFile.openDexFile(DexFile.java:295)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at dalvik.system.DexFile.<init>(DexFile.java:80)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at dalvik.system.DexFile.<init>(DexFile.java:59)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at dalvik.system.DexPathList.loadDexFile(DexPathList.java:279)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at dalvik.system.DexPathList.makePathElements(DexPathList.java:248)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at dalvik.system.DexPathList.<init>(DexPathList.java:120)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at dalvik.system.BaseDexClassLoader.<init>(BaseDexClassLoader.java:48)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at dalvik.system.PathClassLoader.<init>(PathClassLoader.java:65)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at android.app.ApplicationLoaders.getClassLoader(ApplicationLoaders.java:58)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at android.app.LoadedApk.getClassLoader(LoadedApk.java:376)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at android.app.LoadedApk.makeApplication(LoadedApk.java:568)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		... 9 more
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	Suppressed: java.lang.ClassNotFoundException: com.android.exchange.Exchange
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at java.lang.Class.classForName(Native Method)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at java.lang.BootClassLoader.findClass(ClassLoader.java:781)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at java.lang.BootClassLoader.loadClass(ClassLoader.java:841)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 		... 12 more
04-18 22:53:42.483  8078  8078 E AndroidRuntime: 	Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack trace available
~~~

Do you know if it's related to this tool?

Before using it, I had SIM problems (not tested yet with this build) but no "Exchange Services" problem.

Note that I removed the `priv-app` from `out/bullhead/mhc19q/vendor/lge/bullhead/proprietary/priv-app`, but I don't think it's related.

thestinger, 2016-04-18:
Exchange was removed in the marshmallow-dr1.5-release branch used by the android-6.0.1_r24 builds for the Nexus 5X and 6P. It's dead. The fingerprint *should* match automatically, but you need to do a clean build. Incremental builds of the Android Open Source Project are inherently broken and only useful for quick testing.

rom1v, 2016-04-19:
OK, I try a clean build.

rom1v, 2016-04-19:
A clean build solved all my problems :+1:

closed 2016-04-19 by rom1v