anestisb/android-prepare-vendor #8: Boot classes extraction failed

icyerasor, 2016-06-02:
When JDK is set to 1.7 boot class extraction will fail:
```
[*] Preparing environment for 'arm' ABI
[!] Boot classes extraction failed
[-] System partition de-optimization failed
...

icyerasor@aosp:/tmp/android_img_repair.IBHeWV/arm$ java -jar /home/icyerasor/android-prepare-vendor/hostTools/Java/oat2dex.jar boot /tmp/android_img_repair.IBHeWV/arm/boot.oat
Exception in thread "main" java.lang.UnsupportedClassVersionError: org/rh/smaliex/Main : Unsupported major.minor version 52.0
        at java.lang.ClassLoader.defineClass1(Native Method)
        at java.lang.ClassLoader.defineClass(ClassLoader.java:803)
        at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
        at java.net.URLClassLoader.defineClass(URLClassLoader.java:449)
        at java.net.URLClassLoader.access$100(URLClassLoader.java:71)
        at java.net.URLClassLoader$1.run(URLClassLoader.java:361)
        at java.net.URLClassLoader$1.run(URLClassLoader.java:355)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.net.URLClassLoader.findClass(URLClassLoader.java:354)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:425)
        at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:358)
        at sun.launcher.LauncherHelper.checkAndLoadMain(LauncherHelper.java:482)
```

Solution: (install JDK 1.8 &) update-alternatives for java and javac to use 1.8
Depending on what branch you try to compile, you might have to switch back to 1.7 again before building aosp.

Not sure on how to improve this, maybe mention it int he requirements or as one of the causes of "boot classes extraction failed"?

anestisb, 2016-06-02:
Thanks for raising the issue. Commit https://github.com/anestisb/android-prepare-vendor/commit/645a204aa2ded71bcaad7ec14788b6b7e69bdb0c implements an extra check to verify Java 8 is found in running system $PATH

closed 2016-06-02 by anestisb

icyerasor, 2016-06-06:
Awesome, thx!