anestisb/android-prepare-vendor #11: vendor.img in AOSP build not generated for bullhead MTC19V

tryge, 2016-06-29:
So, if I reproduce my actions correctly I've done about the following:

```
cd $HOME/android
make clobber
cd $HOME/android-prepare-vendor
sudo ./extract-all.sh -d bullhead -b MTC19V -o $HOME/android
cd $HOME/android
. build/envsetup.sh
lunch 18
make -j2
make dist
./build/tools/releasetools/sign_target_files_apks -o -d ~/.android-certs out/dist/*-target_files-*.zip signed-target_files.zip
./build/tools/releasetools/img_from_target_files signed-target-files.zip signed-img.zip
```

I see that the generated makefiles are included:
```
including ./bullhead/mtc19v/vendor/lge/bullhead/Android.mk ...
```

However, no `vendor.img` is included in the final `signed-img.zipz and I can't locate any intermediaries. When flashing the generated images the system won't detect the SIM card.

There might be a user error on my side, or maybe some changes in the latest version. In case of the latter, I'll happily make all the necessary changes myself in order to get it working, however I have no clue where to begin looking.

I'll appreciate any hints, thank you.

anestisb, 2016-06-30:
I just fired a build for the build tag you mentioned and everything is working like a charm. 

From the commands you listed it seems that you that have a small error in the vendor dir structure and thus the makefiles are not picked up by the build process. The vendor prepare script output has two additional directory levels (at the head) added for archiving reasons. If you inspect the script output you will see a structure similar to the following:
```
# tree -L 4 bullhead/mtc19v
bullhead/mtc19v
├── bullhead-mtc19v-factory-f3a6bee5.tgz
└── vendor
    └── lge
        └── bullhead
            ├── Android.mk
            ├── BoardConfigVendor.mk
            ├── bullhead-vendor-blobs.mk
            ├── device-vendor.mk
            ├── proprietary
            └── vendor
```

Now since you define as output path the AOSP_ROOT directory, the two extra layers in the dir structure are not aligned with the AOSP build system. You need to place the files starting from ```vendor/``` at the AOSP_ROOT dir.

tryge, 2016-06-30:
Thank you for your feedback, I'll try that immediately.

Regarding having to relocate the output I had two issues (I'm not a native speaker, so this might be why...):
(1) I didn't understand the last line of the output as prompting me to actually move it.
(2) in the `generate-vendor.sh` I found this:

```
INFO:
      * Output should be moved/synced with AOSP root, unless -o is AOSP root
```

therefore believed, that it would suffice that I pass the AOSP root as  output directory to the top level script (I didn't track the actual invocations though).

I'll think about what documentation or output would have prevented the issues for me and send a PR (soonish).

closed 2016-06-30 by tryge

anestisb, 2016-06-30:
You're absolutely right and thanks for noticing. The info bullet was left behind from an earlier version of the script that I was following a different dir structure for the output.

Looking forward to your PR. Thanks.

tryge, 2016-07-01:
I can confirm that it works, thank you.