anestisb/android-prepare-vendor #12: the symlink target for libdmengine ends up with an extra /lib/ in the path

thestinger, 2016-07-01:
I'm not sure if this problem will be reproducible for you, but it happens consistently here. I can look into it deeper but for now I think I'll just edit the generated output to fix it.

I end up with libdmengine being incorrectly referenced like this:

```
$ ls /system/priv-app/DMService/lib/arm/ -l                     
lrw-r--r-- root     root              2009-01-01 00:00 libdmengine.so -> /system/lib/lib/libdmengine.so
lrw-r--r-- root     root              2009-01-01 00:00 libdmjavaplugin.so -> /system/lib/libdmjavaplugin.so
```

It can be seen in the generated Android.mk that the `libdmjavaplugin.so` library is fine, but `libdmengine.so` is not:

```
include $(CLEAR_VARS)
LOCAL_MODULE := libdmjavaplugin_32.so
LOCAL_MODULE_CLASS := FAKE
LOCAL_MODULE_TAGS := optional
LOCAL_MODULE_OWNER := huawei
include $(BUILD_SYSTEM)/base_rules.mk
$(LOCAL_BUILT_MODULE): TARGET := /system/lib/libdmjavaplugin.so
$(LOCAL_BUILT_MODULE): SYMLINK := $(PRODUCT_OUT)/$(TARGET_COPY_OUT_SYSTEM)/priv-app/DMService/lib/arm/libdmjavaplugin.so
$(LOCAL_BUILT_MODULE): $(LOCAL_PATH)/Android.mk
$(LOCAL_BUILT_MODULE):
        $(hide) mkdir -p $(dir $@)
        $(hide) mkdir -p $(dir $(SYMLINK))
        $(hide) rm -rf $@
        $(hide) rm -rf $(SYMLINK)
        $(hide) ln -sf $(TARGET) $(SYMLINK)
        $(hide) touch $@

include $(CLEAR_VARS)
LOCAL_MODULE := libdmengine_32.so
LOCAL_MODULE_CLASS := FAKE
LOCAL_MODULE_TAGS := optional
LOCAL_MODULE_OWNER := huawei
include $(BUILD_SYSTEM)/base_rules.mk
$(LOCAL_BUILT_MODULE): TARGET := /system/lib/lib/libdmengine.so
$(LOCAL_BUILT_MODULE): SYMLINK := $(PRODUCT_OUT)/$(TARGET_COPY_OUT_SYSTEM)/priv-app/DMService/lib/arm/libdmengine.so
$(LOCAL_BUILT_MODULE): $(LOCAL_PATH)/Android.mk
$(LOCAL_BUILT_MODULE):
        $(hide) mkdir -p $(dir $@)
        $(hide) mkdir -p $(dir $(SYMLINK))                                                                                                  
        $(hide) rm -rf $@
        $(hide) rm -rf $(SYMLINK)
        $(hide) ln -sf $(TARGET) $(SYMLINK)
        $(hide) touch $@
```

thestinger, 2016-07-01:
Happens for both angler and bullhead.

anestisb, 2016-07-02:
There was an error in the path construct logic. The original base dir for DSOs was appended inside the loop and thus creating the duplicate subdir levels when more than one symlinks were created under the same root.

It should be fixed now. Please let me know if the fix didn't work for you.

posativ, 2016-07-12:
Build CopperheadOS on sunday (2016-07-10) using Gentoo:

```
~> adb shell ls /system/priv-app/DMService/lib/arm/ -l
lrw-r--r-- root     root              2009-01-01 00:00 libdmengine.so -> /system/lib/libdmengine.so
lrw-r--r-- root     root              2009-01-01 00:00 libdmjavaplugin.so -> /system/lib/libdmjavaplugin.so
~> adb shell getprop ro.vendor.build.date
Sun Jul 10 23:04:05 CEST 2016
```

If that helps.

anestisb, 2016-07-12:
Awesome, thanks for confirming @posativ 

Feel free to re-open if you spot any glitches.

closed 2016-07-12 by anestisb