anestisb/android-prepare-vendor #13: Crash on insufficient memory

Huppys, 2016-08-30:
Hey,

just a little note: I used your repo within Ubuntu 16.04 x64 and experienced a crash because of a too small configured RAM. 

Maybe you could advise to have at least 3-4 GB of RAM.

Best,

Huppys

anestisb, 2016-08-31:
That is weird. Which component/step triggered the out of memory crash?

Huppys, 2016-09-03:
Some different parts after "Start extracting system partition & de-optimize pre-compiled bytecode ..." like extracting the apps like Photos.

anestisb, 2016-09-04:
So all crashes occurred when de-optimizing with oat2dex.jar tool

```
cat hs_err_pid* | grep java_command
java_command: /home/huppys/projects/aosp/android-prepare-vendor/hostTools/Java/oat2dex.jar -o /tmp/android_img_repair.jg1PfC /home/huppys/projects/aosp/android-prepare-vendor/out/bullhead/mhc19q/factory_imgs_data/system/priv-app/PrebuiltGmsCore/oat/arm/PrebuiltGmsCore.odex /tmp/android_img_repair.jg1PfC/arm/dex
java_command: /home/huppys/projects/aosp/android-prepare-vendor/hostTools/Java/oat2dex.jar boot /tmp/android_img_repair.7YsfBL/arm64/boot.oat
java_command: /home/huppys/projects/aosp/android-prepare-vendor/hostTools/Java/oat2dex.jar boot /tmp/android_img_repair.9RFXnZ/arm64/boot.oat
java_command: /home/huppys/projects/aosp/android-prepare-vendor/hostTools/Java/oat2dex.jar boot /tmp/android_img_repair.tz402a/arm/boot.oat
java_command: /home/huppys/projects/aosp/android-prepare-vendor/hostTools/Java/oat2dex.jar -o /tmp/android_img_repair.0ez3H1 ./out/bullhead/mtc20f/factory_imgs_data/system/app/Wallet/oat/arm64/Wallet.odex /tmp/android_img_repair.0ez3H1/arm64/dex
java_command: /home/huppys/projects/aosp/android-prepare-vendor/hostTools/Java/oat2dex.jar -o /tmp/android_img_repair.RTYiDc ./out/bullhead/mtc20f/factory_imgs_data/system/app/Tycho/oat/arm64/Tycho.odex /tmp/android_img_repair.RTYiDc/arm64/dex
java_command: /home/huppys/projects/aosp/android-prepare-vendor/hostTools/Java/oat2dex.jar -o /tmp/android_img_repair.jxti9z ./out/bullhead/mtc20f/factory_imgs_data/system/app/PrebuiltGmail/oat/arm64/PrebuiltGmail.odex /tmp/android_img_repair.jxti9z/arm64/dex
```

I'm guessing that since you mentioned a VM you have assigned 1-2 cores to it. So the issue is probably a combination of low memory and oat2dex.jar tool defaulting to 4 parallel jobs (https://github.com/testwhat/SmaliEx/blob/master/smaliex/src/main/java/org/rh/smaliex/OatUtil.java#L93).

I'll probably submit a patch to SmaliEx project and fix at the external tool level since the developer is not exposing a user option flag to control it from calling args. Will keep this issue updated when progress is made.

Thanks for reporting.

anestisb, 2016-09-07:
Even with jobs set to 1 CPU 2GBs of RAM are not enough. Added message warning to let users know. Will be soon merged to master as well after finishing some other testing of recent changes.

Thanks again.

closed 2016-09-07 by anestisb