anestisb/android-prepare-vendor #14: Archive extension

mikalackis, 2016-09-01:
I just used this tool for my Nexus 5X AOSP project and I noticed that the file downloaded from google's servers (factory image) ends up with .zip extension, so maybe that should be covered in the script?

anestisb, 2016-09-01:
Yeah Google switched from tgz to zip just yesterday. I've already started working around all recent changes at the master-dev branch (https://github.com/anestisb/android-prepare-vendor/tree/master-dev).

Go for it until I complete all testing and merge with master. If you notice any bugs reporting them is more than appreciated. Also just to let you know that Android-N support is not ready yet.

anestisb, 2016-09-01:
Just tested latest ZIP format against Nexus 9 WiFi with Android Nougat images and everything seems ok. Have merged with master, so update your local repo to use factory images under zip format.

So far for Android N only Nexus 9 WiFi has been tested. In your Nexus 5x setup you can use only against older versions for now.

closed 2016-09-01 by anestisb