anestisb/android-prepare-vendor #15: Android-N support for Nexus 5x

anestisb, 2016-09-01:


anestisb, 2016-09-04:
Seems we have some shifted locations for some blobs. We need to branch api levels for "system-proprietary-blobs-api.txt" to deal with such cases.

```
root@bchfs:bullhead# cat system-proprietary-blobs-api.txt | while read -r file; do if [[ ! -f "nrd90m/factory_imgs_data/$file" ]]; then echo "'$file' missing"; fi; done
'system/app/atfwd/atfwd.apk' missing
'system/app/qcrilmsgtunnel/qcrilmsgtunnel.apk' missing
'system/bin/imscmservice' missing
'system/etc/diag/IMS_DS_CM_NAS.cfg' missing
'system/etc/diag/radio_ims.cfg' missing
```

We need to also investigate if any more new blobs need to added in the list.

anestisb, 2016-09-04:
Commit https://github.com/anestisb/android-prepare-vendor/commit/0c6de817a7269361d860e2c32644ccab51977c61 added the base to start working with API specific blob lists without breaking backwards compatibility.

anestisb, 2016-09-07:
Dependency of AOSP targets to vendor proprietary modules support has been added as explained in https://github.com/anestisb/android-prepare-vendor/issues/21

Final step is deal with bytecode repairs. If alternative repair method (https://github.com/anestisb/android-prepare-vendor/issues/22) is working (currently testing) we're done. Otherwise we need to wait until smali adds support new OAT format (https://github.com/anestisb/android-prepare-vendor/issues/18).

anestisb, 2016-09-07:
```
11-09 21:27:26.706   720   720 F libc    : CANNOT LINK EXECUTABLE "/vendor/bin/slim_daemon": library "libloc_api_v02.so" not found
11-09 21:27:26.706   720   720 F libc    : Fatal signal 6 (SIGABRT), code -6 in tid 720 (slim_daemon)
```

That is very strange since the target is defined and included [here](https://android.googlesource.com/device/lge/bullhead/+/android-7.0.0_r4/location/loc_api/loc_api_v02/Android.mk), although is not part of the final AOSP output /system. Haven't fully trace it to identify exactly why, although it's fixed by reverting (https://github.com/anestisb/android-prepare-vendor/commit/a05fe9f3dc4c4aeebb33874481fe42d60718352e, https://github.com/anestisb/android-prepare-vendor/commit/5623228b5ec304088fbda4d0347daddddc63efb6) the `system/lib64/libloc_api_v02.so` & `system/lib/libloc_api_v02.so` proprietary blobs (not as individual shared modules though)

anestisb, 2016-09-07:
Initial testing of a vanilla AOSP build with new DEX repair method (https://github.com/anestisb/android-prepare-vendor/issues/22) has been completed with no issues so far (hurrah).

Manually tested components with no errors/issues:
* Radio working as expected (tested with Greek carriers)
* 4G (tested with Greek carriers)
* WiFi
* Bluetooth (contacts access & file transfer)
* Fingerprint
* Location

anestisb, 2016-09-07:
Also tested with no issues:
* NFC (mostly via Yubikey neo)
* Hotspot tethering
* Camera (Front & Back)
* Multi-user (install app from other user)

mikalackis, 2016-09-08:
Fingers crossed man, waiting for your feedback (and commit ;)) as i got stuck on Nougat nexus5x (de-opt failed).

anestisb, 2016-09-08:
The bytecode repair (de-optimization) issue seems to be resolved with the new repair method via oatdump (see more info [here](https://github.com/anestisb/android-prepare-vendor/issues/22)).

If you pull the master-dev branch you can successfully generate a valid vendor makefile & blobs structure to be included in AOSP.

The only small bits that I'm still working are some dependency resolves for the bullhead gps modules. I thought I had it fixed although something hasn't been cleaned properly in my build env and I regressed it. Hopefully I'll sort it out within the next few hours and merge with master.

If you have time to test the master-dev branch against the Nougat factory images, let me know if faced any errors / issues.

mikalackis, 2016-09-08:
Ok cool, i'm checking out the master-dev branch and will try to build it. I'll update you about the process, thnx

rvimont, 2016-09-08:
From commit 5c963b5863e0cff21d8c56605f8083048ae2cce1:

```
# ./execute-all.sh -d bullhead -b NRD90M -i ~/phone/bullhead-nrd90m-factory-61495c8b.zip -k -o out
[*] Setting output base to 'out/bullhead/nrd90m'
[*] Processing with 'API-24' configuration
[*] Extracting '/home/rom/phone/bullhead-nrd90m-factory-61495c8b.zip'
[*] Unzipping 'image-bullhead-nrd90m.zip'
[*] Copying files from 'system.img.raw' image ...
[*] Copying files from 'vendor.img.raw' image ...
[*] Preparing environment for 'arm' ABI
[*] Preparing environment for 'arm64' ABI
[*] Start processing system partition & de-optimize pre-compiled bytecode ...
[*] '15' APKs will be decompiled along with framework jars
[-] '/priv-app/DiagMon/DiagMon.apk' de-optimization failed consider manual inspection - skipping archive
[-] '/priv-app/DMService/DMService.apk' de-optimization failed consider manual inspection - skipping archive
[-] '/priv-app/CNEService/CNEService.apk' de-optimization failed consider manual inspection - skipping archive
[-] '/priv-app/LifeTimerService/LifeTimerService.apk' de-optimization failed consider manual inspection - skipping archive
[-] '/priv-app/ConnMO/ConnMO.apk' de-optimization failed consider manual inspection - skipping archive
[-] '/priv-app/SprintDM/SprintDM.apk' de-optimization failed consider manual inspection - skipping archive
[-] '/priv-app/DCMO/DCMO.apk' de-optimization failed consider manual inspection - skipping archive
[-] '/priv-app/HotwordEnrollment/HotwordEnrollment.apk' de-optimization failed consider manual inspection - skipping archive
[-] '/app/RCSBootstraputil/RCSBootstraputil.apk' de-optimization failed consider manual inspection - skipping archive
[-] '/app/HiddenMenu/HiddenMenu.apk' de-optimization failed consider manual inspection - skipping archive
[-] '/app/RcsImsBootstraputil/RcsImsBootstraputil.apk' de-optimization failed consider manual inspection - skipping archive
[-] 'out/bullhead/nrd90m/factory_imgs_data/system/framework/conscrypt.jar' not pre-optimized & without 'classes.dex' - skipping
[-] '/framework/android.test.runner.jar' de-optimization failed consider manual inspection - skipping archive
[-] '/framework/uiautomator.jar' de-optimization failed consider manual inspection - skipping archive
[-] 'out/bullhead/nrd90m/factory_imgs_data/system/framework/okhttp.jar' not pre-optimized & without 'classes.dex' - skipping
[*] '/framework/rcsservice.jar' not pre-optimized with sanity checks passed - copying without changes
[-] '/framework/javax.obex.jar' de-optimization failed consider manual inspection - skipping archive
[-] 'out/bullhead/nrd90m/factory_imgs_data/system/framework/ims-common.jar' not pre-optimized & without 'classes.dex' - skipping
[*] '/framework/cneapiclient.jar' not pre-optimized with sanity checks passed - copying without changes
[-] '/framework/wifi-service.jar' de-optimization failed consider manual inspection - skipping archive
[*] '/framework/rcsimssettings.jar' not pre-optimized with sanity checks passed - copying without changes
[-] 'out/bullhead/nrd90m/factory_imgs_data/system/framework/ext.jar' not pre-optimized & without 'classes.dex' - skipping
[-] 'out/bullhead/nrd90m/factory_imgs_data/system/framework/org.apache.http.legacy.boot.jar' not pre-optimized & without 'classes.dex' - skipping
[-] '/framework/com.google.android.media.effects.jar' de-optimization failed consider manual inspection - skipping archive
[-] '/framework/com.google.android.maps.jar' de-optimization failed consider manual inspection - skipping archive
[-] 'out/bullhead/nrd90m/factory_imgs_data/system/framework/framework.jar' not pre-optimized & without 'classes.dex' - skipping
[-] 'out/bullhead/nrd90m/factory_imgs_data/system/framework/core-oj.jar' not pre-optimized & without 'classes.dex' - skipping
[-] 'out/bullhead/nrd90m/factory_imgs_data/system/framework/bouncycastle.jar' not pre-optimized & without 'classes.dex' - skipping
[-] 'out/bullhead/nrd90m/factory_imgs_data/system/framework/apache-xml.jar' not pre-optimized & without 'classes.dex' - skipping
[-] '/framework/services.jar' de-optimization failed consider manual inspection - skipping archive
[-] '/framework/monkey.jar' de-optimization failed consider manual inspection - skipping archive
[-] 'out/bullhead/nrd90m/factory_imgs_data/system/framework/voip-common.jar' not pre-optimized & without 'classes.dex' - skipping
[-] 'out/bullhead/nrd90m/factory_imgs_data/system/framework/core-libart.jar' not pre-optimized & without 'classes.dex' - skipping
[-] 'out/bullhead/nrd90m/factory_imgs_data/system/framework/telephony-common.jar' not pre-optimized & without 'classes.dex' - skipping
[*] Skipping 'framework-res' since it doesn't pair with bytecode
[-] '/framework/pm.jar' de-optimization failed consider manual inspection - skipping archive
[-] 'out/bullhead/nrd90m/factory_imgs_data/system/framework/core-junit.jar' not pre-optimized & without 'classes.dex' - skipping
[*] System partition successfully extracted & de-optimized at 'out/bullhead/nrd90m/factory_imgs_repaired_data'
[!] Target device expects to have following img versions when using output system img
 [*] Booatloder:BHZ11e
 [*] Baseband:M8994F-2.6.33.2.14
[*] Generating blobs for vendor/lge/bullhead
[*] Copying files to 'out/bullhead/nrd90m/vendor/lge/bullhead' ...
cp: impossible d'évaluer « out/bullhead/nrd90m/factory_imgs_repaired_data/system/app/HiddenMenu/HiddenMenu.apk »: Aucun fichier ou dossier de ce type
[-] Vendor generation failed
```

Did I do something wrong?

anestisb, 2016-09-08:
@rvimont you're using master branch. Switch to master-dev branch to test Nougat support. Since not all issues are fixed yet I'll not merge with master until it reaches a stable point.

Yeah dropping root when not necessary is in the todo list, although I don't bother for the time being dealing with all the Linux distro headaches of sudo/su wrapping. I'll start looking at it as soon as I deal with the Nougat compatibility issues. Feel free to contribute if you have time and ideas on how to deal with them.

rvimont, 2016-09-08:
OK thanks :+1:

mikalackis, 2016-09-09:
@anestisb I'm still trying to build with your master-dev and I've encountered this issue while building the ROM:
device/lge/bullhead/location/loc_api/loc_api_v02/location_service_v02.h:78:10: fatal error: 'qmi_idl_lib.h' file not found

That file is nowhere to be found in my source tree. I checked out that you wrote something about libloc_api_v02.so and I do have that file in the vendor generated directory so I'm quite stuck now :) Have you encountered this issue?

anestisb, 2016-09-09:
Yeah it has been effectively fixed by https://github.com/anestisb/android-prepare-vendor/commit/95c297ca9fe14114cf796621cc8dda01c919347c. You have probably already generated them by the time a pushed the change. Git pull latest version from master-dev, re-genarate vendor with the scripts and retry the build.

There is no need to clean AOSP from you previous build. Changes will be picked automatically.

mikalackis, 2016-09-09:
Thanks man, didnt check for commits on master-dev since yesterday and I see you added quite a few :) Ok, doing it again and I'll update you

anestisb, 2016-09-10:
@mikalackis I'm refactoring a lot of stuff in the master-dev branch so it's quite unstable. I'll update here when N5x support reaches a stable state. Today seems to be stable enough for some build with the drop-root support via using fuse-ext2 module. You can give it a try if you want, but I've to want you that it's still on beta phase. Of course feedback is always welcome.

Hopefully, I'll have something stable within the next few days.

anestisb, 2016-09-11:
Support completed & merged with master.

closed 2016-09-11 by anestisb