anestisb/android-prepare-vendor #16: Android-N support for Nexus 6p

anestisb, 2016-09-01:


anestisb, 2016-09-07:
Still missing factory images from Google (https://developers.google.com/android/nexus/images#angler)

thestinger, 2016-09-08:
They only released Nougat OTAs to the beta channel. They haven't upgraded anyone on the normal channel, so it makes sense that those are missing. No kernel source code yet either...

thestinger, 2016-09-09:
They finally uploaded these factory images and started the release.

anestisb, 2016-09-09:
Thanks. Will start working on it since lots of bits have been shifted around.

anestisb, 2016-09-09:
Seems that a lot of stuff have been moved to /vendor partition, although I'm particularly worried about the symlinks located at old /system locations.

e.g.
```
root@bchfs:angler# ls -la nrd90u/factory_imgs_data/system/etc/sap.conf
lrwxrwxrwx 1 root root 20 Jan  1  2009 nrd90u/factory_imgs_data/system/etc/sap.conf -> /vendor/etc/sap.conf
```

Currently scripts cannot deal with symlinks automation in configuration files, except the JNI libraries of APKs located under /system/lib(64). If these new symlinks break functionality, I need to find a way to deal with them in the scripts.

anestisb, 2016-09-09:
```
root@bchfs:nrd90u# find -L factory_imgs_data/system -type l | grep -v "lib/arm" | grep -v 'system/vendor'
factory_imgs_data/system/bin/ims_rtp_daemon
factory_imgs_data/system/bin/imsqmidaemon
factory_imgs_data/system/bin/imsdatadaemon
factory_imgs_data/system/bin/imscmservice
factory_imgs_data/system/etc/lowi.conf
factory_imgs_data/system/etc/data/netmgr_config.xml
factory_imgs_data/system/etc/data/dsi_config.xml
factory_imgs_data/system/etc/cne/profile.txt
factory_imgs_data/system/etc/sap.conf
factory_imgs_data/system/etc/izat.conf
factory_imgs_data/system/etc/pp_calib_data_SAMSUNG_S6E3HA3X01_5P7_1440P_CMD_DUAL0.xml
factory_imgs_data/system/etc/flp.conf
```

anestisb, 2016-09-11:
Completed in master-dev branch. Will merge with master when N5x testing is also completed.

closed 2016-09-11 by anestisb