anestisb/android-prepare-vendor #18: smali not supporting Android N (oat ver. 079)

anestisb, 2016-09-03:
Bytecode de-optimization util is using [baksmali / smali](https://github.com/JesusFreke/smali) tools, which currently don't support Android N ART changes (https://github.com/JesusFreke/smali/issues/378).

The second (although minor) problem is that old "boot.oat" is now split into multiple files prefixed with boot-XXX.oat and having their own ART image file. We'll probably have to deal with it manually, unless SmaliEx maintainers quickly pick up this and implement it.

Support for Android N is blocked until smali project (and then by extend SmaliEx which we actually use to simplify baksmali -> smali iterations) supports recent ART changes.

```
java -jar /aosp_b_prod/prepare_vendor_blobs/hostTools/Java/baksmali.jar -x -o $(pwd)/deopt -c "$(find boot64 -name "*.oat" -printf "%f\n" | while read -r file; do echo -ne "$file:"; done)" -d $(pwd)/boot64 atfwd.odex

Error occurred while loading boot class path files. Aborting.
java.lang.StringIndexOutOfBoundsException: String index out of range: 1599362287
       	at java.lang.String.checkBounds(String.java:385)
       	at java.lang.String.<init>(String.java:462)
       	at org.jf.dexlib2.dexbacked.OatFile$1$1.next(OatFile.java:172)
       	at org.jf.dexlib2.dexbacked.OatFile$1$1.next(OatFile.java:159)
       	at org.jf.dexlib2.analysis.ClassPath.fromClassPath(ClassPath.java:191)
       	at org.jf.baksmali.baksmali.disassembleDexFile(baksmali.java:67)
       	at org.jf.baksmali.main.run(main.java:113)
       	at org.jf.baksmali.main.main(main.java:322)
```

anestisb, 2016-09-03:
cc @thestinger since it will affect CopperheadOS

anestisb, 2016-09-24:
Adding https://github.com/JesusFreke/smali/issues/418 for tracking reference

nychitman1, 2016-10-03:
@anestisb https://github.com/JesusFreke/smali/issues/378 -- JF updated to indicate support now works.

anestisb, 2016-10-04:
Thanks for the update @nychitman1

anestisb, 2016-10-04:
Support is still experimental. Users want to test can use the `--smali` flag from master script to enable it.

closed 2016-10-04 by anestisb