anestisb/android-prepare-vendor #21: Upgrade shared libs from PRODUCT_COPY_FILES to individual modules

anestisb, 2016-09-05:
Nougat AOSP sources indicate that some components have dependencies with proprietary shared objects that are not available from Google. Such an example is bullhead loc_api_V02 (https://android.googlesource.com/device/lge/bullhead/+/android-7.0.0_r1/location/loc_api/loc_api_v02/Android.mk#20)

As such we can no longer include the shared objects that are extracted from factory images as part of the PRODUCT_COPY_FILES option. We need to create a separate target for at least all shared objects that are dependencies of AOSP components. Since we can't easily identify those we need to upgrade the toolchain to create a separate target for all shared objects that need to be included from factory images.

The format should be something like this:
```
include $(CLEAR_VARS)
LOCAL_MODULE := libqmi_cci
LOCAL_SRC_FILES := vendor/lib64/libqmi_cci.so
LOCAL_MODULE_SUFFIX := .so
LOCAL_MODULE_CLASS := SHARED_LIBRARIES
LOCAL_MODULE_PATH := $(PRODUCT_OUT)/$(TARGET_COPY_OUT_VENDOR)/lib64
LOCAL_MODULE_TAGS := optional
LOCAL_MODULE_OWNER := lge
include $(BUILD_PREBUILT)
```

We also need to identify a compatible way to deal with MULTILIB scenarios. Probably search if 32bit version is also available and set the `LOCAL_SRC_FILES_32` flag.

anestisb, 2016-09-05:
First attempt at master-dev branch (https://github.com/anestisb/android-prepare-vendor/commit/63b188da1e46679646dd9a955d5592db2905bbe5). Now running tests for N9 & N5x Nougat releases to confirm it works with AOSP build chains.

anestisb, 2016-09-05:
Error when building against N5x:

```
build/core/base_rules.mk:183: *** vendor/lge/bullhead: MODULE.TARGET.SHARED_LIBRARIES.libgps.utils already defined by hardware/qcom/gps/msm8994/utils.
build/core/ninja.mk:163: recipe for target 'out_userdebug_Linux_7.0.0_r1/sources/build-aosp_bullhead.ninja' failed
```

Fixed by updating N5x blobs list at https://github.com/anestisb/android-prepare-vendor/commit/9562a526f2bb60e01dd829e9b29757188fb3505a

anestisb, 2016-09-05:
More N5x errors:

```
build/core/base_rules.mk:183: *** vendor/lge/bullhead: MODULE.TARGET.SHARED_LIBRARIES.libloc_api_v02 already defined by device/lge/bullhead/location/loc_api/loc_api_v02.
build/core/ninja.mk:163: recipe for target 'out_userdebug_Linux_7.0.0_r1/sources/build-aosp_bullhead.ninja' failed
```

Fixed by updating N5x blobs list at https://github.com/anestisb/android-prepare-vendor/commit/14beaf5573e1b3568d36c54ee5d9aa7e116a74d9

anestisb, 2016-09-05:
More N5x errors:

```
build/core/base_rules.mk:183: *** vendor/lge/bullhead: MODULE.TARGET.SHARED_LIBRARIES.libloc_ds_api already defined by device/lge/bullhead/location/loc_api/ds_api.
build/core/ninja.mk:163: recipe for target 'out_userdebug_Linux_7.0.0_r1/sources/build-aosp_bullhead.ninja' failed
```

Fixed by updating N5x blobs list at https://github.com/anestisb/android-prepare-vendor/commit/dfa6f1e312ca24638e3645442201287eb72db748

anestisb, 2016-09-05:
More N5x errors:

```
build/core/base_rules.mk:183: *** vendor/lge/bullhead: MODULE.TARGET.SHARED_LIBRARIES.libloc_eng already defined by hardware/qcom/gps/msm8994/loc_api/libloc_api_50001.
build/core/ninja.mk:163: recipe for target 'out_userdebug_Linux_7.0.0_r1/sources/build-aosp_bullhead.ninja' failed
```

Fixed by updating N5x blobs list at https://github.com/anestisb/android-prepare-vendor/commit/5a0d4c19779cc203ed453f2024fd452dd1d6eb20

anestisb, 2016-09-06:
The bulk approach wasn't working as expected, as such only manually selected shared libraries are marked for individual target module generation. More info at "https://github.com/anestisb/android-prepare-vendor/commit/858085612c05b52950155839e019935849dfb6ca" commit log.

anestisb, 2016-09-06:
N5x AOSP builds now seem compatible with:
* https://github.com/anestisb/android-prepare-vendor/blob/master-dev/bullhead/shared-proprietary-blobs-api24.txt

N9 doesn't need any shared libraries blobs

N6p libs list is also prepared although due to missing factory image cannot be tested yet
* https://github.com/anestisb/android-prepare-vendor/blob/master-dev/angler/shared-proprietary-blobs-api24.txt

Next step if to extensively test N5x build combined with new bytecode repair method (https://github.com/anestisb/android-prepare-vendor/issues/22).

anestisb, 2016-09-08:
Feature completed & tested. One less thing to worry for Nougat support.

closed 2016-09-08 by anestisb

thestinger, 2016-09-08:
It's only those location components that were a problem for us.

I think it's feasible to convince Google developers to enable `LOCAL_DEX_PREOPT := unstripped` or `false` for the vendor apps, but it's a lot trickier to get them to do it for Nougat since it's already released.