anestisb/android-prepare-vendor #25: AOSP disables verity for the vendor partition on the 5X and 6P (not 9)

thestinger, 2016-09-09:
This seems relevant to android-prepare-vendor, since it goes through a lot of pain to get a proper build of vendor.img where verity can be used:

* https://android.googlesource.com/device/lge/bullhead/+/7fecf8ad27e41fc1ccfc586bac8ffe1c66016873
* https://android.googlesource.com/device/huawei/angler/+/4ff4fb8d3ceb14e1c10811acb65c8feafa6347e1

Not sure if you can override it, but you might want to document it.

I wonder if there's any chance of landing the vendor reassembly scripts upstream. Their blob list already expects you to bundle a stock vendor.img. If they actually moved everything they should to vendor, they could support a sane build process again with only that.

anestisb, 2016-09-09:
Yeah noticed that recently too, although didn't have the time to dig further. I think the extra flags automation added recently at https://github.com/anestisb/android-prepare-vendor/commit/95c297ca9fe14114cf796621cc8dda01c919347c will enable us to override it from generated makefiles Will chase it as soon as I finish with N5x / N6p blob lists updates and update this thread.

To be honest after some basic compatibility with Nougat for supported Nexus devices is achieved, I'll assemble a small draft with all this paranoia and try to escalate / chase with Google. There is no way we can spend so many work hours every time new major release is out just to get the basics work. The whole thing is chaotic and I'm ending-up doing workaround for workarounds of major hacks. I'm mostly using AOSP with Nexus devices to develop pentest ROMs with security tools built-in, but the whole process is getting too expensive. If we don't find some middle ground for the next major release I'm not sure I'll be maintaining this project.

anestisb, 2016-09-09:
Btw another evidence of how fragile things are in AOSP regarding this issue.

Root device mk for AOSP version `aosp_bullhead.mk`, disables dm-verity for /vendor at line 34, although still includes the `device/lge/bullhead/device.mk` at line 36.
```
34: PRODUCT_COPY_FILES += device/lge/bullhead/fstab.aosp_bullhead:root/fstab.bullhead
35: 
36: $(call inherit-product, device/lge/bullhead/device.mk)
37: $(call inherit-product-if-exists, vendor/lge/bullhead/device-vendor.mk)
```

Where `device/lge/bullhead/device.mk` sets again the fstab setup at line 37 with same destination path.
```
34: PRODUCT_COPY_FILES += \
35:     device/lge/bullhead/init.bullhead.rc:root/init.bullhead.rc \
36:     device/lge/bullhead/init.bullhead.usb.rc:root/init.bullhead.usb.rc \
37:     device/lge/bullhead/fstab.bullhead:root/fstab.bullhead \
38:     device/lge/bullhead/ueventd.bullhead.rc:root/ueventd.bullhead.rc \
```

And from build core at lines [17-47](https://android.googlesource.com/platform/build/+/android-7.0.0_r6/core/Makefile#17) only the **first** src from `PRODUCT_COPY_FILES` is processed when more than one entries have the same dst. Since it's not documented at the commit log or inside comments I'm not sure if the committer was actually counting on that functionality or it's working by coincidence.

So my approach would be to append `PRODUCT_COPY_FILES`with the dm-verity enabled fstab before `aosp_bullhead.mk` is included. This will probably happen at the `BoardConfig.mk` level. Same for angler.

thestinger, 2016-09-09:
I guess the fact that android-prepare-vendor needs root for this would be a problem for pushing it upstream. Too bad there isn't a robust FUSE implementation of ext4. Maybe there's a reliable way to extract it though. I seem to remember that 7zip can do it...

jduck, 2016-09-09:
I've been using "fuseext2" without any problem thus far. Perhaps it can be useful for your projects.

anestisb, 2016-09-09:
I think that 7z might do the trick since it supports ext4 since 2015 (http://www.7-zip.org/history.txt). Have already started working it at drop-root branch [here](https://github.com/anestisb/android-prepare-vendor/tree/drop-root). Will start testing as soon as I finish the the N6p support.

@juck thanks for mentioning. That might actually prove useful for some projects I'm working on.

anestisb, 2016-09-11:
Fixed at https://github.com/anestisb/android-prepare-vendor/commit/d21a2419a924dc1ced156e851e6bec972c40d96b

closed 2016-09-11 by anestisb

thestinger, 2016-09-12:
I guess that's due to 7z being Windows-centric. It's unfortunate that it doesn't work since I think that would be cleaner than needing FUSE. Oh well. It's still better than a hard dependency on root.