anestisb/android-prepare-vendor #27: Invalid input directory structure

mikalackis, 2016-09-12:
Pulled down the latest code on master branch (although had the same issue with master-dev before your merge with master), here is the log with the error:
./execute-all.sh -d bullhead -b NRD90S -o nexus5x_nougat
[*] Setting output base to 'nexus5x_nougat/bullhead/nrd90s'
[*] Downloading image from 'https://dl.google.com/dl/android/aosp/bullhead-nrd90s-factory-d6bf1f56.zip'
--2016-09-12 09:33:34--  https://dl.google.com/dl/android/aosp/bullhead-nrd90s-factory-d6bf1f56.zip
Resolving dl.google.com (dl.google.com)... 172.217.19.110, 2a00:1450:4016:804::200e
Connecting to dl.google.com (dl.google.com)|172.217.19.110|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1243254222 (1,2G) [application/zip]
Saving to: ‘nexus5x_nougat/bullhead/nrd90s/bullhead-nrd90s-factory-d6bf1f56.zip’

100%[====================================================================================================================================================================>] 1.243.254.222  410KB/s   in 38m 46s

2016-09-12 10:12:20 (522 KB/s) - ‘nexus5x_nougat/bullhead/nrd90s/bullhead-nrd90s-factory-d6bf1f56.zip’ saved [1243254222/1243254222]

[*] Processing with 'API-24' configuration
[*] Extracting 'nexus5x_nougat/bullhead/nrd90s/bullhead-nrd90s-factory-d6bf1f56.zip'
[*] Unzipping 'image-bullhead-nrd90s.zip'
[-] Invalid input directory structure
  Usage: gen-prop-blobs-list.sh [options]
    OPTIONS:
      -i|--input      : Root path of /vendor partition
      -o|--output     : Path to save generated "proprietary-blobs.txt" file
      --sys-list      : File list with proprietary blobs in /system partition
      --bytecode-list : File list with proprietary bytecode archive files
      --dep-dso-list  : File list with DSO files with individual targets
[-] 'proprietary-blobs.txt' generation failed

Any help on this maybe? I will be available to perform any additional tests you ask :)

anestisb, 2016-09-12:
Most probably fuse-ext2 is failing for some reason to mount images, although doesn't return an error code. Run the `extract-factory-images.sh` in debug mode (uncomment `set -x` at line 9) and check the fuse-ext2 output.

Let me know that the error was so that I can improve the error handling and covert that case too.

mikalackis, 2016-09-12:
Here is the debug log: http://pastebin.com/ykB2JmBz

I dont see any explicit error reported in the log, do you see something interesting?

anestisb, 2016-09-12:
Damn I forgot the file log redirect part. Can you please remove the `&>"$MOUNT_LOG"` part of the mount command at line 93 so that you can get the complete output & repeat the debug step?

I'm looking now for ways to improve error handling of fuse-ext2.

mikalackis, 2016-09-12:
Ok, removed the part you mentioned and did two tests:
1. Non sudo test log: http://pastebin.com/iWB4b3xA
Seems that there is a permission problem with fuse-ext2
2. Ran the same test with sudo, log: http://pastebin.com/L0npyLp2
Seems that this did the trick, I have the APKs now and all the other blobs.

I tried running the system-img-repair script once the sudo worked but now I get this:
[*] All bytecode files under system partition will be repaired
[*] Repairing bytecode under /system partition using oatdump method
[-] '/app/CtsShimPrebuilt/CtsShimPrebuilt.apk' not pre-optimized & without 'classes.dex' - skipping
[*] '/app/Hangouts/Hangouts.apk' is multi-dex - adjusting recursive archive adds
find: paths must precede expression: *2*_repaired.dex

Anyways, im gonna wait for your info on logs regarding fuse-ext2 and ill go step by step :) Thnx for the help, appreciate it

anestisb, 2016-09-12:
Ok now makes sense were the problem is:
```
fusermount: failed to open /etc/fuse.conf: Permission denied
fusermount: option allow_other only allowed if 'user_allow_other' is set in /etc/fuse.conf
```

Previous lines indicate that:
* No read prermissions for `/etc/fuse.conf` -- `sudo chmod +r /etc/fuse.conf` to fix
* `user_allow_other` option not set in fuse.conf -- Edit `/etc/fuse.conf` and add/edit the `user_allow_other` keyword

The whole idea is that we get rid of the sudo part. It's just that some installations distribute/deploy different configuration files.

thestinger, 2016-09-12:
The code checking `/etc/fuse.conf` is running as the current user though, isn't it? It could be tricked...

thestinger, 2016-09-12:
It seems this is meant to protect the filesystem from itself, which makes it weird that they don't provide an easy way to override this even though it appears that is possible.

mikalackis, 2016-09-12:
Ok, at the moment of installation, my /etc/fuse.conf had that user_allow_other commented out, so I enabled it. I also did the chmod +r command too.
After that, I ran the .execute-all command again but it failed reading proprietary-blobs.txt with permission denied. Did the sudo execute-all again, here is the los:
http://pastebin.com/cnJffQj9

This line is probably where the error happened: 
 [[ ! -d nexus5x_nougat/bullhead/nrd90s/factory_imgs_repaired_data/vendor ]]

I checked out the dir structure and i have a system folder but vendor seems to be a symlink and its probably causing an error. Do you know what could I check about that?

anestisb, 2016-09-12:
@thestinger yeah it's indeed weird. I probably didn't notice since I've installed manually and not from distro packages.

fusermount seems to always default to `/etc/fuse.conf` without allowing user override of config:
https://github.com/libfuse/libfuse/blob/fuse_2_9_bugfix/util/fusermount.c#L38

On the other hand Ubuntu & Debian packages (https://launchpad.net/ubuntu/+source/fuse/2.8.6-2ubuntu2.1) seem to use the following perms for post-install script:
```
                if [ -z "$2" ]; then
                        chown root:fuse /etc/fuse.conf
                        chmod 0640 /etc/fuse.conf
                fi
```

So you either +r other or add user to fuse group. It seems kind of unnecessary restrictions from package maintainers, although I might miss something here.

I'm working now some improved error handling and extra notes around these errors. It's just the asynchronous background mode that complexes things since mount-points are not directly visible after mount. I want to avoid some sleep type of things and do something more elegant.

anestisb, 2016-09-12:
@mikalackis your previous run with sudo has created the `proprietary-blobs.txt` file under the target device dir, although your next run with low privilege user cannot override it when regenerating since it doesn't have the permissions to do it. Delete manually the file and repeat the process and all should be ok.

mikalackis, 2016-09-12:
Ok, I manually removed everything, did a clean execute but still having the same invalid directory structure. The symlink I mentioned (vendor) points to factory_imgs_data/vendor directory which in my case is empty. Is there any other log I could provide?

anestisb, 2016-09-12:
Just to clarify the latest status of your problem. System partition is mounted correctly and available for scripts to continue extracting data by vendor partition isn't?

mikalackis, 2016-09-12:
Yes, that seems to be the problem. In factory_imgs_data folder, system folder has all the files but the vendor folder is completely empty and thats whats probably preventing the script from further execution.

anestisb, 2016-09-12:
Alrighty thanks for the update. Not sure exactly were the problem is, but hopefully the improved error handling / logging changes I'm working will help us isolate it. Will update here when changes are ready so that you can rerun against your env.

Thanks for comments & feedback.

anestisb, 2016-09-12:
I've upgraded the error handling around fuse mounts in version 0.1.2. Can you please pull the latest version and retry after ensuring the following:

* You don't have any local git changes (git status && git diff)
* You haven't any mounted directories (mount & umount manually the ones accidentally left behind)
* The permissions of the entire project tree & the output directory are owned by the low privilege user that you're working with (use chown if you've modified the default ones or checked out the repo as root).

mikalackis, 2016-09-12:
Thanks for the heads up.

I did everything you said, I even removed everything and checked out your repo from scratch but still no luck. I tried with Nougat factory image and with Marshmallow factory image and on both it fails at generate-vendor.sh. I will post the logs tomorrow

mikalackis, 2016-09-13:
I tried it again today and still no luck. It seems that data in vendor and system folder in factory_img_data get removed at some point. I tricked the system by sleeping the generate-vendor script, then backing up these folders and then re-run just generate-vendor. Just flash my phone with a build with this vendor files but it doesnt boot for some reason

Crystal-Rom, 2016-09-13:
Had the same issue, pulled from master branch and there's no way to make it. i attached the log here, also tried  fuse-ext2 fixes (in fact i didn't notice any error about that). That's all i know
[debug_log.txt](https://github.com/anestisb/android-prepare-vendor/files/470382/debug_log.txt)

anestisb, 2016-09-13:
Ok spotted the problem. The lack of absolute path definition for the output location you're using is messing with the vendor symbolic link causing the problem.

I haven't come across the issue because I was always providing output directory as full path and not as a relative from current path. Running something like the following will resolve the issue for your cases:

```
anestisb@bchfs:android-prepare-vendor$ ./execute-all.sh -d bullhead -b NRD90S -o $(pwd)/out -k -i /aosp_b_prod/android-prepare-vendor/bullhead/nrd90s/bullhead-nrd90s-factory-d6bf1f56.zip
```

The important part is the `-o $(pwd)/out` argument. I'll add a path expand/complete at the master script to avoid such cases automatically.

By the way there is no reason to sudo the script as the whole idea of the fuses was to get rid of the root permission requirements for the mounts.

Crystal-Rom, 2016-09-13:
confirmed, it worked

mikalackis, 2016-09-13:
damn and i used the full path before... thanks for clearing this out, it worked :) im doing a new build with new vendor blobs, hopefully this time it will boot

anestisb, 2016-09-14:
Fixed.

closed 2016-09-14 by anestisb