anestisb/android-prepare-vendor #28: /system JARs missing signature

anestisb, 2016-09-12:
Currently vendor generate script is marking all framework JARs under /system as pre-signed.

```
include $(CLEAR_VARS)
LOCAL_MODULE := com.google.widevine.software.drm
LOCAL_MODULE_TAGS := optional
LOCAL_MODULE_OWNER := lge
LOCAL_MODULE_PATH := $(PRODUCT_OUT)/$(TARGET_COPY_OUT_SYSTEM)/framework
LOCAL_SRC_FILES := proprietary/framework/com.google.widevine.software.drm.jar
LOCAL_CERTIFICATE := PRESIGNED
LOCAL_MODULE_CLASS := JAVA_LIBRARIES
LOCAL_MODULE_SUFFIX := $(COMMON_JAVA_PACKAGE_SUFFIX)
LOCAL_DEX_PREOPT := false
include $(BUILD_PREBUILT)
```

However the new DEX repair method appends bytecode back to archive and deletes old signature. When marked as presigned it build chain skips zip-align/re-sign and thus breaking the archive.

Need to investigate if JARs need to be marked as pre-signed when extracted (thus not delete META-INF) or can be treated in same way as APKs and use `LOCAL_CERTIFICATE := platform` flag.

anestisb, 2016-09-12:
Tested and seems ok. Merged with master.

closed 2016-09-12 by anestisb

thestinger, 2016-09-12:
Thanks. Getting close to something I can consider a proper build :).