anestisb/android-prepare-vendor #30: Unnecessary Blobs Angler and Non-working makelinks script

krash86, 2016-09-13:
There are a ton of blobs that are not need for Angler. AOSP reproduces many of them. 

I have already submitted a pull request for some that were missed that are needed. Would you like me to slim down the ones that are not needed and submit another pull request?

Also I noticed that the AOSP makelinks script you have appended to a makefile in vendor is not working. The one in the tree does work, when called. See: https://github.com/krash86/android_device_huawei_angler/commit/faf48e289ede6cb7e02dc4ac21bd0d27a9ee3849

Thank you for your awesome work. This tool is a real gem of a find for me.

anestisb, 2016-09-14:
Yes please do. Anything that gets build from AOSP on both userdebug & release versions should be removed from the blobs list.

Regarding the extra symbolic links part I've noticed yesterday that some builds have it some not. Not sure why. Will investigate and let you know. As in general I want to avoid at all cost any changes that the AOSP tree since it's a maintenance cost I cannot afford for my use cases of the custom ROMS.

krash86, 2016-09-14:
I am running userdebug builds. I had to call the script that came in the tree for the links to propagate properly.

anestisb, 2016-09-17:
Ok the makelinks portion you've mentioned is fixed. Thanks for reporting.

Will have a look at the blob changes you've mentioned at first chance.

closed 2016-09-21 by anestisb