anestisb/android-prepare-vendor #31: N6P basename failing

Joshndroid, 2016-09-18:
I'm getting an issue in which the vendor generation keeps failing.

I have tried your versions over the last few days and still get the same error.

Using the angler NRD90U firmware i get down to this;

basename: invalid option -- '>'
Try 'basename --help' for more information.
[-] Vendor generation failed

is this an issue that is to surpassed? is there anything further required to assist?
This issue occurs whether -g flag used or not.

Thanks

anestisb, 2016-09-18:
Something in your system paths is causing the problem. Can you paste the exact command with the arguments that you're calling? Please put the data inside backquotes so that content is not altered by markdown render.

You can also enable the debug flag at the `execute-all.sh` and the `vendor-generate.sh` scripts and trace back the error from the fault basename command call. Debug mode can be enabled by uncommenting the `set -x` at the top of the scripts source code.

Joshndroid, 2016-09-18:
I am running this command - 

./execute-all.sh -d angler -b NRD90U -i /media/joshndroid/files/Device\ Backups/angler/angler-nrd90u-factory-7c9b6a2b.zip -o /media/joshndroid/files/Device\ Backups/angler/vendor -g

Joshndroid, 2016-09-18:
I changed the location of the input and output files. My guess is your script really doesnt like the space in the file directory, even though i was using correct syntax ' \ ' to parse the space. As all i did was change the input and output to new location with no space in the directory (everything else the same) then it completed successfully

closed 2016-09-18 by Joshndroid

anestisb, 2016-09-18:
Yeah there was a problem when output path was containing spaces. Issue has been fixed at commit https://github.com/anestisb/android-prepare-vendor/commit/f712d9136f5b10f336d1137a5d8e150645fe8695

Thanks for reporting.

Joshndroid, 2016-09-18:
Awesome, first time stumbling on your work. great stuff!!