anestisb/android-prepare-vendor #34: Include bootloader & radio images firmware

anestisb, 2016-09-24:
Update extract & vendor generate scripts to also include bootloader & baseband images at the generated vendor blobs.

anestisb, 2016-09-24:
Completed.

closed 2016-09-24 by anestisb

thestinger, 2016-09-24:
BTW, OTA updates with bootloader.img on the Nexus 5X require update-binary functionality that's not included in AOSP. It generates edify calls to some LG plugin. In CopperheadOS, I bundle the stock OTA update-binary.

anestisb, 2016-09-24:
@thestinger Yeah I remember discussing it a few months ago. For my use cases I don't mess with OTA so not really affected. Added a warning though at the README in case others are.

I was dealing with baseband & bootloader stuff with a separate set of script, although decided to merge them to minimize maintenance cost.