anestisb/android-prepare-vendor #37: Duplicates in vendor and device tree

Joshndroid, 2016-10-04:
Hey Anestisb,

I have a few nexus devices and no doubt have been using your tool to assist in extraction of vendor. i have noticed in both angler and bullhead that there are a couple of duplicates. I was thinking you may need an ifdef or something to decide whether to pull from the vendor or not. At present i just delete them as the issue arises when pulling from new factory images.

Files i see are; (in android.mk and in /lib and /lib64)
loc_api_v02
ds_api

from a guess they are gps related eh?

anestisb, 2016-10-05:
By duplicate I'm guessing you mean that you have one copy under /system/lib{64} and one under /vendor/lib{64}. Is that the case?

If so, then something has been mixed in your environment since I cannot reproduce. Have a look at the following for a few builds against Angler & Bullhead:
```
$ find out_* -iname "libloc_api_v02.so"
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/system/lib/libloc_api_v02.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/system/lib64/libloc_api_v02.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/symbols/system/lib/libloc_api_v02.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/symbols/system/lib64/libloc_api_v02.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/obj/lib/libloc_api_v02.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/obj/PACKAGING/target_files_intermediates/aosp_bullhead-target_files-eng.anestisb/SYSTEM/lib/libloc_api_v02.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/obj/PACKAGING/target_files_intermediates/aosp_bullhead-target_files-eng.anestisb/SYSTEM/lib64/libloc_api_v02.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/obj/SHARED_LIBRARIES/libloc_api_v02_intermediates/PACKED/libloc_api_v02.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/obj/SHARED_LIBRARIES/libloc_api_v02_intermediates/LINKED/libloc_api_v02.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/obj_arm/lib/libloc_api_v02.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/obj_arm/SHARED_LIBRARIES/libloc_api_v02_intermediates/PACKED/libloc_api_v02.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/obj_arm/SHARED_LIBRARIES/libloc_api_v02_intermediates/LINKED/libloc_api_v02.so
out_release_Linux_7.0.0_r6/aosp_prod/target/product/angler/symbols/vendor/lib/libloc_api_v02.so
out_release_Linux_7.0.0_r6/aosp_prod/target/product/angler/symbols/vendor/lib64/libloc_api_v02.so
out_release_Linux_7.0.0_r6/aosp_prod/target/product/angler/obj/lib/libloc_api_v02.so
out_release_Linux_7.0.0_r6/aosp_prod/target/product/angler/obj/PACKAGING/target_files_intermediates/aosp_angler-target_files-eng.anestisb/VENDOR/lib/libloc_api_v02.so
out_release_Linux_7.0.0_r6/aosp_prod/target/product/angler/obj/PACKAGING/target_files_intermediates/aosp_angler-target_files-eng.anestisb/VENDOR/lib64/libloc_api_v02.so
out_release_Linux_7.0.0_r6/aosp_prod/target/product/angler/obj/SHARED_LIBRARIES/libloc_api_v02_intermediates/PACKED/libloc_api_v02.so
out_release_Linux_7.0.0_r6/aosp_prod/target/product/angler/obj/SHARED_LIBRARIES/libloc_api_v02_intermediates/LINKED/libloc_api_v02.so
out_release_Linux_7.0.0_r6/aosp_prod/target/product/angler/vendor/lib/libloc_api_v02.so
out_release_Linux_7.0.0_r6/aosp_prod/target/product/angler/vendor/lib64/libloc_api_v02.so
out_release_Linux_7.0.0_r6/aosp_prod/target/product/angler/obj_arm/lib/libloc_api_v02.so
out_release_Linux_7.0.0_r6/aosp_prod/target/product/angler/obj_arm/SHARED_LIBRARIES/libloc_api_v02_intermediates/PACKED/libloc_api_v02.so
out_release_Linux_7.0.0_r6/aosp_prod/target/product/angler/obj_arm/SHARED_LIBRARIES/libloc_api_v02_intermediates/LINKED/libloc_api_v02.so
out_userdebug_Linux_7.0.0_r9/aosp_prod/target/product/angler/symbols/vendor/lib/libloc_api_v02.so
out_userdebug_Linux_7.0.0_r9/aosp_prod/target/product/angler/symbols/vendor/lib64/libloc_api_v02.so
out_userdebug_Linux_7.0.0_r9/aosp_prod/target/product/angler/obj/lib/libloc_api_v02.so
out_userdebug_Linux_7.0.0_r9/aosp_prod/target/product/angler/obj/PACKAGING/target_files_intermediates/aosp_angler-target_files-eng.anestisb/VENDOR/lib/libloc_api_v02.so
out_userdebug_Linux_7.0.0_r9/aosp_prod/target/product/angler/obj/PACKAGING/target_files_intermediates/aosp_angler-target_files-eng.anestisb/VENDOR/lib64/libloc_api_v02.so
out_userdebug_Linux_7.0.0_r9/aosp_prod/target/product/angler/obj/SHARED_LIBRARIES/libloc_api_v02_intermediates/PACKED/libloc_api_v02.so
out_userdebug_Linux_7.0.0_r9/aosp_prod/target/product/angler/obj/SHARED_LIBRARIES/libloc_api_v02_intermediates/LINKED/libloc_api_v02.so
out_userdebug_Linux_7.0.0_r9/aosp_prod/target/product/angler/vendor/lib/libloc_api_v02.so
out_userdebug_Linux_7.0.0_r9/aosp_prod/target/product/angler/vendor/lib64/libloc_api_v02.so
out_userdebug_Linux_7.0.0_r9/aosp_prod/target/product/angler/obj_arm/lib/libloc_api_v02.so
out_userdebug_Linux_7.0.0_r9/aosp_prod/target/product/angler/obj_arm/SHARED_LIBRARIES/libloc_api_v02_intermediates/PACKED/libloc_api_v02.so
out_userdebug_Linux_7.0.0_r9/aosp_prod/target/product/angler/obj_arm/SHARED_LIBRARIES/libloc_api_v02_intermediates/LINKED/libloc_api_v02.so
```

```
$ find out_* -iname "libloc_ds_api.so"
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/system/lib/libloc_ds_api.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/system/lib64/libloc_ds_api.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/symbols/system/lib/libloc_ds_api.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/symbols/system/lib64/libloc_ds_api.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/obj/lib/libloc_ds_api.so
out_release_Linux_7.0.0_r4/aosp_prod/target/product/bullhead/obj/PACKAGING/target_files_intermediates/aosp_bullhead-target_files-eng.anestisb/SYSTEM/lib/libloc_ds_api.so
[... cut ...]
As you can see there is one copy only of the mentioned libs under /system for Bullhead and under /vendor for Angler. Also it's not possible to compile the version of these libraries available in AOSP tree since they're broken due to missing private libraries.

My best guess is that you either mixed some makefiles between the two devices and end-up with these duplicates (do a clean generate & build to confirm) or you've included some other vendor files on the top that are messing with the ones generated from the script.

At any case I cannot help much if I cannot reproduce.

Joshndroid, 2016-10-05:
No, when building aosp (i even checked the android git device tree) and using clean vendor, it will state that there is a duplicate module for those listed above and that it is defined within the device/huawei/angler (or device/lge/bullhead). Removing those 2 libs from vendor, OR removing the offending packages from the device tree allow build to continue and GPS function. I have had this same issue from initial vendor creation and with this lastest round of source releases a few days ago (again starting clean). Does that make more sense?

anestisb, 2016-10-05:
Ok got you now. Since the mentioned module definitions from AOSP are broken (cannot get build due to missing private library dependencies) I tried to exclude them from building without modifying any AOSP source code.

This is achieved by un-setting the `BOARD_VENDOR_QCOM_GPS_LOC_API_HARDWARE` variable at the generated `vendor/huawei/angler/BoardConfigVendor.mk` Having it empty effectively means that the AOSP sources of the same modules are never included for build due to [this](https://android.googlesource.com/device/lge/bullhead/+/android-7.0.0_r14/location/loc_api/Android.mk).

If you get a duplicate definition when makefiles are parsed prior to start building, then it means that the above mechanism is somewhere broken. You need to chase what is different in your env / process that results in both being included. I have no reports for this issue before so something in your env is triggering the problem.

From my side I can remove those two libs from having a standalone target and let them be part of the bulk `PRODUCT_COPY_FILES` copy, although that is not defensive enough since users will have a half broken location library for their devices being build and might possible end up with duplicate libraries if Google at some point fixes the lib deps and we don't notice in time. So I prefer to leave it as is since it's an early indicator that something is not working as expected when building.

anestisb, 2016-10-11:
Closing since it doesn't seem like an issue here. Let me know if you have new evidence that indicate otherwise.

closed 2016-10-11 by anestisb