anestisb/android-prepare-vendor #39: how to fix the issue "fuse-ext2: command not found"

k3vinlusec, 2016-10-11:
root@k3vinlu:/home/k3vinlusec/Downloads/android-prepare-vendor-master# ./execute-all.sh -d bullhead -b NBD90W -o ./bullhead/
[*] Setting output base to '/home/k3vinlusec/Downloads/android-prepare-vendor-master/bullhead/bullhead/nbd90w'
....
....
[?] I have read and agree with the above terms and conditions - ACKNOWLEDGE [y|n]: y
[*] Downloading image from 'https://dl.google.com/dl/android/aosp/bullhead-nbd90w-factory-d2f11c5f.zip'
--2016-10-10 17:14:11--  https://dl.google.com/dl/android/aosp/bullhead-nbd90w-factory-d2f11c5f.zip
Resolving dl.google.com (dl.google.com)... 216.58.195.78, 2607:f8b0:4005:804::200e
Connecting to dl.google.com (dl.google.com)|216.58.195.78|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1243266718 (1.2G) [application/zip]
Saving to: ‘/home/k3vinlusec/Downloads/android-prepare-vendor-master/bullhead/bullhead/nbd90w/bullhead-nbd90w-factory-d2f11c5f.zip’

100%[========================================================================================>] 1,243,266,718 5.77MB/s   in 3m 31s 

2016-10-10 17:17:42 (5.63 MB/s) - ‘/home/k3vinlusec/Downloads/android-prepare-vendor-master/bullhead/bullhead/nbd90w/bullhead-nbd90w-factory-d2f11c5f.zip’ saved [1243266718/1243266718]

[*] Processing with 'API-24 config-naked' configuration
[*] Extracting '/home/k3vinlusec/Downloads/android-prepare-vendor-master/bullhead/bullhead/nbd90w/bullhead-nbd90w-factory-d2f11c5f.zip'
[*] Unzipping 'image-bullhead-nbd90w.zip'
[-] '/tmp/android_img_extract.MTMVz1/bullhead-nbd90w-factory-d2f11c5f/images/system.img.raw' mount failed
/home/k3vinlusec/Downloads/android-prepare-vendor-master/scripts/extract-factory-images.sh: line 93: fuse-ext2: command not found
[-] Factory images data extract failed

anestisb, 2016-10-11:
You need to install fuse-ext2 tool at your system. More info [here](https://github.com/alperakcan/fuse-ext2).

k3vinlusec, 2016-10-11:
thx， resovled

closed 2016-10-11 by anestisb