anestisb/android-prepare-vendor #40: nexus 6p in osx 10.11 can't build

yixiaoqingchen, 2016-10-11:
nexus 6p in osx 10.11 can't build

anestisb, 2016-10-11:
Can you provide something more specific to work with? An output log or error message?

yixiaoqingchen, 2016-10-12:
./execute-all.sh -d angler -b MTC20F -o ../vender/ -i ../angler-mtc20f-factory-4355fe06.zip -k

[*] Setting output base to '/Users/yuanfeng/androidcode/vender/angler/mtc20f'
[*] Processing with 'API-23 config-naked' configuration
[*] Extracting '/Users/yuanfeng/androidcode/angler-mtc20f-factory-4355fe06.zip'
[*] Unzipping 'image-angler-mtc20f.zip'
mv: /Users/yuanfeng/androidcode/vender/angler/mtc20f/factory_imgs_data/radio/radio-angler-angler-03.61.img: set owner/group (was: 501/0): Operation not permitted
mv: /Users/yuanfeng/androidcode/vender/angler/mtc20f/factory_imgs_data/radio/bootloader-angler-angler-03.54.img: set owner/group (was: 501/0): Operation not permitted
/Users/yuanfeng/androidcode/android-prepare-vendor/scripts/system-img-repair.sh: line 784: readarray: command not found
[-] System partition bytecode repair failed

yixiaoqingchen, 2016-10-12:
GNU bash, version 3.2.57(1)-release (x86_64-apple-darwin15)
Copyright (C) 2007 Free Software Foundation, Inc.

yixiaoqingchen, 2016-10-12:
after upgrad bash version to 4.3.46(1)-release. This problem has resolved. thanks

anestisb, 2016-10-12:
Yeah bash 4.x is required. Thanks for highlighting that. Have added an extra check [here](https://github.com/anestisb/android-prepare-vendor/commit/b2ab9a3d78cf97d15019466b3fd7d5d2911245db) to notify users with older bash versions.

closed 2016-10-12 by anestisb