anestisb/android-prepare-vendor #42: Breaks on build NBD90W

knoy, 2016-10-18:
```
michael@ubuntu:~/vendortest2/android-prepare-vendor$ ./execute-all.sh -d bullhead -b NBD90W -i bullhead-nbd90w-factory-d2f11c5f.zip -o output
[*] Setting output base to '/home/michael/vendortest2/android-prepare-vendor/output/bullhead/nbd90w'
[*] Processing with 'API-24 config-naked' configuration
[*] Extracting '/home/michael/vendortest2/android-prepare-vendor/bullhead-nbd90w-factory-d2f11c5f.zip'
[*] Unzipping 'image-bullhead-nbd90w.zip'
error: file_write: incomplete write
Cannot write output file
[-] simg2img failed to convert system.img from sparse
[-] Factory images data extract failed
```

thestinger, 2016-10-18:
You'er probably running out of space in /tmp.

knoy, 2016-10-19:
Yup, had a 6GB ramdisk setup in fstab for /tmp. Thought that would be enough
Cheers

closed 2016-10-19 by knoy