anestisb/android-prepare-vendor #45: Cannot do ANYTHING

404Anonymous, 2016-11-11:
Please fix this....
This is the result:
Input: 
android-prepare-vendor$ ./execute-all.sh -d bullhead -i /fast-datavault/nexus-vendor-blobs/bullhead/mtc20k/bullhead-mtc20k-factory-4a950470.zip -b mtc20k -o /fast-datavault/nexus-vendor-blobs
output:
[-] 'fusermount' command not found

anestisb, 2016-11-11:
You need to install [fuse-ext2](https://github.com/alperakcan/fuse-ext2) to use the scripts

404Anonymous, 2016-11-12:
Not helping still
installed that

anestisb, 2016-11-12:
The reason the scripts are failing is simply because the `fusermount` command (which should be installed with fuse-ext2) is not found in your system's `PATH`. If you don't provide any more info about your system I can't help much.

Did you install fuse-ext2 from project upstream github repo or from your system's package manager? Which Linux distribution are you running?

closed 2016-12-06 by anestisb