anestisb/android-prepare-vendor #46: Updater process ended with error 6

Crystal-Rom, 2016-11-11:
Me again, I had an issue in this week, basically (I don't know exactly with which update, but was with NBD90W's vendor files) but the generated otapackage.zip can't be flashed in twrp, when installing starts it gets stopped almost immediately with "Updater process ended with error 6" (and no explaination of what)
So I tried to manually edit the updater-script, checked EOLs, repaced with Unix's ones, reflashed, still error 6. 
This week i tried to extract from NRD91N's factory, and still the same error. Finally I tried importing into the aosp old NRD90S's vendor, that was extracted from factory in September(almost) compiled (after a make installclean) then I tried to flash the generated .zip and this time it worked, but I can't understand why.

anestisb, 2016-11-12:
I don't perform OTA build updates myself so I can't help much until you track down the problem to something more specific. As far as I'm aware few other projects are using the same scripts and deliver OTA builds for Nexus devices without any problems. So probably it's related to your environment somehow.

thestinger, 2016-11-12:
Nexus 5X is a special snowflake though.

thestinger, 2016-11-12:
That's already in the README though. Need to do the same thing as the release.sh script included in the CopperheadOS build instructions.

Crystal-Rom, 2016-11-13:
Yeah, actually this fixed the error, switching update-binary, thank you.

closed 2016-11-13 by Crystal-Rom