anestisb/android-prepare-vendor #47: symlinks in vendor are not generated

thestinger, 2016-12-05:
Nexus 6P has these symlinks within vendor and at the moment only the 2 for the app libraries are included:

```
vendor/app/ims/lib/arm64/libimscamera_jni.so
vendor/app/ims/lib/arm64/libimsmedia_jni.so
vendor/etc/firmware/wcd9320/wcd9320_anc.bin
vendor/etc/firmware/wcd9320/wcd9320_mad_audio.bin
vendor/etc/firmware/wcd9320/wcd9320_mbhc.bin
vendor/lib64/eglSubDriverAndroid.so
vendor/lib64/libEGL_adreno.so
vendor/lib64/libGLESv1_CM_adreno.so
vendor/lib64/libGLESv2_adreno.so
vendor/lib64/libq3dtools_adreno.so
vendor/lib64/libq3dtools_esx.so
```

thestinger, 2016-12-05:
These don't appear to be needed for anything, but it would be nice to replicate this as closely as possible to rule this out as a potential cause for future problems that come up.

anestisb, 2016-12-05:
Let's wait for Google to push the new tags. If they don't create the symlinks themselves from the device upstream makefiles we can push them at the `extra-modules-api24.txt` config as we already do for some that are not properly handled from AOSP.

closed 2016-12-06 by thestinger