anestisb/android-prepare-vendor #48: 7.1.1 (December security update) appears to break bytecode generation

thestinger, 2016-12-05:
I haven't looked into this in detail yet but the files are definitely still present with stripped apks alongside the oat files as before.

anestisb, 2016-12-05:
Both oatdump & smali bytecode repair methods are failing because the OAT version at API 25 has been bumped to `0x3038380`.

e.g. when enable debug
```
Invalid oat header for '/fast-datavault/nexus-vendor-blobs/bullhead/nmf26f/factory_imgs_data/system/app/HiddenMenu/oat/arm64/HiddenMenu.odex': Invalid oat version, expected 0x3037390, got 0x3038380.
```

I'll be able to build the standalone host tools from the art project for API 25 as soon as Google publishes the sources for 7.1.1 release. Will check the latest upstream smali repo to verify their current support status against new OAT & ART versions.

anestisb, 2016-12-07:
Both oatdump & smali bytecode repair methods appear to be working fine now. Have tested some builds for both N6p & N5x and haven't noticed any issues so far.

thestinger, 2016-12-07:
Yeah, this works well now.

closed 2016-12-07 by thestinger