anestisb/android-prepare-vendor #49: Vendor binaries for nexus 5x/6p and 7.1.1(NMF26F)

mikalackis, 2016-12-07:
I just checked out binaries page on google (https://developers.google.com/android/drivers) and it turns out there are binaries now for 5x and 6p. Im gonna try and and make a build with these, anyone else had a shot with these yet?

anestisb, 2016-12-07:
Had a quick look yesterday, although didn't fire any builds yet. Let us know how that worked for you.

Google provided blobs are still missing APK bytecode, although I'm not sure how much functionality is affected at 7.1 builds.

mikalackis, 2016-12-07:
Ok, did a build, flashed it. I get that idiotic 'There's an internal problem bla bla bla' message on boot. Other then that, didnt notice any problem but that message is something that bothers me on a personal level so im going back to using your script.

thestinger, 2016-12-07:
That's because vendor.img isn't being regenerated, so the fingerprint will not match (producing that warning) and verified boot won't work even though it works fine if you use android-prepare-vendor to get a proper build.

thestinger, 2016-12-07:
They started disabling verified boot for vendor.img in AOSP builds which is why you can boot at all. It's enabled again by android-prepare-vendor and works just fine. They should really release vendor.img as a tarball with a build system to package it up... no reason to do things the way they do it.

anestisb, 2016-12-17:
Well I think the pros/cons have been covered. In short note the currently available vendor blobs for Nexus devices do not appear to be feature complete for release AOSP based ROMs.

Feel free to re-open if you think there're more bits around to discuss.

closed 2016-12-17 by anestisb