anestisb/android-prepare-vendor #50: N6p /vendor audio config files diff between AOSP & prod

anestisb, 2016-12-10:
Contents of `/vendor/etc/audio_effects.conf` appear to be different between AOSP builds and production release images for Nexus 6p. The scripts automatically pick the config file when extracting data from production image, although the file is already included from master `device.mk` [here](https://android.googlesource.com/device/huawei/angler/+/android-7.1.1_r4/device.mk#69).

Since AOSP build system ignores destination duplicates of PRODUCT_COPY_FILES keeping only the first occurrence, the version we include from the scripts is ignored.

Investigate if we need to prioritize /vendor PRODUCT_COPY_FILES to always be selected from build system by appending them at the generated `BoardConfigVendor.mk ` instead of the current `angler-vendor-blobs.mk`

```
diff -u angler/nmf26f/factory_imgs_data/vendor/etc/audio_effects.conf /fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/angler/vendor/etc/audio_effects.conf
--- angler/nmf26f/factory_imgs_data/vendor/etc/audio_effects.conf	2009-01-01 10:00:00.000000000 +0200
+++ /fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/angler/vendor/etc/audio_effects.conf	2016-12-06 18:54:13.928473755 +0200
@@ -18,21 +18,21 @@
   visualizer_hw {
     path /system/lib/soundfx/libqcomvisualizer.so
   }
-  fmas {
-    path /system/lib/soundfx/libfmas.so
+  downmix {
+    path /system/lib/soundfx/libdownmix.so
   }
   proxy {
     path /system/lib/soundfx/libeffectproxy.so
   }
-  loudness_enhancer {
-    path /system/lib/soundfx/libldnhncr.so
-  }
   offload_bundle {
     path /system/lib/soundfx/libqcompostprocbundle.so
   }
   qcom_pre_processing {
     path /system/lib/soundfx/libqcomvoiceprocessing.so
   }
+  loudness_enhancer {
+    path /system/lib/soundfx/libldnhncr.so
+  }
 }
 
 # Default pre-processing library. Add to audio_effect.conf "libraries" section if
@@ -95,8 +95,18 @@
     }
   }
   virtualizer {
-    library fmas
-    uuid 36103c50-8514-11e2-9e96-0800200c9a66
+    library proxy
+    uuid d3467faa-acc7-4d34-acaf-0002a5d5c51b
+
+    libsw {
+      library bundle
+      uuid 1d4033c0-8557-11df-9f2d-0002a5d5c51b
+    }
+
+    libhw {
+      library offload_bundle
+      uuid 509a4498-561a-4bea-b3b1-0002a5d5c51b
+    }
   }
   equalizer {
     library proxy
@@ -187,8 +197,8 @@
     }
   }
   downmix {
-    library fmas
-    uuid 36103c51-8514-11e2-9e96-0800200c9a66
+    library downmix
+    uuid 93f04452-e4fe-41cc-91f9-e475b6d1d69f
   }
   loudness_enhancer {
     library loudness_enhancer
@@ -204,6 +214,15 @@
   }
 }
 
+pre_processing {
+    voice_communication {
+        aec {
+        }
+        ns {
+        }
+    }
+}
+
 # Default pre-processing effects. Add to audio_effect.conf "effects" section if
 # audio HAL implements support for them.
 #
@@ -256,11 +275,6 @@
 #        ...
 #    }
 
-pre_processing {
-    voice_communication {
-        aec {
-        }
-        ns {
-        }
-    }
-}

```

thestinger, 2016-12-23:
Could you filter it out of PRODUCT_COPY_FILES? Is there any reason not to modify it?

anestisb, 2016-12-23:
AOSP device makefiles structure has two ways of importing external makefiles from `ANDROID_BUILD_TOP/vendor`:
* Directly via `-include <file>` statement (include is file present)
* Via a call to product inherit help macros `(call inherit-product, <file>)`

Dealing with import order or variable value overrides is straightforward for the first case, although not always possible for the second since the order matters. The product.mk helper macros maintain internally a few lists making it impossible to override at later point something already set (unless a side-way exists), since the previous sets are already pushed in the lists. That includes the list of PRODUCT_COPY_FILES already defined from another makefile before the one we try to apply some additional logic.

I've bypassed this restriction in the past by using the 'BoardConfigVendor.mk` which is always included with the first method very early in the makefile parsing process. As such any PRODUCT_COPY_FILES defined there can be placed at the top of the runtime lists maintained from product.mk. As such they will always reach output regardless of potential double includes from other AOSP device specific makefiles (just some warnings are printed when building).

Some ROM builders prefer to modify the AOSP root build makefiles to always consider the last PRODUCT_COPY_FILES destination instead of the first. However, I prefer a solution that doesn't involve modifying the AOSP root src tree.

Until I noticed this inconsistency for N6p, I haven't come across any AOSP device makefiles messing  with /vendor blobs in such a way.

So long story short, I know how to fix this inconsistency, although I need to test if it's actually required and that it does not break anything. At any case the current state of angler blobs seem to have fully functional audio stacks, so not sure yet if and how angler is affected. Bullhead and flounder does not appear to have this issue.

Testing will have to wait a few days since I don't have an available N6p device at the moment to run it.

thestinger, 2016-12-23:
Does bullhead have the same audio_effects.xml or is it just in /system/etc instead of vendor? It might still be a mismatch with the stock one.

anestisb, 2016-12-23:
Both angler & bullhead have `audio_effects.conf` under both `/system/etc` and `/vendor/etc`. Not sure where each one is used but they are never the same.

```
$ find /fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/angler -name "audio_effects.conf"
/fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/angler/system/etc/audio_effects.conf
/fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/angler/obj/PACKAGING/target_files_intermediates/aosp_angler-target_files-eng.anestisb/SYSTEM/etc/audio_effects.conf
/fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/angler/obj/PACKAGING/target_files_intermediates/aosp_angler-target_files-eng.anestisb/VENDOR/etc/audio_effects.conf
/fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/angler/vendor/etc/audio_effects.conf
$ find /fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/bullhead -name "audio_effects.conf"
/fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/bullhead/system/etc/audio_effects.conf
/fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/bullhead/obj/PACKAGING/target_files_intermediates/aosp_bullhead-target_files-eng.anestisb/SYSTEM/etc/audio_effects.conf
/fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/bullhead/obj/PACKAGING/target_files_intermediates/aosp_bullhead-target_files-eng.anestisb/VENDOR/etc/audio_effects.conf
/fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/bullhead/vendor/etc/audio_effects.conf
```

For bullhead, both configs match the release image:
```
$ diff -u bullhead/nmf26f/factory_imgs_data/vendor/etc/audio_effects.conf /fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/bullhead/vendor/etc/audio_effects.conf
$ diff -u bullhead/nmf26f/factory_imgs_data/system/etc/audio_effects.conf /fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/bullhead/system/etc/audio_effects.conf
```

On the other hand, angler matches only /system/etc one but not the vendor:
```
$ diff -u angler/nmf26f/factory_imgs_data/system/etc/audio_effects.conf /fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/angler/system/etc/audio_effects.conf
$ diff -u angler/nmf26f/factory_imgs_data/vendor/etc/audio_effects.conf /fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/angler/vendor/etc/audio_effects.conf
--- angler/nmf26f/factory_imgs_data/vendor/etc/audio_effects.conf	2009-01-01 10:00:00.000000000 +0200
+++ /fast-datavault/aosp_prod/out_release_Linux_7.1.1_r1/aosp_prod/target/product/angler/vendor/etc/audio_effects.conf	2016-12-06 18:54:13.928473755 +0200
@@ -18,21 +18,21 @@
   visualizer_hw {
     path /system/lib/soundfx/libqcomvisualizer.so
   }
-  fmas {
-    path /system/lib/soundfx/libfmas.so
+  downmix {
+    path /system/lib/soundfx/libdownmix.so
   }
   proxy {
     path /system/lib/soundfx/libeffectproxy.so
   }
-  loudness_enhancer {
-    path /system/lib/soundfx/libldnhncr.so
-  }
   offload_bundle {
     path /system/lib/soundfx/libqcompostprocbundle.so
   }
   qcom_pre_processing {
     path /system/lib/soundfx/libqcomvoiceprocessing.so
   }
+  loudness_enhancer {
+    path /system/lib/soundfx/libldnhncr.so
+  }
 }

 # Default pre-processing library. Add to audio_effect.conf "libraries" section if
@@ -95,8 +95,18 @@
     }
   }
   virtualizer {
-    library fmas
-    uuid 36103c50-8514-11e2-9e96-0800200c9a66
+    library proxy
+    uuid d3467faa-acc7-4d34-acaf-0002a5d5c51b
+
+    libsw {
+      library bundle
+      uuid 1d4033c0-8557-11df-9f2d-0002a5d5c51b
+    }
+
+    libhw {
+      library offload_bundle
+      uuid 509a4498-561a-4bea-b3b1-0002a5d5c51b
+    }
   }
   equalizer {
     library proxy
@@ -187,8 +197,8 @@
     }
   }
   downmix {
-    library fmas
-    uuid 36103c51-8514-11e2-9e96-0800200c9a66
+    library downmix
+    uuid 93f04452-e4fe-41cc-91f9-e475b6d1d69f
   }
   loudness_enhancer {
     library loudness_enhancer
@@ -204,6 +214,15 @@
   }
 }

+pre_processing {
+    voice_communication {
+        aec {
+        }
+        ns {
+        }
+    }
+}
+
 # Default pre-processing effects. Add to audio_effect.conf "effects" section if
 # audio HAL implements support for them.
 #
@@ -256,11 +275,6 @@
 #        ...
 #    }

-pre_processing {
-    voice_communication {
-        aec {
-        }
-        ns {
-        }
-    }
-}
+#
+# TODO: add default audio pre processor configurations after debug and tuning phase
+#
```

anestisb, 2016-12-28:
Have pushed a fix (https://github.com/anestisb/android-prepare-vendor/commit/e6020a9f504fea87bd5ab7e72c57049d5998aab8) for this at the [vendor_priority](https://github.com/anestisb/android-prepare-vendor/tree/vendor_priority) branch.

However, so far I haven't noticed any problems with the audio stacks. @thestinger do you have different evidence? I've seen that you pushed a commit at the angler device mk preventing the AOSP version from being including.

I'm considering not merging this fix yet since it's another hack on the top of the others. Trying to keep them small in number since things will probably break again at next release.

thestinger, 2016-12-28:
I don't have any evidence of issues on the N6P. N9 audio is really weird (the UI sounds are ridiculously loud) but it's not clear why, particularly since the N9 barely has anything in system so there's less that can go wrong. Maybe it's a BoardConfig.mk issue.

thestinger, 2016-12-28:
I just decided that I'd rather be shipping the same file as Google. Far more chance of it being correct than one that they never test / don't care about. I doubt that the config is meant to be different for AOSP + vendor blobs properly bundled.

anestisb, 2016-12-28:
Makes sense. Will leave things as is for now, although will keep an eye for similar cases while working the Pixel compatibility.

closed 2016-12-28 by anestisb