anestisb/android-prepare-vendor #51: Unbound variable error

kcy1019, 2017-01-04:
Error:
`/home/lucent/android-prepare-vendor/scripts/generate-vendor.sh: Line 484: PKGS_SSLINKS: unbound variable`
Cause:
`PKGS_SSLINKS` is an empty variable(no packages are added to the variable)

Fix:
Test whether `PKGS_SSLINKS` is empty before using it.

Error can be recreated with this command:
`sudo ./execute-all.sh -d bullhead -b MTC20K -o ../ -g -y --smaliex`

Environment:
- elementary OS 0.4 Loki  (based on Ubuntu 16.04 / xenial)
- GNU bash, version 4.3.46(1)-release (x86_64-pc-linux-gnu)

anestisb, 2017-01-04:
Xmm your issue brought to surface a few problematic bits.

* `MTC20K` for bullhead is missing the following bytecode packages. Can you confirm that? If that is the case something is terrible wrong with the `MTC20K` images since they deviate from the rest of Marshmallow builds.
  * system/app/HiddenMenu/HiddenMenu.apk
  * system/app/RCSBootstraputil/RCSBootstraputil.apk
  * system/app/RcsImsBootstraputil/RcsImsBootstraputil.apk
* `MTC20K` has been released at the same time with Nougat `NRD90M`, so if I'm not mistaken it actually never reached production OTA. Have you verified that the factory `MTC20K` images are working? To be honest the last Marshmallow build I've worked with was the `MTC20F` and not `MTC20K`.
* `PKGS_SSLINKS` is indeed undefined (no elements appended to array) if no symbolic links are detected from factory images (`S_SLINKS_SRC` is empty). However, so far I've never came across any factory images without any symbolic links. So I'm guessing that something is not working as expected at your env.

I can accept your patch (thanks btw), although I'm guessing that the generated output will be incomplete. Have you tried building AOSP against the generated bullhead blobs?

anestisb, 2017-01-04:
Ignore the first bullet. It seems that we've hit another bug that is triggered only when generating from a macOS. It seems that either simg2img or fuse-ext2 (probably the later) is not working as expected missing entries. The following diff indicates files missed when generating from MAC:

```
$ diff -u mac_mtc20k.txt linux_mtc20k.txt
--- mac_mtc20k.txt	2017-01-04 20:37:16.000000000 +0200
+++ linux_mtc20k.txt	2017-01-04 20:37:24.000000000 +0200
@@ -3,7 +3,13 @@
 bullhead/mtc20k/factory_imgs_data/system/app/Bluetooth/Bluetooth.apk:9260d0fb26c50ffb3da6e4069cc0e076
 bullhead/mtc20k/factory_imgs_data/system/app/Bluetooth/oat/arm/Bluetooth.odex:da52b4290679e2159c5ec6b6214c4c87
 bullhead/mtc20k/factory_imgs_data/system/app/BluetoothMidiService/BluetoothMidiService.apk:91073d348f48f85c3a307c8166afa8bf
+bullhead/mtc20k/factory_imgs_data/system/app/BluetoothMidiService/oat/arm64/BluetoothMidiService.odex:4530a1517c06b6a0023cbc6542adb629
+bullhead/mtc20k/factory_imgs_data/system/app/BullheadLayout/BullheadLayout.apk:972b77d8fabefa9aeb29986b0b7f0cd9
+bullhead/mtc20k/factory_imgs_data/system/app/BullheadLayout/oat/arm64/BullheadLayout.odex:877862149469b58a9fc93d20f663ad30
 bullhead/mtc20k/factory_imgs_data/system/app/CalculatorGoogle/CalculatorGoogle.apk:345130b0cb4d3d970bae2327cb92bd72
+bullhead/mtc20k/factory_imgs_data/system/app/CalculatorGoogle/oat/arm64/CalculatorGoogle.odex:a145a9a2d2a06ce150164dca0b8ff889
+bullhead/mtc20k/factory_imgs_data/system/app/CalendarGooglePrebuilt/CalendarGooglePrebuilt.apk:57c0f2b401d27b0ce8ee65e063ce67c8
+bullhead/mtc20k/factory_imgs_data/system/app/CalendarGooglePrebuilt/oat/arm64/CalendarGooglePrebuilt.odex:87f6974202332bcd1f7760f72d45a164
 bullhead/mtc20k/factory_imgs_data/system/app/CaptivePortalLogin/CaptivePortalLogin.apk:d813a4369791e8f7d2c70fc835251a9d
 bullhead/mtc20k/factory_imgs_data/system/app/CaptivePortalLogin/oat/arm64/CaptivePortalLogin.odex:2464de47c7d45f983b19c910f8e0330c
 bullhead/mtc20k/factory_imgs_data/system/app/CertInstaller/CertInstaller.apk:7aacd43b3d1aa685b6512ac9ccb0dd7b
@@ -22,6 +28,10 @@
 bullhead/mtc20k/factory_imgs_data/system/app/Drive/oat/arm64/Drive.odex:7284406140576278410fcb5ff386fffe
 bullhead/mtc20k/factory_imgs_data/system/app/FaceLock/FaceLock.apk:1b6ca424757c862222661a72df8a65b1
 bullhead/mtc20k/factory_imgs_data/system/app/FaceLock/oat/arm64/FaceLock.odex:d73f36cd52f87e086a5c207475dadad0
+bullhead/mtc20k/factory_imgs_data/system/app/GoogleCamera/GoogleCamera.apk:df17d60a00d9532b12327cbe5f8007b7
+bullhead/mtc20k/factory_imgs_data/system/app/GoogleCamera/oat/arm64/GoogleCamera.odex:52961de2ff049b6f523680fde62af739
+bullhead/mtc20k/factory_imgs_data/system/app/GoogleContactsSyncAdapter/GoogleContactsSyncAdapter.apk:b99ededa6897f8ebd54eee652be710b0
+bullhead/mtc20k/factory_imgs_data/system/app/GoogleContactsSyncAdapter/oat/arm64/GoogleContactsSyncAdapter.odex:53bcba9825a42fb1c938af62f70c1dfe
 bullhead/mtc20k/factory_imgs_data/system/app/GoogleHindiIME/GoogleHindiIME.apk:033ab854fe0c07a3fc2ea0d3bffc4c93
 bullhead/mtc20k/factory_imgs_data/system/app/GoogleHindiIME/oat/arm64/GoogleHindiIME.odex:609bd8e75f620ba301b649cbb35fe835
 bullhead/mtc20k/factory_imgs_data/system/app/GoogleHome/GoogleHome.apk:0b3a06e3e003bd140b0ec8525ad60c08
@@ -35,14 +45,29 @@
 bullhead/mtc20k/factory_imgs_data/system/app/HTMLViewer/HTMLViewer.apk:c54264b488461304e22349cb171c4f43
 bullhead/mtc20k/factory_imgs_data/system/app/HTMLViewer/oat/arm64/HTMLViewer.odex:83487ea0307c92c0683500b658b87706
 bullhead/mtc20k/factory_imgs_data/system/app/Hangouts/Hangouts.apk:0101359b4a10468bd8ce5e649b208389
+bullhead/mtc20k/factory_imgs_data/system/app/Hangouts/oat/arm/Hangouts.odex:6b77481a562b1575e4c5d4d6083d55a1
+bullhead/mtc20k/factory_imgs_data/system/app/HiddenMenu/HiddenMenu.apk:a50731a5ab478950a7d3156eb667204f
+bullhead/mtc20k/factory_imgs_data/system/app/HiddenMenu/oat/arm64/HiddenMenu.odex:e5fd1b68b799a360d42c382cc238ea7c
 bullhead/mtc20k/factory_imgs_data/system/app/KeyChain/KeyChain.apk:eaca5540bb6e2641bc84cad3645042e2
+bullhead/mtc20k/factory_imgs_data/system/app/KeyChain/oat/arm64/KeyChain.odex:1944e27cfa31628cc07258de2f96f585
+bullhead/mtc20k/factory_imgs_data/system/app/KoreanIME/KoreanIME.apk:3a5ec342bb3c67e3ccbe424063dd7cf8
+bullhead/mtc20k/factory_imgs_data/system/app/KoreanIME/oat/arm64/KoreanIME.odex:cdd61bb0198cd5da7ad23440ec09d8f1
 bullhead/mtc20k/factory_imgs_data/system/app/LatinImeGoogle/LatinImeGoogle.apk:1f9c02b03027b4341765bc8b4805730b
 bullhead/mtc20k/factory_imgs_data/system/app/LatinImeGoogle/oat/arm64/LatinImeGoogle.odex:d13a725625e62444fa9f15b66bccbfe1
 bullhead/mtc20k/factory_imgs_data/system/app/LiveWallpapersPicker/LiveWallpapersPicker.apk:d257afad6612d0b212f7d8e2214c2e5e
+bullhead/mtc20k/factory_imgs_data/system/app/LiveWallpapersPicker/oat/arm64/LiveWallpapersPicker.odex:1d1b27c77ec2153625e6b96bddd7e8fc
+bullhead/mtc20k/factory_imgs_data/system/app/Maps/Maps.apk:7af9c6808885afcbf9bf637ddc2b5b53
+bullhead/mtc20k/factory_imgs_data/system/app/Maps/oat/arm64/Maps.odex:e3fc7f5b67161a8959babd936a6a4198
+bullhead/mtc20k/factory_imgs_data/system/app/Music2/Music2.apk:2c27b51d325f700f5cdc70a29fcc719a
+bullhead/mtc20k/factory_imgs_data/system/app/Music2/oat/arm64/Music2.odex:93b9d3d6fdb4ad8899e1ace61cd042d2
 bullhead/mtc20k/factory_imgs_data/system/app/NfcNci/NfcNci.apk:07fbce22c357f7421d4a8ca07d439c4d
 bullhead/mtc20k/factory_imgs_data/system/app/NfcNci/oat/arm64/NfcNci.odex:df252f4a6e554b1eaf6c598181e6bd7c
[... cut ...]
+bullhead/mtc20k/factory_imgs_data/system/app/RcsImsBootstraputil/oat/arm64/RcsImsBootstraputil.odex:a1252868a5999b033ca424cd6a7edcfe
 bullhead/mtc20k/factory_imgs_data/system/app/SetupSmartDeviceOverlay/SetupSmartDeviceOverlay.apk:ab554c944e42d5aeec398254baaa078c
 bullhead/mtc20k/factory_imgs_data/system/app/SetupSmartDeviceOverlay/oat/arm64/SetupSmartDeviceOverlay.odex:3d8a5dfba122875acdf9ae0df2838adf
 bullhead/mtc20k/factory_imgs_data/system/app/Stk/Stk.apk:3d575c24a169f6b0d981cfe8b93eb681
```

anestisb, 2017-01-04:
Seems that my initial analysis was wrong. You're right that for some images the `PKGS_SSLINKS` array can be empty if all detected symbolic links are located under an apps directory. The bug has been introduced at commit https://github.com/anestisb/android-prepare-vendor/commit/b54fcbc4be5d05e0655a0f5e475685e6fe0b3340

Will do some more regression testing and apply your patch. Thanks again.

anestisb, 2017-01-04:
Fixed

closed 2017-01-04 by anestisb