anestisb/android-prepare-vendor #57: Add a flag to execute-all to keep apk signature

ezio84, 2017-02-07:
Hi guys, thanks for this awesome project. 
It'd be nice to have a flag for the execute-all.sh to keep the META-INF in the final deodexed apk, it's been useful at least for me today to quickly update angler vendor blobs with the latest February factory image.

Bye and thanks again.

anestisb, 2017-02-11:
I'm confused as to what you're trying to achieve here. What is the point of leaving an old signature inside the APK since it would be resigned?

Resigning is required since a new file (the deodex bytecode) is appended to the APK. Normally the build system is expected to replace the key + manifest files, I just force it manually as a defensive mechanism to ensure that old signatures don't propagate to repaired files.

ezio84, 2017-02-12:
The reason is that HotwordEnrollment.apk didn't work correctly after resigning it, Ok Google everywhere was not able to start.

anestisb, 2017-02-12:
Well, google apps support was never properly implemented. Are you repairing "HotwordEnrollment" from factory images using the gplay config profile from this script, or are you trying to install gapps afterwards?

Which device and build version are you using?

ezio84, 2017-02-12:
Yes, extracted and repaired HotwordEnrollment from latest Angler factory img (N4F26O) with the "gplay" flag.

The related .mk part:
include $(CLEAR_VARS)
LOCAL_MODULE := HotwordEnrollment
LOCAL_MODULE_TAGS := optional
LOCAL_SRC_FILES := HotwordEnrollment/HotwordEnrollment.apk
LOCAL_CERTIFICATE := PRESIGNED
LOCAL_MODULE_CLASS := APPS
LOCAL_PRIVILEGED_MODULE := true
LOCAL_MODULE_SUFFIX := $(COMMON_ANDROID_PACKAGE_SUFFIX)
include $(BUILD_PREBUILT)

Keeping the original META-INF for all vendor blobs apk, everything works well (also facelock and other things).

anestisb, 2017-05-13:
Do you still have this issue?

In a recent batch of changes I've enforced all APKs (repaired or not) to be resigned with the platform key. As such the generate scripts will no more set the `LOCAL_CERTIFICATE := PRESIGNED` for extracted APKs.

ezio84, 2017-05-17:
Ye it seems i still need this:
https://github.com/ezio84/vendor-blobs-extractfromfactory/commit/40f267c3512604abbb236d6df73a157fbcdd854e
and the presigned flag to keep the original META-INF folder in the extracted apk (e.g. HotwordEnrollment for Ok Google Everywhere).

anestisb, 2017-05-17:
I see. So HotwordEnrollment is required to keep the original Google signature (probably has signature protected interfaces utilized by other apps).

This effectively means that we cannot repair it (de-optimize) in any Nexus/Pixel configs. I'll remove it from the config. Users will have to install it as part of the GApps installation (which I think is already happening from the distributed installation packages).

Thanks for the info.

ezio84, 2017-05-17:
Afaik that apk has been removed lately from common gapps packages (beans and opengapps) because it's become device-specific, so it must be put in the priv-app folder by the rom flash zip.

Can't say more instead about other vendor apk because the hotword one is the one i can test straight: without the original meta-inf, the ok google everywhere option is greyed out. With the original one, i can enable it.

anestisb, 2017-05-17:
Ok I'll investigate further and see if I can hack around it. However, if the application is pre-optimized and the original bytecode is stripped from the APK there are not much I can do.

Have you obtained a good copy at some point and use across your angler builds or are you alwayer repairing it from the latest factory image?

ezio84, 2017-05-17:
I always take it from latest factory img.
Here's a copy of the actual working one, if it can be useful to you: https://github.com/ezio84/abc-n_vendor_huawei/tree/n2/angler/proprietary/priv-app/HotwordEnrollment
It's been extracted with the above commit to  keep the original meta-inf, with these flags:
./execute-all.sh -d angler -i "factoryimg_zip_path" --gplay -y -o "extracted_vendor_path" -b N2G47O -k --deodex-all

anestisb, 2017-05-17:
Thanks for the info. I'll have a deeper look into it.

Btw I've replaced the `--gplay` flag since it was misleading as to its purpose. The new flag is `--full` and aims to utilize a series of configuration settings that include non-essential vendor blobs that might be required from some setups (e.g. US carriers, GApps). My goal is to keep a very minimal default for the `naked` configuration profile that provide the smallest set of binaries required to deploy a functional AOSP build to Nexus/Pixel devices.

ezio84, 2017-05-17:
Ye i noticed that, good one. Tnx yo you dude!!

anestisb, 2017-05-17:
Have just checked bullhead, angler & sailfish, and it seems that the HotwordEnrollment APK is always pre-optimized. Bullhead & angler factory system images strip the `classes.dex` from the APK, while the Pixel one does not.

So this effectively means that for Pixel devices the APK can be included with the `LOCAL_CERTIFICATE := PRESIGNED`, although for bullhead & angler it cannot.

What you're effectively doing in your setup by keeping the old META-INF while still appending the repaired `classes.dex` in the APK zip archive, is that you have an unsigned entry. If you run jarsigner against the APK you've repaired you'll see that the `classes.dex` that is appended is not signed thus vulnerable to tamper attacks. As such it is not safe to use in production ROMs. If this setup is fine for your ROM use cases you can keep using it, although I cannot upstream since it introduces potential security issues to generated images. Also I'm very surprised that the Package & Asset Managers are not complaining with a runtime exception for the unsigned bytecode entry.

Now if the HotwordEnrollment APK is indeed device specific I don't think that the issue can be reliably solved for bullhead & angler devices. If you have any further ideas, they're more than welcome.

Also cc'ing @thestinger just to be sure that CopperheadOS is not using any blobs with PRESIGNED certificate and thus might be affected for potential unsigned entries.

ezio84, 2017-05-18:
Thanks for your explanation. Atm that's the only solution i found to make that apk working.

anestisb, 2017-05-18:
Yeah I don't think that this issue can be properly solved for angler/bullhead

anestisb, 2017-06-06:
I've pushed a workaround for Pixel & Pixel XL, although as mentioned above there nothing I can do from my side to provide a complete fix for angler & bullhead.

I'm keeping this issue OPEN for the time being for quick reference purposes.

robbyoconnor, 2017-07-24:
@anestisb what about excluding that APK from deodexing?

anestisb, 2017-07-24:
If the DEX bytecode is not repaired from the optimized OAT file, it cannot be imported as an APK to the AOSP build system since the bytecode file is missing.

Technically the factory image OAT files have all the information required from patchoat to relocate/repair offsets based on device's ART images, although the AOSP build system does not appear to support OAT precompiled imports that can be patched instead of being compiled from scratch.

So in short note I don't think that there is a way to work around the issue with this approach, if I'm not missing anything.

robbyoconnor, 2017-07-24:
hmm I guess -- I could go the route of using an existing vendor tree but it won't be up to date at all...and that's not great at all. Extracting them from the latest security fixes is the ideal way..

anestisb, 2017-07-24:
Yeah I know it's really frustrating considering that the updates of the "com.android.hotwordenrollment" are no longer available in the Play Store due to the device specific bits.

So I guess the only workaround is repairing but not signing the bytecode entry, as described in previous comment. However, I cannot upstream this due to security reasons.

robbyoconnor, 2017-07-24:
I actually did that and left the signatures in place. Just don't remove the signatures I guess?

anestisb, 2017-09-12:
Seems that the same problem is present in API-26 for N5x & N6p devices. New apk is labeled as `system/priv-app/HotwordEnrollmentWCD9330/HotwordEnrollmentWCD9330.apk` and still has the `classes.dex` stripped.

anestisb, 2017-11-30:
Renamed issue since it will be open forever for reference purposes.

anestisb, 2017-12-07:
For the record the problem affects also API-27 (Google keeps stripping `classes.dex` from APKs).

```
$ find opm1.171019.011/factory_imgs_data -iname "HotwordEnrollment*.apk" | while read -r file; do zipinfo $file; done
Archive:  opm1.171019.011/factory_imgs_data/system/priv-app/HotwordEnrollmentTGoogleWCD9330/HotwordEnrollmentTGoogleWCD9330.apk
Zip file size: 35975 bytes, number of entries: 10
-rw----     1.0 fat    13379 bx stor 09-Jan-01 00:00 res/drawable-xxxhdpi-v4/ic_app.png
-rw----     1.0 fat       10 bx stor 09-Jan-01 00:00 res/raw/version
-rw----     1.0 fat     2300 bx stor 09-Jan-01 00:00 resources.arsc
-rw----     2.0 fat     3072 bl defN 09-Jan-01 00:00 AndroidManifest.xml
-rw----     2.0 fat      632 bl defN 09-Jan-01 00:00 res/layout/main.xml
-rw----     2.0 fat    30712 bl defN 09-Jan-01 00:00 res/raw/en_all.mmap
-rw----     2.0 fat      540 bl defN 09-Jan-01 00:00 res/xml/enrollment_application.xml
-rw----     2.0 fat      815 bl defN 09-Jan-01 00:00 META-INF/CERT.SF
-rw----     2.0 fat     1452 bl defN 09-Jan-01 00:00 META-INF/CERT.RSA
-rw----     2.0 fat      735 bl defN 09-Jan-01 00:00 META-INF/MANIFEST.MF
10 files, 53647 bytes uncompressed, 32970 bytes compressed:  38.5%
Archive:  opm1.171019.011/factory_imgs_data/system/priv-app/HotwordEnrollmentXGoogleWCD9330/HotwordEnrollmentXGoogleWCD9330.apk
Zip file size: 35940 bytes, number of entries: 10
-rw----     1.0 fat    13379 bx stor 09-Jan-01 00:00 res/drawable-xxxhdpi-v4/ic_app.png
-rw----     1.0 fat       10 bx stor 09-Jan-01 00:00 res/raw/version
-rw----     1.0 fat     2300 bx stor 09-Jan-01 00:00 resources.arsc
-rw----     2.0 fat     3072 bl defN 09-Jan-01 00:00 AndroidManifest.xml
-rw----     2.0 fat      632 bl defN 09-Jan-01 00:00 res/layout/main.xml
-rw----     2.0 fat    30516 bl defN 09-Jan-01 00:00 res/raw/en_all.mmap
-rw----     2.0 fat      540 bl defN 09-Jan-01 00:00 res/xml/enrollment_application.xml
-rw----     2.0 fat      815 bl defN 09-Jan-01 00:00 META-INF/CERT.SF
-rw----     2.0 fat     1452 bl defN 09-Jan-01 00:00 META-INF/CERT.RSA
-rw----     2.0 fat      735 bl defN 09-Jan-01 00:00 META-INF/MANIFEST.MF
10 files, 53451 bytes uncompressed, 32935 bytes compressed:  38.4%
Archive:  opm1.171019.011/factory_imgs_data/system/priv-app/HotwordEnrollmentOKGoogleWCD9330/HotwordEnrollmentOKGoogleWCD9330.apk
Zip file size: 537410 bytes, number of entries: 48
-rw----     1.0 fat    13379 bx stor 09-Jan-01 00:00 res/drawable-xxxhdpi-v4/ic_app.png
-rw----     1.0 fat       10 bx stor 09-Jan-01 00:00 res/raw/version
-rw----     1.0 fat     4504 bx stor 09-Jan-01 00:00 resources.arsc
-rw----     2.0 fat     3072 bl defN 09-Jan-01 00:00 AndroidManifest.xml
-rw----     2.0 fat      632 bl defN 09-Jan-01 00:00 res/layout/main.xml
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/ar_eg.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/cmn_hans_cn.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/cmn_hant_tw.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/cs_cz.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/da_dk.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/de_de.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/en_au.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/en_gb.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/en_ie.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/en_in.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/en_ph.mmap
-rw----     2.0 fat    34084 bl defN 09-Jan-01 00:00 res/raw/en_us.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/es_419.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/es_ar.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/es_es.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/es_mx.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/es_us.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/fa_ir.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/fi_fi.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/fil_ph.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/fr_fr.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/hi_in.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/hr_hr.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/id_id.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/it_it.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/ja_jp.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/ko_kr.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/ms_my.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/nb_no.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/nl_nl.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/pl_pl.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/pt_br.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/ro_ro.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/ru_ru.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/sv_se.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/th_th.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/tr_tr.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/vi_vn.mmap
-rw----     2.0 fat    33796 bl defN 09-Jan-01 00:00 res/raw/yue_hant_hk.mmap
-rw----     2.0 fat      768 bl defN 09-Jan-01 00:00 res/xml/enrollment_application.xml
-rw----     2.0 fat     4254 bl defN 09-Jan-01 00:00 META-INF/CERT.SF
-rw----     2.0 fat     1452 bl defN 09-Jan-01 00:00 META-INF/CERT.RSA
-rw----     2.0 fat     4174 bl defN 09-Jan-01 00:00 META-INF/MANIFEST.MF
48 files, 1350577 bytes uncompressed, 529503 bytes compressed:  60.8%
```

closed 2018-01-06 by anestisb