anestisb/android-prepare-vendor #58: marlin/sailfish flashlight quick settings icon greyed out after first boot of an update

thestinger, 2017-02-11:
I am pretty sure this is caused by missing something in AOSP / android-prepare-vendor. It might have another local cause though, so it would be good if someone could confirm that they hit this.

anestisb, 2017-02-11:
Issue has been confirmed. After a hardware reboot, the flashlight icon grays-out and cannot be enabled.

I'll investigate the issue.

thestinger, 2017-02-11:
It's quite weird since the camera and camera flash still work.

anestisb, 2017-02-12:
After a factory reset the flashlight is working fine until the next reboot. It seems that there is some kind of post-boot configuration that conflicts or destroys some assets access of the CameraService. I'm still investigating, although evidence so far indicate that is not a missing blob. Seems some kind of configuration conflict (probably against data stored under /data).

mke208, 2017-02-12:
Just noticed something regarding this issue: if you remove the flashlight widget and then re-add it, it starts working ...  Device is marlin.

anestisb, 2017-02-12:
It seems some kind of race condition, since `CameraManagerGlobal` service is requesting resources from `CameraService` service before the latter has finished initialization. It seems to try two times (probably one for each cameraID - front and back) and then abort, leaving some runtime objects null.

```
05-22 11:59:53.749  1396  1396 I CameraManagerGlobal: Connecting to camera service
05-22 11:59:53.749  1396  1396 E CameraManagerGlobal: Camera service is unavailable
05-22 11:59:53.780  1396  1396 I CameraManagerGlobal: Connecting to camera service
05-22 11:59:53.780  1396  1396 E CameraManagerGlobal: Camera service is unavailable
```

As such I don't think the issue is related to the vendor blobs.

anestisb, 2017-02-12:
Attaching camera logs vs factory

[aosp_camera_log.txt](https://github.com/anestisb/android-prepare-vendor/files/769506/aosp_camera_log.txt)
[factory_camera_log.txt](https://github.com/anestisb/android-prepare-vendor/files/769507/factory_camera_log.txt)

It's quite clear that `CameraManagerGlobal` is not blocking until `CameraService` has completed initialization. Issue should be reported to upstream as it doesn't appear to be connected to vendor blobs.

thestinger, 2017-02-12:
Does this happen with stock though? Maybe they aren't using the AOSP code...

anestisb, 2017-02-12:
The `CameraManagerGlobal` code from AOSP appears to be identical between AOSP and factory images. However, I don't know which component or configuration value is triggering the problem in AOSP builds but not factory. It might be related to the actual camera app and the native libraries it deploys, although it's hard to confirm since the app implements lots of functionality.

thestinger, 2017-02-22:
Removing the flashlight quick tile and adding it back works around it. It's weird, because it seems to try to register a callback to be notified when this service is available: https://developer.android.com/reference/android/hardware/camera2/CameraManager.html#registerTorchCallback(android.hardware.camera2.CameraManager.TorchCallback, android.os.Handler). It seems the service itself is somehow unavailable...

thestinger, 2017-02-22:
Alright I did a partial workaround for this so that it at least starts working once it's touched (assuming camera service is loaded by then): https://github.com/CopperheadOS/platform_frameworks_base/commit/79ea29c5205ed6a2a923a8a4f6aaf7f935274724. Before then it's greyed out. It really needs to wait until the service is ready, but I'm not sure how to do that at the moment if it's even possible without deeper surgery.

thestinger, 2017-02-25:
I think one of our users hit this on the Nexus 5X: https://github.com/copperhead/bugtracker/issues/597#issuecomment-282476247. I'm glad to have added that recovery hack because it adds some robustness against this kind of failure. I'd like to solve it properly and upstream that though...

vamsirocks, 2017-03-31:
Hi i need help how to build pixel aosp without including recovery partition, such that device cant enter recovery mode

thestinger, 2017-03-31:
I don't think that has anything to do with this...

anestisb, 2017-04-01:
@vamsirocks please open a different issue since what you're asking is not related to this thread.

anestisb, 2017-05-13:
I think that the issue has been fixed in upstream AOSP. Just tested a NHG47L android-7.1.2_r11 sailfish build and the flashlight appears to work fine across device reboots.

closed 2017-05-18 by anestisb