anestisb/android-prepare-vendor #59: [Pixel] netmgrd crash loop

anestisb, 2017-02-11:
`/system/bin/netmgrd` enters a crash loop polluting the logs with the following. Issue is under investigation

```
02-11 23:10:13.187  4236  4236 W netmgrd : type=1400 audit(0.0:524): avc: denied { dac_override } for capability=1 scontext=u:r:netmgrd:s0 tcontext=u:r:netmgrd:s0 tclass=capability permissive=0
02-11 23:10:13.188  4236  4236 I libmdmdetect: No supported ESOC's found
02-11 23:10:13.189  4236  4236 I libmdmdetect: slpi subsystem located
02-11 23:10:13.190  4236  4236 I libmdmdetect: Found internal modem: modem
02-11 23:10:13.190  4236  4236 I libmdmdetect: modem subsystem found
02-11 23:10:13.193  4236  4236 E Diag_Lib:  Diag_LSM_Init: Failed to open handle to diag driver, error = 13
02-11 23:10:13.194  4236  4236 I libmdmdetect: No supported ESOC's found
02-11 23:10:13.195  4236  4236 I libmdmdetect: slpi subsystem located
02-11 23:10:13.195  4236  4236 I libmdmdetect: Found internal modem: modem
02-11 23:10:13.195  4236  4236 I libmdmdetect: modem subsystem found
02-11 23:10:13.196  4236  4236 E QC-NETMGR-LIB: QC-NETMGR
```

thestinger, 2017-02-11:
FWIW, I've also been running into this for some time now. I assumed it was a CopperheadOS feature causing it but I guess it's not after all.

anestisb, 2017-02-11:
Yeah a default AOSP src is triggering the same issue. Seems that some dlopen is failing. I've fired a debug build to strace the problem.

anestisb, 2017-02-12:
The https://github.com/anestisb/android-prepare-vendor/commit/e5e30384d9bb459c3005fba60484a29a7a7b554b commit appears to fixes the issue. Will do some additional testing later today against a release ROM build, since at the moment I'm chasing the remaining issues against a debug build.

anestisb, 2017-02-12:
Fix verified with release builds.

closed 2017-02-12 by anestisb