anestisb/android-prepare-vendor #60: Idea: replace fuse-ext2 with debugfs

relan, 2017-02-12:
debugfs(8) is an utility from official e2fsprogs, so it's very well tested and maintained. It can recursively extract files from ext2/ext3/ext4 images:

```sh
# $1 is ext2/ext3/ext4 image file name
debugfs -R 'ls -p' $1 | cut -d/ -f6 | while read entry; do
    debugfs -R "rdump \"$entry\" ." $1
done
```

It cannot extract all top-level directories and files in a single command, so we have to list top-level entries first (`ls -p`) and then extract each of them (`rdump $entry .` — works recursively).

fuse-ext2 fails even to list `vendor.img` from volantis' N4F26Q for me. Given how many complains here in the issues are related to fuse-ext2 and FUSE, I think it's worth switching to debugfs. This will also simplify `extract-factory-images.sh` a lot.

e2fsprogs are available in Homebrew, so Mac users should be fine.

anestisb, 2017-02-12:
Which platform (macOS or Linux) fails to extract the `vendor.img` contents from volantis N4F26Q image?

Indeed `fuse-ext2` is not the most robust option, although I've never noticed an issue under Linux workstations, except some config glitches that are enumerated in the README file. Do you have issues with fuse-ext2 when working under a Linux workstation?

For macOS indeed the tool has some functional issues like the https://github.com/anestisb/android-prepare-vendor/issues/53. I report them as soon as I notice.

I'll investigate the option you mentioned and see if I can add some experimental initial support. Of course PR are more than welcome.

relan, 2017-02-12:
> Which platform (macOS or Linux) fails to extract the vendor.img contents from volantis N4F26Q image?

Linux (CentOS 7 x86-64), fuse-ext2 v0.0.9. It did mount the image but its contents was empty. The patch from #53 didn't help. I gave up and dirty-hacked `extract-factory-images.sh` and `execute-all.sh` to use debugfs. This solved my issue.

> Do you have issues with fuse-ext2 when working under a Linux workstation?

Never tried fuse-ext2 for anything else. All my experience with `android-prepare-vendor` happened on a CentOS 7 system.

anestisb, 2017-02-12:
Have added some initial experimental support for debugfs. You can test it invoking the master script with the `--debugfs` flag. Ensure you've pulled the latest master.

Let me know if that worked for your setup since I don't have a CentOS vm available at the moment to test.

relan, 2017-02-13:
With `--debugfs` it worked flawlessly on my CentOS 7.3. Thanks for a quick fix!

closed 2017-02-13 by relan