anestisb/android-prepare-vendor #61: [Pixel] Services not spawned

anestisb, 2017-02-12:
The following services are not spawned despite the fact that vendor binaries are included:
* AtCmdFwd:[com.qualcomm.atfwd.IAtCmdFwd]
* cneservice:[com.quicinc.cne.ICNEManager]
* ims:[com.android.ims.internal.IImsService]
* rcs:[com.android.ims.internal.IRcsService]

```bash
--- aosp_svc_dump.txt	2017-02-12 13:15:56.000000000 +0200
+++ orig_svc_dump.txt	2017-02-12 13:26:53.000000000 +0200
@@ -1,3 +1,4 @@
+AtCmdFwd:[com.qualcomm.atfwd.IAtCmdFwd]
 DockObserver:[]
 SurfaceFlinger:[android.ui.ISurfaceComposer]
 accessibility:[android.view.accessibility.IAccessibilityManager]
@@ -19,6 +20,7 @@
 bluetooth_manager:[android.bluetooth.IBluetoothManager]
 carrier_config:[com.android.internal.telephony.ICarrierConfigLoader]
 clipboard:[android.content.IClipboard]
+cneservice:[com.quicinc.cne.ICNEManager]
 commontime_management:[]
 connectivity:[android.net.IConnectivityManager]
 connectivity_metrics_logger:[android.net.IConnectivityMetricsLogger]
@@ -46,6 +48,7 @@
 graphicsstats:[android.view.IGraphicsStats]
 hardware_properties:[android.os.IHardwarePropertiesManager]
 imms:[com.android.internal.telephony.IMms]
+ims:[com.android.ims.internal.IImsService]
 input:[android.hardware.input.IInputManager]
 input_method:[com.android.internal.view.IInputMethodManager]
 iphonesubinfo:[com.android.internal.telephony.IPhoneSubInfo]
@@ -90,6 +93,7 @@
 print:[android.print.IPrintManager]
 processinfo:[android.os.IProcessInfoService]
 procstats:[com.android.internal.app.procstats.IProcessStats]
+rcs:[com.android.ims.internal.IRcsService]
 recovery:[android.os.IRecoverySystem]
 restrictions:[android.content.IRestrictionsManager]
 rttmanager:[android.net.wifi.IRttManager]
```

mke208, 2017-02-12:
Hi,
might it be related to this ? 
https://forum.xda-developers.com/general/help/mobile-data-aosp-pixel-please-help-t3533304

strace on netmgrd shows this:
openat(AT_FDCWD, "/system/etc/data/netmgr_config.xml", O_RDONLY) = -1 ENOENT (No such file or directory)

anestisb, 2017-02-12:
As of commit https://github.com/anestisb/android-prepare-vendor/commit/e5e30384d9bb459c3005fba60484a29a7a7b554b `/system/etc/data/netmgr_config.xml` is included in the blobs list, so it's not this one that caused the problem.

It appears to be some missing bytecode files. I have already fixed (push is about to happen soon) the first two (AtCmdFwd and cneservice) and I'm chasing now the other two.

anestisb, 2017-02-12:
`ims:[com.android.ims.internal.IImsService]` is the only still broken and is under investigation. The code is present, although for some reason the service doesn't start.

Not to mention the next level of pain that we're facing here with google partially pushing vendor code in AOSP. While some code is now present in AOSP, the original modules that included them as deps are not present in the makefile structure, resulting into skipping them. So now we not only need to identify the missing blobs, but also the broken dependencies and include them.

mke208, 2017-02-12:
The issue with missing /system/etc/data is also present on Pixel XL ...

anestisb, 2017-02-12:
@mke208 true. I'll port the changes from sailfish to marlin as soon as I finalize them. Since I don't have a Pixel XL device to test, it's hard for me to verify that fixes are working there as well. If you have one available, it would be really helpful to give it a spin and verify them.

mke208, 2017-02-12:
I have added the missing files and i am building now. Ill get back with an update.

anestisb, 2017-02-12:
Final service fixed from https://github.com/anestisb/android-prepare-vendor/commit/48b5a67ddd5442be1e910f137ad38b039e0edd66

Issue should be fixed across the border now for both Pixel & XL

closed 2017-02-12 by anestisb

mke208, 2017-02-12:
Tested this, everything seems fine now.

anestisb, 2017-02-12:
Awesome, thanks @mke208

mke208, 2017-02-12:
Will keep looking around and do more testing. 
Btw, I appreciate the work you are doing.
Thanks

commit 992e31a5 (2017-02-12): sailfish config updates

Partially fixes #61