anestisb/android-prepare-vendor #66: Bootloader and baseband version check in android-info.txt

Julien-Roger, 2017-06-02:
First, thanks for this great project, that saved hours for me. We had the "NO SIM CARD"/Baseband unknown on Pixel device without understanding why. Now, with the generated vendor folder, it is OK.

However I have noticed that when using the generated vendor folder for marlin or sailfish (pixel xl/pixel), generated android-info.txt file contains 2 lines:
require version-bootloader=8996-012001-1702151126
require version-baseband=8996-012901-1702171013

When using fastboot flashall for updating the device, we get an error because the device bootloader version is not matching the required value in the android-info.txt file.
This could be resolved easily by removing those 2 lines in this file, but it would be better this lines not to be generated at AOSP build time.

Thanks

thestinger, 2017-06-02:
> When using fastboot flashall for updating the device, we get an error because the device bootloader version is not matching the required value in the android-info.txt file.

That's not the process for flashing everything. The standard process for a complete flash is generating the factory images and using `flash-all.sh`. If you aren't doing it via the proper process, then you should at least be flashing the bootloader and radio firmware manually.

> This could be resolved easily by removing those 2 lines in this file, but it would be better this lines not to be generated at AOSP build time.

They should not be removed. It's correct to have them, and you should just use the correct flashing process. If you want full documentation in making signed, production releases and generating factory images and over-the-air updates, https://copperhead.co/android/docs/building covers everything. The only thing it doesn't cover is android-prepare-vendor since that's already done for people building CopperheadOS and signed incremental over-the-air updates which are straightforward to generate in the same way.

Julien-Roger, 2017-06-02:
I am just following the process described by Google in https://source.android.com/source/running

$ fastboot flashall -w

thestinger, 2017-06-02:
Their documentation is very incomplete. That isn't an android-prepare-vendor issue. There's no Google documentation covering the full set of steps to build a signed, production release. They don't have documentation on including all of the vendor components either so it's not unexpected that their flashing instructions are incomplete. I pointed you to complete instructions.

Julien-Roger, 2017-06-02:
OK thanks will read this carefully

thestinger, 2017-06-02:
If you want to avoid your immediate issue, flash the radio and bootloader images first. However, if you're doing something aimed at production usage or even just running the CTS in a meaningful way / testing against a production build you'll need to go through the extra steps to do a signed build which for Pixels involves the extra totally undocumented step of generating a key in the right format for the kernel keyring and including it there.

anestisb, 2017-06-04:
The `require` version tags have been placed to ensure that latest vendor blobs and builds are always flashed with the matching latest radio & bootloader images. It is very dangerous to mix setups and flash newer vendor & system images while using older bootloader and radio version.

@thestinger provided a link to detailed instructions on how to create fully signed images and automate flash/OTA process.

As an alternative in case you don't want signed images you can try any of the following:
* Manually flash bootloader and radio before executing the `fastboot flashall -w` command from the output directory
   ```
   fastboot flash bootloader bootloader.img
   fastboot reboot-bootloader
   fastboot flash radio radio.img
   fastboot reboot-bootloader
   fastboot flashall -w
   ```

* Execute `make dist` (distribution build) from AOSP root and flash the bootloader, radio & packed device images archive in similar manner to `flash-all.sh`. Be aware that distribution output is stored under `out/<aosp_src>/dist` and not `out/<aosp_src>/target`

Also it is strongly recommended that you use the fastboot host binary that is build along with the matching target from AOSP. This ensures that you don't have any compatibility issues when using fastboot command from Android SDK installed system-wide.

thestinger, 2017-06-04:
Those require version tags have actually been really helpful for us because they solve the issue of people doing partial flashes. AOSP already takes care of detecting a mismatch between vendor and system.

closed 2017-06-13 by anestisb

thestinger, 2017-06-16:
It turns out that AOSP actually knows how to use that bootloader / radio metadata to check at runtime like the system/vendor comparison, but they disabled it. There's also a disable boot image check.

I ended up enabling them for us, although their radio check doesn't work for the early boot check since the radio usually isn't up by then, which might be why they disabled this.

https://github.com/CopperheadOS/platform_frameworks_base/commit/9177098395c068676b8a1f4ea8f3cd2bd4236093