anestisb/android-prepare-vendor #67: system-proprietary-blobs file not found

Mawth, 2017-06-10:
With the latest bullhead factory image I get the following error when attempting to execute the scripts:

`./execute-all.sh -d bullhead -i ~/aosp7bullhead/aosp_b_prod/prepare_vendor_blobs/bullhead/n2g47w/bullhead-n2g47w-factory-e4179e51.zip -b n2g47w -o ~/aosp7bullhead/aosp_b_prod/prepare_vendor_blobs
[*] Setting output base to '/home/nighthawk/aosp7bullhead/aosp_b_prod/prepare_vendor_blobs/bullhead/n2g47w'
[*] Extracting '/home/nighthawk/aosp7bullhead/aosp_b_prod/prepare_vendor_blobs/bullhead/n2g47w/bullhead-n2g47w-factory-e4179e51.zip'
[*] Unzipping 'image-bullhead-n2g47w.zip'
[*] Processing with 'API-25 config-naked' configuration
[-] system-proprietary-blobs file not found
  Usage: gen-prop-blobs-list.sh [options]
    OPTIONS:
      -i|--input      : Root path of /vendor partition
      -o|--output     : Path to save generated "proprietary-blobs.txt" file
      --sys-list      : File list with proprietary blobs in /system partition
      --bytecode-list : File list with proprietary bytecode archive files
      --dep-dso-list  : File list with DSO files with individual targets
[-] 'proprietary-blobs.txt' generation failed`

This error would seem to be an issue starting with the June update, since I am following the same process as previous images that have executed successfully.

anestisb, 2017-06-11:
Just tested against bullhead n2g47w and it appears to work fine in my devices. So I'm guessing that there is some path issues in your setup that are not handled correctly from the scripts.

Can you please enable the debug mode in the `gen-prop-blobs-list.sh` script (uncomment line 12 in the file) and paste the verbose output here?

What is the base directory that you're running the scripts from?

Mawth, 2017-06-11:
[*] Setting output base to '/home/nighthawk/aosp7bullhead/aosp_b_prod/prepare_vendor_blobs/bullhead/n2g47w'
[*] Extracting '/home/nighthawk/aosp7bullhead/aosp_b_prod/prepare_vendor_blobs/bullhead/n2g47w/bullhead-n2g47w-factory-e4179e51.zip'
[*] Unzipping 'image-bullhead-n2g47w.zip'
[*] Processing with 'API-25 config-naked' configuration
+ sysTools=("find" "sed" "sort")
+ declare -a sysTools
+ trap 'abort 1' SIGINT SIGTERM
+ for i in '"${sysTools[@]}"'
+ command_exists find
+ type find
+ for i in '"${sysTools[@]}"'
+ command_exists sed
+ type sed
+ for i in '"${sysTools[@]}"'
+ command_exists sort
+ type sort
+ INPUT_DIR=
+ OUTPUT_DIR=
+ IN_SYS_FILE=
+ IN_BYTECODE_FILE=
+ IN_DEP_DSO_FILE=
+ [[ 10 -gt 1 ]]
+ arg=--input
+ case $arg in
++ echo /home/nighthawk/aosp7bullhead/aosp_b_prod/prepare_vendor_blobs/bullhead/n2g47w/factory_imgs_data/vendor
++ sed 's:/*$::'
+ INPUT_DIR=/home/nighthawk/aosp7bullhead/aosp_b_prod/prepare_vendor_blobs/bullhead/n2g47w/factory_imgs_data/vendor
+ shift
+ shift
+ [[ 8 -gt 1 ]]
+ arg=--output
+ case $arg in
++ echo /home/nighthawk/aosp7bullhead/aosp_b_prod/bullhead
++ sed 's:/*$::'
+ OUTPUT_DIR=/home/nighthawk/aosp7bullhead/aosp_b_prod/bullhead
+ shift
+ shift
+ [[ 6 -gt 1 ]]
+ arg=--sys-list
+ case $arg in
+ IN_SYS_FILE=/home/nighthawk/aosp7bullhead/aosp_b_prod/bullhead/config-naked/system-proprietary-blobs-api25.txt
+ shift
+ shift
+ [[ 4 -gt 1 ]]
+ arg=--bytecode-list
+ case $arg in
+ IN_BYTECODE_FILE=/home/nighthawk/aosp7bullhead/aosp_b_prod/bullhead/config-naked/bytecode-proprietary-api25.txt
+ shift
+ shift
+ [[ 2 -gt 1 ]]
+ arg=--dep-dso-list
+ case $arg in
+ IN_DEP_DSO_FILE=/home/nighthawk/aosp7bullhead/aosp_b_prod/bullhead/config-naked/dep-dso-proprietary-blobs-api25.txt
+ shift
+ shift
+ [[ 0 -gt 1 ]]
+ check_dir /home/nighthawk/aosp7bullhead/aosp_b_prod/prepare_vendor_blobs/bullhead/n2g47w/factory_imgs_data/vendor Input
+ local dirPath=/home/nighthawk/aosp7bullhead/aosp_b_prod/prepare_vendor_blobs/bullhead/n2g47w/factory_imgs_data/vendor
+ local dirDesc=Input
+ [[ /home/nighthawk/aosp7bullhead/aosp_b_prod/prepare_vendor_blobs/bullhead/n2g47w/factory_imgs_data/vendor == '' ]]
+ [[ ! -d /home/nighthawk/aosp7bullhead/aosp_b_prod/prepare_vendor_blobs/bullhead/n2g47w/factory_imgs_data/vendor ]]
+ check_dir /home/nighthawk/aosp7bullhead/aosp_b_prod/bullhead Output
+ local dirPath=/home/nighthawk/aosp7bullhead/aosp_b_prod/bullhead
+ local dirDesc=Output
+ [[ /home/nighthawk/aosp7bullhead/aosp_b_prod/bullhead == '' ]]
+ [[ ! -d /home/nighthawk/aosp7bullhead/aosp_b_prod/bullhead ]]
+ check_file /home/nighthawk/aosp7bullhead/aosp_b_prod/bullhead/config-naked/system-proprietary-blobs-api25.txt system-proprietary-blobs
+ local filePath=/home/nighthawk/aosp7bullhead/aosp_b_prod/bullhead/config-naked/system-proprietary-blobs-api25.txt
+ local fileDesc=system-proprietary-blobs
+ [[ /home/nighthawk/aosp7bullhead/aosp_b_prod/bullhead/config-naked/system-proprietary-blobs-api25.txt == '' ]]
+ [[ ! -f /home/nighthawk/aosp7bullhead/aosp_b_prod/bullhead/config-naked/system-proprietary-blobs-api25.txt ]]
+ echo '[-] system-proprietary-blobs file not found'
[-] system-proprietary-blobs file not found
+ usage
+ cat
++ basename /home/nighthawk/aosp7bullhead/aosp_b_prod/scripts/gen-prop-blobs-list.sh
  Usage: gen-prop-blobs-list.sh [options]
    OPTIONS:
      -i|--input      : Root path of /vendor partition
      -o|--output     : Path to save generated "proprietary-blobs.txt" file
      --sys-list      : File list with proprietary blobs in /system partition
      --bytecode-list : File list with proprietary bytecode archive files
      --dep-dso-list  : File list with DSO files with individual targets
+ abort 1
+ exit 1
[-] 'proprietary-blobs.txt' generation failed

Mawth, 2017-06-11:
I removed the entire android-prepare-vendor folder from my directory and re-synced. Still no luck. Output can be seen [here](https://pastebin.com/qDx4eXBE).

anestisb, 2017-06-12:
Aha, so after you fresh downloaded the project you have came across a different issue than the one originally stated in your logs. The second issue matches https://github.com/anestisb/android-prepare-vendor/issues/68, which has been resolved. 

So it seems that the original missing file issue was due to a corrupted local repository in your setup.

Can you please git pull from latest master and try again?

Mawth, 2017-06-13:
You are correct. Problem solved with latest master. Thank you for your help.

closed 2017-06-13 by Mawth