anestisb/android-prepare-vendor #69: Radio and bootloader updates via OTA

mke208, 2017-06-14:
Hello,
Can anyone check if bootloader and baseband also get updated via the OTA zip file for Marlin / Sailfish AB updates ? 
For my AOSP builds i needed to use a 'workaround':
-manually extract partition images from bootloader.img and radio.img
-add them to device/google/marlin/device-common.mk as follows:
AB_OTA_PARTITIONS += \
    boot \
    system \
    aboot \
    apdp \
    bootlocker \
    cmnlib32 \
    cmnlib64 \
    devcfg \
    hosd \
    hyp \
    keymaster \
    modem \
    pmic \
    rpm \
    tz \
    xbl
without this, the radio/bootloader partitions are not added to AB payload,
the factory image however works just fine.
Can someone else check with an AOSP build or maybe copperhead,
to see if this is really a bug, or i messed up my AOSP  tree ? 

Thanks

thestinger, 2017-06-14:
I think you're correct, but how did you get that list? It's missing sbl1 and perhaps other images isn't it?

thestinger, 2017-06-14:
From gpt-utils.h:

#define PTN_XBL                         "xbl"
#define PTN_SWAP_LIST                   PTN_XBL, "sbl1", "rpm", "tz", "aboot", "hyp", "lksecapp", "keymaster", "cmnlib", "cmnlib32", "cmnlib64", "pmic", "apdp", "devcfg", "hosd", "keystore", "msadp"
#define AB_PTN_LIST PTN_SWAP_LIST, "boot", "system", "vendor", "modem", "bluetooth"

thestinger, 2017-06-14:
Are you sure you aren't just supposed to use `AB_OTA_PARTITIONS += bootloader modem vendor`?

thestinger, 2017-06-14:
Did you get the list from `fastboot getvar all`? I'm not entirely sure if I'm being using the correct full list...

mke208, 2017-06-14:
hi,
i used an old unpacker, just for testing:
https://github.com/NVISO-BE/nexus_5_bootloader_unpacker
it is 2 years old, but it did extract some images ... maybe there is a way
to extract the images from google's payload.bin ? 
right now my marlin device works with no issues, but still, there are some missing 
images ...

thestinger, 2017-06-14:
FWIW, I'd greatly appreciate help on this... I haven't set up a proper way of handling it for CopperheadOS. It's one of many things that is not going well right now.

mke208, 2017-06-14:
by the way, as i remember, there is no sbl1_a / sbl1_b partition , nor a bluetooth partition ... however i am missing msadp... 
however the phone appears to be working correctly, bootloader and radio shows the latest version, the only visible difference right now is that on the Google boot logo i also have the "powered by android" logo which wasn't there before... 

from my testing, if you add an image that has no /dev/block/bootdevice/by-name/*_a or _b correspondent, updater will fail (from within the OS or recovery).

maybe the updater might have some code for extracting the correct images ?

thestinger, 2017-06-14:
Can you explain exactly what you're doing right now? Where are you extracting the images and how are you referencing them in the build system?

thestinger, 2017-06-14:
It's device/google/marlin/releasetools.py that's supposed to deal with this but clearly isn't right now.

mke208, 2017-06-14:
i compiled the bootloader_unpacker.c from the old nexus_5_bootloader_unpacker repo,
- ran ./bump radio.img and ./bump bootloader.img
- moved resulting *.img in vendor/google-devices/marlin/radio/
- added them to vendor/google-devices/marlin/AndroidBoardVendor.mk as follows:

$(call add-radio-file,radio/bootloader.img,version-bootloader)
$(call add-radio-file,radio/radio.img,version-baseband)
$(call add-radio-file,radio/aboot.img)
$(call add-radio-file,radio/apdp.img)
$(call add-radio-file,radio/bootlocker.img)
$(call add-radio-file,radio/cmnlib32.img)
$(call add-radio-file,radio/cmnlib64.img)
$(call add-radio-file,radio/devcfg.img)
$(call add-radio-file,radio/hosd.img)
$(call add-radio-file,radio/hyp.img)
$(call add-radio-file,radio/keymaster.img)
$(call add-radio-file,radio/modem.img)
$(call add-radio-file,radio/pmic.img)
$(call add-radio-file,radio/rpm.img)
$(call add-radio-file,radio/tz.img)
$(call add-radio-file,radio/xbl.img)

-added them to device/google/marlin/device-common.mk as follows:
AB_OTA_PARTITIONS +=
boot
system
aboot
apdp
bootlocker
cmnlib32
cmnlib64
devcfg
hosd
hyp
keymaster
modem
pmic
rpm
tz
xbl

they all go to target files, but the original radio.img and bootloader.img go to factory image, the extracted ones only go to OTA image.
i believe some partitions are missing, and there might be a better way of doing this, but it's a start ...

mke208, 2017-06-14:
i believe device/google/marlin/releasetools.py is not related to AB images ...  device/google/marlin also contains a radio/filesmap , but i guess those are for the classic update system ...

mke208, 2017-06-14:
there should be a way to extract the correct images from payload.bin using some code from updater or recovery ...

thestinger, 2017-06-15:
Here's a way to get them without that, they're in the stock factory images:

```
strcat@thinktank i ~/downloads/marlin-n2g47o master % unzip -v image-marlin-n2g47o.zip 
Archive:  image-marlin-n2g47o.zip
 Length   Method    Size  Cmpr    Date    Time   CRC-32   Name
--------  ------  ------- ---- ---------- ----- --------  ----
     130  Defl:N       96  26% 2009-01-01 00:00 3cc19cf1  android-info.txt
1721073028  Defl:N 838149491  51% 2009-01-01 00:00 5aa2be93  system_other.img
24526848  Defl:N 17288250  30% 2009-01-01 00:00 301da845  hosd.img
 1662976  Defl:N   458438  72% 2009-01-01 00:00 ab419da3  tz.img
   53248  Defl:N    19457  64% 2009-01-01 00:00 c5d3ec96  devcfg.img
 1839104  Defl:N  1079594  41% 2009-01-01 00:00 f7170294  xbl.img
   16384  Defl:N     2838  83% 2009-01-01 00:00 693116ae  apdp.img
 2629632  Defl:N   859945  67% 2009-01-01 00:00 b51b731e  aboot.img
248052500  Defl:N 106987982  57% 2009-01-01 00:00 e766451c  vendor.img
  262144  Defl:N   145808  44% 2009-01-01 00:00 7f33ccf8  cmnlib64.img
1622564756  Defl:N 857791721  47% 2009-01-01 00:00 74e3a919  system.img
   45056  Defl:N    12685  72% 2009-01-01 00:00 afa28ba4  pmic.img
   77824  Defl:N    31568  59% 2009-01-01 00:00 36f14054  bootlocker.img
  262144  Defl:N    85702  67% 2009-01-01 00:00 04a74fb2  hyp.img
  208896  Defl:N   127784  39% 2009-01-01 00:00 94f76c6d  cmnlib32.img
  360448  Defl:N   185958  48% 2009-01-01 00:00 bced3711  keymaster.img
  233472  Defl:N   113807  51% 2009-01-01 00:00 b9fcdc99  rpm.img
26621158  Defl:N 19392645  27% 2009-01-01 00:00 cabc624e  boot.img
58605568  Defl:N 31388572  46% 2009-01-01 00:00 2c317251  modem.img
--------          -------  ---                            -------
3709095316         1874122341  50%                            19 files
```

@anestisb, can you add all of these as radio files and add them to `AB_OTA_PARTITIONS`? The bootloader and radio still need to be added as radio files as they already are, but not `AB_OTA_PARTITIONS`.

anestisb, 2017-06-15:
Yup, will try to add it today.

thestinger, 2017-06-15:
And BTW if anyone wonders why system_other exists, they have a standard filter to shove the DEXPREOPT code into there to save room in their system. It gets flashed to the alternate slot system partition via factory images and used until an over-the-air update occurs. I turn off that nonsense for CopperheadOS since we update apps via over-the-air updates and want full preoptimization.

thestinger, 2017-06-15:
@mke208 This appears to work, but I'm not confident that all the bootloader partitions are being updated. Does it bump the version in the 'bootloader' screen on your device, if you start from an older bootloader and sideload an update with a new one? It bumps the radio version here, but it doesn't seem to increase the apparent bootloader version. I'll try a clean build...

thestinger, 2017-06-15:
Nevermind, I was just missing one.

mke208, 2017-06-15:
Yes it does update bootloader too. However msadp still seems to be missing, maybe it does not get updated ?

thestinger, 2017-06-15:
Not sure about that. Maybe it's included in another image.

mke208, 2017-06-15:
did you manage to update the bootloader/radio ? does the "powered by android" appear on the boot logo ?

thestinger, 2017-06-15:
Yeah, it worked out fine.

mke208, 2017-06-15:
We should take a note of the msadp partition, maybe an update will be released some time in the future.
Maybe a change in android-prepare-vendor scripts to check for bl/radio partitions and create makefiles ?

thestinger, 2017-06-15:
They aren't really marked as being bootloader / radio partitions in that zip. I think the best option is making a list of the files expected to be in the image zip and warning if something unexpected is in there.

anestisb, 2017-06-15:
Have pushed an update to include the mentioned images. Have tested it against the latest sailfish tags and it seems to include the mentioned files in the OTA builds.

```
anestisb@android-server:/fvault/aosp_src_stash$ zipinfo out_userdebug_Linux_7.1.2_r17/aosp_src/dist/aosp_sailfish-target_files-eng.anestis.zip IMAGES/*
-rw-r--r--  2.0 unx 25269544 b- defN 09-Jan-01 00:00 IMAGES/boot.img
-rw-r--r--  2.0 unx 879489480 b- defN 09-Jan-01 00:00 IMAGES/system.img
-rw-r--r--  2.0 unx   100102 b- defN 09-Jan-01 00:00 IMAGES/system.map
-rw-r--r--  2.0 unx 247913212 b- defN 09-Jan-01 00:00 IMAGES/vendor.img
-rw-r--r--  2.0 unx    26695 b- defN 09-Jan-01 00:00 IMAGES/vendor.map
-rw-r--r--  2.0 unx 243176596 b- defN 09-Jan-01 00:00 IMAGES/system_other.img
-rw-r--r--  2.0 unx 139675980 b- defN 09-Jan-01 00:00 IMAGES/userdata.img
-rw-r--r--  2.0 unx 25269544 b- defN 09-Jan-01 00:00 IMAGES/boot.img
-rw-r--r--  2.0 unx 879489480 b- defN 09-Jan-01 00:00 IMAGES/system.img
-rw-r--r--  2.0 unx 25269544 b- defN 09-Jan-01 00:00 IMAGES/boot.img
-rw-r--r--  2.0 unx 879489480 b- defN 09-Jan-01 00:00 IMAGES/system.img
-rw-r--r--  2.0 unx  2629632 b- defN 09-Jan-01 00:00 IMAGES/aboot.img
-rw-r--r--  2.0 unx    16384 b- defN 09-Jan-01 00:00 IMAGES/apdp.img
-rw-r--r--  2.0 unx    77824 b- defN 09-Jan-01 00:00 IMAGES/bootlocker.img
-rw-r--r--  2.0 unx   208896 b- defN 09-Jan-01 00:00 IMAGES/cmnlib32.img
-rw-r--r--  2.0 unx   262144 b- defN 09-Jan-01 00:00 IMAGES/cmnlib64.img
-rw-r--r--  2.0 unx    53248 b- defN 09-Jan-01 00:00 IMAGES/devcfg.img
-rw-r--r--  2.0 unx 24526848 b- defN 09-Jan-01 00:00 IMAGES/hosd.img
-rw-r--r--  2.0 unx   262144 b- defN 09-Jan-01 00:00 IMAGES/hyp.img
-rw-r--r--  2.0 unx   360448 b- defN 09-Jan-01 00:00 IMAGES/keymaster.img
-rw-r--r--  2.0 unx 58605568 b- defN 09-Jan-01 00:00 IMAGES/modem.img
-rw-r--r--  2.0 unx    45056 b- defN 09-Jan-01 00:00 IMAGES/pmic.img
-rw-r--r--  2.0 unx   233472 b- defN 09-Jan-01 00:00 IMAGES/rpm.img
-rw-r--r--  2.0 unx  1662976 b- defN 09-Jan-01 00:00 IMAGES/tz.img
-rw-r--r--  2.0 unx  1839104 b- defN 09-Jan-01 00:00 IMAGES/xbl.img
``` 

Let me know if the patches worked for you. I'm currently away from my lab, so testing capabilities are very limited.

mke208, 2017-06-15:
system.img and boot.img should not be added to vendor/google_devices/*/radio ...

anestisb, 2017-06-15:
Fixe in https://github.com/anestisb/android-prepare-vendor/commit/a254ac775775f9b80c07ca6bfda3d42d1bea7c34

mke208, 2017-06-15:
the partitions also need to be added to AB_OTA_PARTITIONS var in BoardConfigVendor.mk

thestinger, 2017-06-15:
@mke208 It seems to do that already.

mke208, 2017-06-15:
Oh i guess i missed it

mke208, 2017-06-15:
off topic: a progress bar on download  / install for copperheados updater would be nice to have ...

thestinger, 2017-06-15:
If you're looking at CopperheadOS on a Nexus 9 / Nexus 5X / Nexus 6P, that's not our current Updater app and isn't something that we wrote. The Updater app that we wrote is only for A/B devices.

thestinger, 2017-06-15:
If you're looking at CopperheadOS on a Nexus 9 / Nexus 5X / Nexus 6P, that's not our current Updater app and isn't something that we wrote. The Updater app that we wrote is only for A/B devices and only does background updates, so there's no UI to show progress. It does log progress of downloads, verification and installation.

thestinger, 2017-06-15:
Oh and the reason the old updater doesn't show progress is because OpenShift + Cloudflare with caching results in the content length being dropped since Cloudflare really wants to use chunked encoding.

mke208, 2017-06-15:
No, i was talking about the new updater ...

mke208, 2017-06-15:
So i guess this issue has finally been solved, right ?

mke208, 2017-06-15:
i took the liberty of modifying copperhead AB updater to show notifications, progress, and ask before doing anything ... i have been working on a version of android that doesn't do anything behind user's back ...

thestinger, 2017-06-15:
I started with it being fully automated so I could focus on making it fully reliable and secure. It automatically resumes when the download or installation is interrupted. I wouldn't necessarily mind having an opt-in to more user control, but not by default and it needs to be done very carefully to avoid ruining the reliability and automatic resume. The only thing that truly needs to work 100% in each release is booting far enough to get network access and updating to the next release.

anestisb, 2017-06-16:
@mke208 yeah I think the issue is resolved.

closed 2017-06-16 by anestisb

mke208, 2017-06-16:
Yes you are correct, but anyway if something gets broken, the user an always boot in recovery and install a downloaded image. Also the AB system is supposed to prevent bricking the device by booting to the alternate slot if something goes wrong ..

thestinger, 2017-06-16:
@mke208 BTW, https://github.com/CopperheadOS/platform_frameworks_base/commit/9177098395c068676b8a1f4ea8f3cd2bd4236093 enables checks to detect future issues like this.

I also added https://github.com/CopperheadOS/platform_packages_apps_Settings/commit/98930230723868f9a2db4a5fd03db75808523e8f and changes to detect future silent disabling of verified boot which happened once before when they disabled vendor.img verification for AOSP: https://github.com/CopperheadOS/platform_packages_apps_Settings/commit/1c3da1b94e36fee3ca335b6b25112e2ada1db958 https://github.com/CopperheadOS/platform_packages_apps_Settings/commit/2675e41877201201d3246f1bd4938ffa3373b0a0.

thestinger, 2017-06-16:
Submitted https://android-review.googlesource.com/#/c/416942/ too, although I don't know why they disabled these other than the fact that their radio check was wrong which I dealt with already.

thestinger, 2017-06-16:
Worth noting that the OEM unlocking toggle in AOSP doesn't work without Google Play Services unless you remove wipe data from the recovery menu though. The Nexus 5X also has an LG download mode in the bootloader firmware that it's not supposed to have and it allows flashing valid images and wiping with a locked bootloader. AFAIK, the Pixel bootloader firmware is fine as is the Nexus 6P and Nexus 9. Pixel C and Nexus 6 are probably fine too.

mke208, 2017-06-16:
Hmm the OEM unlocking seems to be working on Pixel XL ...

thestinger, 2017-06-16:
By not working I mean that you can simply boot to bootloader -> start recovery -> hold power + press volume up -> wipe data / cache, and then boot the OS and toggle off OEM unlocking. CopperheadOS removes that wipe menu entry in user builds. OEM unlocking is pretty much supposed to be an anti-theft feature but it depends on Google Play Services enforcing that a device get tied to a Google account. Google Play stores the Google account information in the factory reset protection partition alongside the oem unlocking toggle and then requires logging into the account once a device is tied to it, even after a wipe.

thestinger, 2017-06-16:
https://github.com/CopperheadOS/platform_packages_apps_Updater/commit/6c04c00521a441836d5495ec30c2cbb6dfc25756 is related since it's something recovery sideloading already does, and makes it possible to generate an update package for a specific device serial number. That way sideloading a valid signed package for a specific device can be used as an alternative to the removed wipe option if a user forgets the password (the update can simply add back wipe).

Anyway, just one of the other non-obvious issues with stock vs. AOSP.

mke208, 2017-06-16:
well i believe the anti theft is useless against service software like QPST etc ... 
most devices can go into a qualcomm diag mode that allows flashing a valid image.
i have never tried it though so i am not sure.

thestinger, 2017-06-16:
There's not supposed to be a default enabled download mode for Nexus / Pixel. Nexus 5X was an exception and I don't think LG was supposed to do that.

mke208, 2017-06-16:
Another nice to have feature would be a backup option similar to google's to a custom server 
or maybe to an encrypted USB storage. This would make switching devices very easy ... Right now 
when i change devices i flash an userdebug os and rsync /data to encrypted external storage, rsync again to the new device, restorecon and flash the user build ... 
Not only for switching devices, but a backup is always good to have ... Many have complained that Nexus 5X storage breaks after about 2 years of use ...

thestinger, 2017-06-16:
You can use `adb backup` / `adb restore`, which is a frontend to the same infrastructure. The whole backup infrastructure is part of AOSP, it's only the cloud sync implementation that's not. There's a sample, useless LocalTransport backup service implementation for Compatibility Test Suite testing and a useful implementation could be made from that. Venturing pretty far off-topic now though :).