anestisb/android-prepare-vendor #71: Pixels fail to save / restore time between reboots

thestinger, 2017-06-30:
The time service fails to save / restore the current time on Pixels, either due to a problem when loading it or because it's not being synced to hardware. To reproduce this, the SIM card needs to be removed and WiFi disabled to prevent the phone from syncing the time via either the mobile network or SNTP using WiFi.

If you can't reproduce this, let me know. I think it's an issue with AOSP + android-prepare-vendor but there might be some other variables.

I'm still trying to figure out what's going on... the time server seems to run and load the time, but it's always a placeholder epoch date. I can't figure out if the issue is that it's not ever setting the hardware clock but that seems like a good guess.

thestinger, 2017-06-30:
I think it might be system/app/TimeService that's missing.

anestisb, 2017-07-01:
I can reproduce the issue with AOSP + android-prepare-vendor for a sailfish device build against the latest tags.

Checking your fix now.

anestisb, 2017-07-01:
Fix is also confirmed to work. Thanks.

PR #72 (2017-07-01): add TimeService apk for Pixels
This provides handling for android.intent.action.TIME_SET and
android.intent.action.DATE_CHANGED syncing the time to hardware,
otherwise the time isn't saved to be loaded on the next boot.

Fixes #71
commit c01d494a (2017-07-01): add TimeService apk for Pixels

This provides handling for android.intent.action.TIME_SET and
android.intent.action.DATE_CHANGED syncing the time to hardware,
otherwise the time isn't saved to be loaded on the next boot.

Fixes #71
closed 2017-07-01 by anestisb