anestisb/android-prepare-vendor #75: VoLTE / VoWiFi doesn't work anywhere

thestinger, 2017-07-31:
I don't know what has to be done to fix this, but I suspect that it comes down to GoogleCarrierConfig vs. CarrierConfig resource differences. If you compare vendor.xml you'll see that the AOSP copy is empty vs. a whole bunch of stuff in Google's copy. I've run into other configuration differences like this across Settings, Telephony, etc. that they're doing via overlays. It's frustrating...

One option is simply bundling GoogleCarrierConfig alongside or instead of CarrierConfig, but I don't really want to do that... I think it makes more sense to extract the configuration and apply it to CarrierConfig via overlays.

However, I don't have access to a carrier with VoLTE or VoWiFi myself. In fact, my carrier doesn't even have LTE at the moment.

I don't think there's anything stopping this from working without Play. Other than Sprint / Verizon, I don't think any additional proprietary apps should be required either.

thestinger, 2017-07-31:
Here's one fix that I'm already doing:

* https://github.com/CopperheadOS/device_google_marlin/commit/8abecb6a5c64387b1b31cd9201d327650eb39c36
* https://github.com/CopperheadOS/device_lge_bullhead/commit/f05cd240bd690a9c0913f25d16380be14bff04bd
* https://github.com/CopperheadOS/device_huawei_angler/commit/4c71ad42d4edaa008868cc8a349f18fe8748bca3

It makes the Preferred networks option match ro.telephony.default_network (different for 5X vs. the others).

I've only confirmed that this works rather than comparing it vs. stock, but I'm now thinking that it'd be a good idea to go through Telephony, Settings, etc. and compare the xml resources with stock to try to figure out what they're changing with overlays in their internal device repositories.

anestisb, 2017-08-01:
To be honest I haven't played much with these techs so not sure where we need to focus. I will check if we have a carrier here in Greece that supports them and do some testing.

Yes I guess we can diff around the configs and create the overrides, but without someone to test them I'm not sure what conclusions can be made. I'm working on some ideas to create a few testing branches to play around with some eval configs.

My main concern at this point is that some issues might require patching at the AOSP tree level which creates a new set of problems for my use cases. At any case let's isolate the problems first and then we'll see which approach is optimal for everyone.

thestinger, 2017-08-01:
I started caring about VoLTE because the CopperheadOS LTE only mode (for attack surface reduction) isn't much use for most people including myself without voice calls. I use Signal calls when I can but I can't exactly get my dental office, bank, etc. to do that. I think it's really close to working already. I don't think there's really anything missing, it's just something messed up with the configuration. Frustrating that they keep this stuff internal...

thestinger, 2017-08-01:
AFAICT, other ROMs like PureNexus have gotten this working by bundling GoogleCarrierConfig. I need to find someone willing to test and then I can try getting it to work doing stuff like that. If that works, it means the configuration just needs to be reverse engineered back to xml and done for CarrierConfig via overlays.

KyleSanderson, 2017-08-03:
Sure no problem, I've been using VoLTE on Rogers for quite a number of months now. Let me know where I can grab the Marlin build from.

thestinger, 2017-08-03:
Splitting out the known CarrierConfig issue to #78. It may or may not be the cause of this across assorted carriers. There might be multiple problems...

thestinger, 2017-08-04:
CNEService.apk is included in config-naked for Pixels but not the Nexus 5X / 6P. I think that might be one of the issues and it probably makes sense to include it, since it's a Qualcomm app rather than a carrier app.

KyleSanderson, 2017-08-04:
@thestinger When should I be prepared to flash my Stock Marlin (Pixel XL) to Copperhead OS? Is there a build coming soon? Your recent feedback seems very schizophrenic.

thestinger, 2017-08-04:
> Your recent feedback seems very schizophrenic.

I don't know what you're mean by schizophrenic. I'm trying to get VoLTE working for us so that the LTE only feature is more useful. It didn't really matter before since I don't think many people care about traditional voice calls having HD audio or data dropping to 3G during those calls. This issue is just about VoLTE / VoWiFI issues in the scope of android-prepare-vendor which should be all the remaining issues after the latest CopperheadOS release which dealt with the issues introduced by CopperheadOS itself rather than baseline AOSP issues that android-prepare-vendor deals with for us and others.

> When should I be prepared to flash my Stock Marlin (Pixel XL) to Copperhead OS? Is there a build coming soon?

Official builds of CopperheadOS for Pixels aren't going to be available for free. The options are building each release from source which is free but inconvenient (official over-the-air updates are only for official builds), buying a Pixel with CopperheadOS from our store or sending a Pixel to us to be flashed for the same amount of money as what we add to the Pixel prices. Pixels have been available from our store for a while now and sources have been available since it was supported. I don't think it has ever been unclear that we were moving away from giving releases away from free.

KyleSanderson, 2017-08-04:
>> I don't think it has ever been unclear that we were moving away from giving releases away from free.

> Mar 2nd
> So next year when we've added 2nd generation Pixel phones, those will become the product and Pixels will likely become free like Nexus now.

> This Week
> Official builds / updates aren't going to be free for Pixels. OpenShift used to provide us with free hosting but that ended and Cloudflare bugs prevent scaling up the update server with Cloudflare for the modern update infrastructure.
> CopperheadOS for Pixels will remain free for non-commercial usage via sources. It makes more sense to do it that way so people that are using it for free are in the position to contribute back since they're already regularly building it.

Not to hijack this issues thread, but you've been asking the community to help support your ROM, even pinning this on Reddit which is why I thought I'd reach out. Previously it was the excuse of you should be paid for dev time, which is true, and that the Pixels would become free and supported like their Nexus brethren when the Pixel 2 was to come out. Now it's a year later and you're pushing it back to OpenShift revoking funding? Kimsufi (OVH) has offered unlimited transit, in Canada servers, and 500G of storage for $7/m. I'm building Copperhead right now on one with a giant swap partition and it's just fine... There's a plethora of free Continuous Integration tools that can also save monumentally on cost.

There's a ton of excuses and complete backtracking from you guys and it's very disheartening to see this. I appreciate the vision and what you're trying to accomplish, but forcing commercialization for the infrastructure cost of a cup of coffee at Dineen's or Dark Horse is not great. People are willing to pay to have a supported ROM, take advantage of that.

thestinger, 2017-08-04:
> and that the Pixels would become free and supported like their Nexus brethren when the Pixel 2 was to come out

Pixel support is **already** available for free *via the published sources*. It's not mandatory to publish the sources. It's our choice to do that and we've never gotten any real contributions to the OS itself in return for that. 

It was stated that official builds would only become available **after** we'd moved on to next generation devices. Not when 2nd generation Pixels come out, but *after* we add support for them. That doesn't mean immediately upon supporting the next generation devices either. It means it *won't happen before* and there has **never** been a timeline given for when they will be available.

> Previously it was the excuse of you should be paid for dev time, which is true, and that the Pixels would become free and supported like their Nexus brethren when the Pixel 2 was to come out.

That wasn't stated and I don't see how that's an excuse. There's no obligation to give away official builds for free. If you're not happy with the sources being given away for free we can stop doing that.

> Now it's a year later and you're pushing it back to OpenShift revoking funding? 

Again, completely misrepresenting what was said.

> Kimsufi (OVH) has offered unlimited transit, in Canada servers, and 500G of storage for $7/m. I'm building Copperhead right now on one with a giant swap partition and it's just fine... There's a plethora of free Continuous Integration tools that can also save monumentally on cost.

You don't know the details of the requirements and how it's set up / has to be set up, maybe you shouldn't make assumptions about how much it costs...

CopperheadOS development / maintenance takes a lot of time and effort. I work on it for 100 hours a week. There is not much time for anything else... That time isn't free.

> There's a ton of excuses and complete backtracking from you guys and it's very disheartening to see this. I appreciate the vision and what you're trying to accomplish, but forcing commercialization for the infrastructure cost of a cup of coffee at Dineen's or Dark Horse is not great.

There's no excuses and backtracking. Simply stop being dishonest and manipulative.

>  People are willing to pay to have a supported ROM, take advantage of that.

People can either purchase a Pixel from us with the OS or send their Pixel to be flashed if they want the official builds / over-the-air updates. That also comes with customer support which is part of the value. If you don't want to do that, you can build it from source and flash it on your own. There is not secret sauce. All of the sources are public. It's simply a source-based distribution for non-customers.

thestinger, 2017-08-05:
I don't see what any of this has to do with VoLTE / VoWiFi. Someone that's building from source for Pixels helped to get VoLTE working for T-Mobile for our latest release by testing changes that I provided to them until I figured out what had to be done to fix it for them. That's how this is supposed to work: they get something for us for free, and they contributed back by providing testing for various attempted workarounds until I found the solution to the problem they were running into. I'm working on the same thing with someone that uses the OS (the free offering) for their Nexus 5X with T-Mobile.

Getting it working everywhere that it does on stock will be harder (GoogleCarrierConfig vs. CarrierConfig issue).

I'm really not interested in debating our business model or whether we give away enough for free. There's no reason we're obligated to give away *anything* other than the kernel sources... which is a minuscule portion of the overall project. If people would like us to have an entirely commercial offering where payment is required for source code access, keep chipping away at my willingness to keep giving away years of endless hard work for free.

I feel obligated to contribute back to the projects that we build on and we do that (Android Open Source Project, Linux kernel, Chromium, android-prepare-vendor, etc.). That really has nothing to do with the licensing for CopperheadOS and whether official builds are free or only sources. I don't owe anyone CopperheadOS official builds. Pixels are available for sale with CopperheadOS to people willing to pay the cost we've set based on the development effort required to make the product. Sources are not made available as a way for people to use the OS for free without contributing back. Sure, people can do that, and it doesn't violate the license (if it's non-commercial) but that isn't why it's published.

thestinger, 2017-08-05:
Closing in favour of more specific issues now that I've resolved some problems and narrowed down others.

closed 2017-08-05 by thestinger