anestisb/android-prepare-vendor #76: verizon sepolicy isn't included for Pixels

thestinger, 2017-07-31:
I don't care about this right now so I'm not included it for CopperheadOS, but you might want it for your full configuration. Google used a comparison against "sailfish" for a value that's "aosp_sailfish" for AOSP:

* https://github.com/CopperheadOS/device_google_marlin/commit/6b8f29230e202cfa86d0f683f83ef754cf58d3b6

thestinger, 2017-07-31:
I think they're doing this on purpose to differentiate from AOSP vs. stock within AOSP. It's weird but at least they didn't keep the sepolicy in their internal repository. I don't want them to hide even more stuff there... it's far worse than I thought it was after looking into resources, etc. already.

anestisb, 2017-08-01:
Damn, that's a new group of problems since it does not appear to be safe to override the `TARGET_PRODUCT` as the `aosp_` filters in makefiles will miss stuff. So it seems that we need to track them down per device config and enable them manually.

Ok will start looking at it when I return from my holidays.

thestinger, 2017-08-05:
@anestisb It also seems that the standard sepolicy won't work unless the signatures are left intact.

device/marlin/sepolicy/verizon/keys.conf / device/marlin/sepolicy/verizon/mac_permissions.xml / device/marlin/sepolicy/verizon/seapp_contexts set it up so that the device/google/marlin/sepolicy/verizon/verizon.x509.pem signature is expected.

anestisb, 2017-09-10:
Xmm another tricky case. I see 3 options:
1. keys.conf to include platform certificate
2. modify mac_permissions.xml to assign seinfo tags with a different method than signature (e.g. package name)
3. Exclude Verizon bytecode from re-signing (<apk>::PRESIGNED annotation in configs). Of course this assumes non-stripped APKs in factory images thus can be applied only for Pixel/Pixel XL

Have you received any reports related to Verizon broken functionalities for CopperheadOS builds?

anestisb, 2017-09-10:
It seems that package name cannot be used without a signature (it is effectively a second level on top of sig) so option 2. is invalid. So we either need to override keys.conf or leave original signature for Verizon apps.

thestinger, 2017-09-10:
> Have you received any reports related to Verizon broken functionalities for CopperheadOS builds?

I use the naked configuration so those apps aren't included and LTE (including LTE+) works fine on Verizon. I think VoLTE / VoWiFi have no way of working without bundling them all, but I'm okay with Verizon only having 3G voice calls, etc. until they feel like adopting the standard way of doing things. Visual voicemail works fine without any of those apps, just with the boolean in the telephony configuration for it overridden.

thestinger, 2017-09-10:
VoLTE / VoWiFi are only working for us on Pixels anyway, but it might be tied to me dealing with the CarrierConfig vendor.xml overlay there while not dealing with it on the 5X/6P. I don't really have time to aim for the same level of perfection with the 5X/6P.

anestisb, 2017-09-12:
Xmm something weird is happening here. From the Verizon signed APKs it appears that only the "system/priv-app/obdm_stub/obdm_stub.apk" is using the certificate declared in "device/google/marlin/sepolicy/verizon/verizon.x509.pem". Doesn't sound right.

anestisb, 2017-12-24:
No related issues reported around this matter so closing.

closed 2017-12-24 by anestisb