anestisb/android-prepare-vendor #77: Android O support metabug

thestinger, 2017-08-03:
I've taken a look at the developer previews and there are a lot of changes to the vendor code on Pixels with a lot moving to vendor. They even moved wpa_supplicant to vendor...

It's hopefully going to be easy for Pixels since there will be less in system along with the apks being unstripped.

For the Nexus 5X and 6P, I assume the oat/odex files are going to be a problem again.

I'll see if it works a bit later to start figuring out what's going to be an issue for the real release.

thestinger, 2017-08-23:
https://github.com/anestisb/android-prepare-vendor/pull/82 deals with most of this, other than providing a new version of oatdump and the fact that the images are at https://developers.google.com/android/images instead of the usual place.

anestisb, 2017-09-09:
The old URL redirects to the new one so that part is covered. I might replace it at some point later, although I'll have to first check that the cookies hack does not break.

The oatdump host utils for Linux have been updated for API-26 and I plan to update the macOS too at some point on Monday.

I've started diff-ing AOSP builds with production Oreo images so hopefully I've covered most bits within the next few days. As soon as I'm done with Pixel I'll go with 5x / 6p.

thestinger, 2017-09-10:
Ah, they added a redirect and moved over the old layout. They had a separate page with only Oreo images for a while, without the old system being updated.

thestinger, 2017-09-12:
I'll close this in favour of the per device issues. Baseline Oreo support works well.

closed 2017-09-12 by thestinger