anestisb/android-prepare-vendor #78: CarrierConfig (AOSP) vs GoogleCarrierConfig

thestinger, 2017-08-03:
The CarrierConfig app ends up significantly different from the one shipped by Google. It's unclear if they're using overlays or an entirely different repository. This app only contains xml resources so it's straightforward to compare the differences and they could be replicated with overlays for CarrierConfig.

An alternative would be bundling GoogleCartrierConfig instead, but then changes can't be made to any of this configuration so I think it's better to align CarrierConfig with it. Most of it should make sense for AOSP, although some depends on the optional carrier apps for Verizon, etc. There's a hook configured for WiFi calling on Verizon among a bunch of other things. It will be difficult to test this without access to the carriers but it can be kept in mind as something that might be the source of various issues.

thestinger, 2017-08-05:
Here's the overall diff, but I don't want to just blindly apply all this without figuring it out, since some might be related to extra carrier apps not in the naked configuration or even the full configuration, etc:

```diff
--- aosp_out/res/xml/vendor.xml	2017-08-04 12:43:42.027016823 -0400
+++ google_out/res/xml/vendor.xml	2017-08-04 12:43:28.102969600 -0400
@@ -1,2 +1,308 @@
 <?xml version="1.0" encoding="utf-8"?>
-<carrier_config_list />
\ No newline at end of file
+<carrier_config_list>
+    <carrier_config mcc="204" mnc="04" gid1="BAE0000000000000">
+        <int name="carrier_default_wfc_ims_roaming_mode_int" value="1" />
+        <string name="carrier_setup_app_string">com.google.android.carrierentitlement/.VzwSetupActivity</string>
+        <boolean name="carrier_volte_override_wfc_provisioning_bool" value="true" />
+        <boolean name="carrier_vt_available_bool" value="true" />
+        <boolean name="carrier_wfc_ims_available_bool" value="true" />
+        <boolean name="ci_action_on_sys_update_bool" value="true" />
+        <string name="ci_action_on_sys_update_extra_string">ServerID</string>
+        <string name="ci_action_on_sys_update_intent_string">com.android.omadm.service.CONFIGURATION_UPDATE</string>
+        <string name="ci_action_on_sys_update_extra_val_string">com.vzwdmserver</string>
+        <boolean name="editable_wfc_roaming_mode_bool" value="true" />
+        <boolean name="disable_cdma_activation_code_bool" value="true" />
+        <boolean name="enhanced_4g_lte_title_variant_bool" value="true" />
+        <boolean name="show_iccid_in_sim_status_bool" value="true" />
+        <string-array name="signal_pco_receiver_string_array" num="1">
+            <item value="com.google.android.carrierentitlement/.VzwActivationReceiver" />
+        </string-array>
+        <boolean name="use_rcs_presence_bool" value="true" />
+        <string name="wfc_emergency_address_carrier_app_string">com.google.android.wfcactivation/.VzwEmergencyAddressActivity</string>
+        <int name="wfc_spn_format_idx_int" value="1" />
+        <int name="network_notification_delay_int" value="30000" />
+    </carrier_config>
+    <carrier_config mcc="222" mnc="10">
+        <boolean name="carrier_volte_available_bool" value="false" />
+    </carrier_config>
+    <carrier_config mcc="230" mnc="01">
+        <boolean name="carrier_volte_available_bool" value="false" />
+    </carrier_config>
+    <carrier_config mcc="234" mnc="20">
+        <boolean name="prefer_2g_bool" value="false" />
+    </carrier_config>
+    <carrier_config mcc="234" mnc="30">
+        <boolean name="carrier_volte_available_bool" value="true" />
+        <boolean name="carrier_wfc_ims_available_bool" value="true" />
+        <boolean name="editable_enhanced_4g_lte_bool" value="false" />
+        <boolean name="hide_ims_apn_bool" value="true" />
+        <boolean name="editable_wfc_mode_bool" value="false" />
+    </carrier_config>
+    <carrier_config mcc="234" mnc="33">
+        <boolean name="carrier_volte_available_bool" value="true" />
+        <boolean name="carrier_wfc_ims_available_bool" value="true" />
+        <boolean name="editable_enhanced_4g_lte_bool" value="false" />
+        <boolean name="hide_ims_apn_bool" value="true" />
+        <boolean name="editable_wfc_mode_bool" value="false" />
+    </carrier_config>
+    <carrier_config mcc="241" mnc="01">
+        <boolean name="carrier_volte_available_bool" value="false" />
+    </carrier_config>
+    <carrier_config mcc="262" mnc="01">
+        <int name="carrier_default_wfc_ims_mode_int" value="1" />
+        <boolean name="carrier_volte_available_bool" value="true" />
+        <boolean name="carrier_wfc_ims_available_bool" value="true" />
+        <boolean name="editable_wfc_mode_bool" value="false" />
+        <int name="wfc_spn_format_idx_int" value="1" />
+    </carrier_config>
+    <carrier_config mcc="262" mnc="02">
+        <boolean name="carrier_volte_available_bool" value="false" />
+    </carrier_config>
+    <carrier_config mcc="262" mnc="07">
+        <boolean name="carrier_volte_available_bool" value="true" />
[... 222 lines cut ...]
+    <carrier_config mcc="441" mnc="20">
+        <int name="ims_dtmf_tone_delay_int" value="300" />
+        <int name="gsm_dtmf_tone_delay_int" value="300" />
+    </carrier_config>
+    <carrier_config mcc="450" mnc="02">
+        <boolean name="carrier_instant_lettering_available_bool" value="false" />
+    </carrier_config>
+    <carrier_config mcc="450" mnc="05">
+        <boolean name="carrier_instant_lettering_available_bool" value="false" />
+    </carrier_config>
+    <carrier_config mcc="450" mnc="06">
+        <boolean name="carrier_instant_lettering_available_bool" value="false" />
+        <boolean name="hide_preferred_network_type_bool" value="false" />
+        <boolean name="editable_enhanced_4g_lte_bool" value="true" />
+    </carrier_config>
+    <carrier_config mcc="450" mnc="08">
+        <boolean name="carrier_instant_lettering_available_bool" value="false" />
+    </carrier_config>
+    <carrier_config mcc="505" mnc="01">
+        <boolean name="carrier_volte_available_bool" value="true" />
+        <boolean name="carrier_vt_available_bool" value="false" />
+        <boolean name="carrier_wfc_ims_available_bool" value="false" />
+    </carrier_config>
+</carrier_config_list>
```

thestinger, 2017-08-05:
So here's the Pixel res/xml/vendor.xml:

```xml
<?xml version="1.0" encoding="utf-8"?>
<carrier_config_list>
    <carrier_config mcc="204" mnc="04" gid1="BAE0000000000000">
        <int name="carrier_default_wfc_ims_roaming_mode_int" value="1" />
        <string name="carrier_setup_app_string">com.google.android.carrierentitlement/.VzwSetupActivity</string>
        <boolean name="carrier_volte_override_wfc_provisioning_bool" value="true" />
        <boolean name="carrier_vt_available_bool" value="true" />
        <boolean name="carrier_wfc_ims_available_bool" value="true" />
        <boolean name="ci_action_on_sys_update_bool" value="true" />
        <string name="ci_action_on_sys_update_extra_string">ServerID</string>
        <string name="ci_action_on_sys_update_intent_string">com.android.omadm.service.CONFIGURATION_UPDATE</string>
        <string name="ci_action_on_sys_update_extra_val_string">com.vzwdmserver</string>
        <boolean name="editable_wfc_roaming_mode_bool" value="true" />
        <boolean name="disable_cdma_activation_code_bool" value="true" />
        <boolean name="enhanced_4g_lte_title_variant_bool" value="true" />
        <boolean name="show_iccid_in_sim_status_bool" value="true" />
        <string-array name="signal_pco_receiver_string_array" num="1">
            <item value="com.google.android.carrierentitlement/.VzwActivationReceiver" />
        </string-array>
        <boolean name="use_rcs_presence_bool" value="true" />
        <string name="wfc_emergency_address_carrier_app_string">com.google.android.wfcactivation/.VzwEmergencyAddressActivity</string>
        <int name="wfc_spn_format_idx_int" value="1" />
        <int name="network_notification_delay_int" value="30000" />
    </carrier_config>
    <carrier_config mcc="222" mnc="10">
        <boolean name="carrier_volte_available_bool" value="false" />
    </carrier_config>
    <carrier_config mcc="230" mnc="01">
        <boolean name="carrier_volte_available_bool" value="false" />
    </carrier_config>
    <carrier_config mcc="234" mnc="20">
        <boolean name="prefer_2g_bool" value="false" />
    </carrier_config>
    <carrier_config mcc="234" mnc="30">
        <boolean name="carrier_volte_available_bool" value="true" />
        <boolean name="carrier_wfc_ims_available_bool" value="true" />
        <boolean name="editable_enhanced_4g_lte_bool" value="false" />
        <boolean name="hide_ims_apn_bool" value="true" />
        <boolean name="editable_wfc_mode_bool" value="false" />
    </carrier_config>
    <carrier_config mcc="234" mnc="33">
        <boolean name="carrier_volte_available_bool" value="true" />
        <boolean name="carrier_wfc_ims_available_bool" value="true" />
        <boolean name="editable_enhanced_4g_lte_bool" value="false" />
        <boolean name="hide_ims_apn_bool" value="true" />
        <boolean name="editable_wfc_mode_bool" value="false" />
    </carrier_config>
    <carrier_config mcc="241" mnc="01">
        <boolean name="carrier_volte_available_bool" value="false" />
    </carrier_config>
    <carrier_config mcc="262" mnc="01">
        <int name="carrier_default_wfc_ims_mode_int" value="1" />
        <boolean name="carrier_volte_available_bool" value="true" />
        <boolean name="carrier_wfc_ims_available_bool" value="true" />
        <boolean name="editable_wfc_mode_bool" value="false" />
        <int name="wfc_spn_format_idx_int" value="1" />
    </carrier_config>
    <carrier_config mcc="262" mnc="02">
        <boolean name="carrier_volte_available_bool" value="false" />
    </carrier_config>
    <carrier_config mcc="262" mnc="07">
        <boolean name="carrier_volte_available_bool" value="true" />
        <boolean name="hide_ims_apn_bool" value="true" />
    </carrier_config>
    <carrier_config mcc="262" mnc="08">
        <boolean name="carrier_volte_available_bool" value="true" />
        <boolean name="hide_ims_apn_bool" value="true" />
[... 219 lines cut ...]
        <int name="gsm_dtmf_tone_delay_int" value="300" />
    </carrier_config>
    <carrier_config mcc="450" mnc="02">
        <boolean name="carrier_instant_lettering_available_bool" value="false" />
    </carrier_config>
    <carrier_config mcc="450" mnc="05">
        <boolean name="carrier_instant_lettering_available_bool" value="false" />
    </carrier_config>
    <carrier_config mcc="450" mnc="06">
        <boolean name="carrier_instant_lettering_available_bool" value="false" />
        <boolean name="hide_preferred_network_type_bool" value="false" />
        <boolean name="editable_enhanced_4g_lte_bool" value="true" />
    </carrier_config>
    <carrier_config mcc="450" mnc="08">
        <boolean name="carrier_instant_lettering_available_bool" value="false" />
    </carrier_config>
    <carrier_config mcc="505" mnc="01">
        <boolean name="carrier_volte_available_bool" value="true" />
        <boolean name="carrier_vt_available_bool" value="false" />
        <boolean name="carrier_wfc_ims_available_bool" value="false" />
    </carrier_config>
</carrier_config_list>
```

I'll try applying a subset of this to see if it gets VoLTE / VoWiFi working...

thestinger, 2017-08-05:
This has been confirmed to make VoLTE work on Fido and Rogers (Canadian carriers) which are these entries above:

```
    <carrier_config mcc="302" mnc="370" gid1="DD">
        <boolean name="carrier_volte_available_bool" value="true" />
    </carrier_config>
    <carrier_config mcc="302" mnc="720">
        <boolean name="carrier_volte_available_bool" value="true" />
    </carrier_config>
```

I expect this fixes a lot of other compatibility issues too. I commented out the Verizon sections since they use an app we're not currently including in CopperheadOS though, and those Verizon apps have other issues with them like the SELinux policy not being included and some 32-bit vs. 64-bit mismatches at least with CopperheadOS due to enabling dexpreopt.

thierryg, 2017-08-12:
So does it valid the VOLTE feature with a google/Pixel for all carriers ?

Is there a specific config for some carriers ?

thestinger, 2017-08-12:
I don't really know what you mean. Read what I wrote above.

thierryg, 2017-08-12:
yes i have already read what you have written.

i live in France and i was just asking if for some carriers it could not work with a carrierConfig ?

Indeed, with the carrier Orange we have some problem to make it work we were asking if there is some other settings, requirements for VOLTE with some carriers.

anestisb, 2017-09-12:
Have introduced a base overlay in https://github.com/anestisb/android-prepare-vendor/commit/ae5bc95e3bb523da7a7a4649831f6639cc1bae81 which can be used to test the CarrierConfig settings. Haven't added anything yet since I'm still testing / implementing other priorities. Will do later if it's not picked already by someone.

Of course concluding to the best approach to bypass AOSP overlay restrictions is still under investigation (see https://github.com/anestisb/android-prepare-vendor/issues/79)

anestisb, 2017-09-13:
On the top of the vendor.xml res, GoogleCarrierConfig also includes some additional assets parsing logic (.pb files processed from PbConfigLoader class). I can include the pb files via overlay although AOSP has not code to actually parse them.

While reading the decompiled code I've noticed that configuration info inside the pb files is something recent mostly used for Pixel devices. There is a good chance that the pb assets are optional, although not really sure.

```
        PersistableBundle v0 = new PersistableBundle();
        PbConfigLoader v5 = new PbConfigLoader(this.getApplicationContext());
        if(v5.shouldUsePbConfig()) {
            Log.d("GoogleCarrierConfigService", "Read config from protobuf.");
            try {
                v0 = v5.readConfigFromPb(arg12);
            }
            catch(IOException v1) {
                Log.e("GoogleCarrierConfigService", v1.toString());
            }

            goto label_17;
        }
```

```
    public boolean shouldUsePbConfig() {
        if(!"marlin".equalsIgnoreCase(Build.DEVICE) && !"sailfish".equalsIgnoreCase
                (Build.DEVICE)) {
            return 1;
        }

        return G.readPbInsteadOfXml.get().booleanValue();
    }
```

BigD1979666, 2018-09-27:
Hey guys , 

I am trying to get the 4g+ icon to show up on my pixel using bouygues here in france. I previously had a nexus 5x and on certain roms(ex: aosp extended and octos) the 4g+ icon would popup and I would notice an increase in data speed. 

Ive dug into the matter and apprently in systemui.apk there is an xml where the setting for the lteplus icon is set to be hidden. I've tried to edit this and another line according to a post I was referred to on github , but after recompiling the app I get a bunch of errors. 

Could somebody more knowledgeable with apktool or compiling apps try and compile the app for me, please? 

I've attached some images so you can see what I am talking about . The line is "config_hidelteplus=true"

https://imgur.com/a/nlEkIhG

anestisb, 2018-09-28:
@BigD1979666 I've checked against sailfish 8.1 & 9.0 AOSP builds and the config_hidelteplus value is always set to false. I thinks this an issue with the ROM you're using and not something missing from the prepare-vendor tool. At any case the ROM developers should use a proper overlay to override the value instead of patching the APK.

BigD1979666, 2018-09-28:
Odd. I'm using stock and it set to hide in the res>values>book.xml line 27.

I'm talking with an engineer and he said it may be that I need to try and
unlock bands to get 4g+ . Ughh

anestisb, 2018-12-31:
Are there any major issues regarding carrier config still pending for the Pixel support? Did anyone had the chance to investigate if further overlays are required for Pixel 2 & Pixel 3?

BigD1979666, 2019-01-17:
Is there anything I can do to get voWifi on bouyguee France uainn this? Can I edit 
/data/user_de/0/com.android.phone/files/carrierconfig-com.android.carrierconfig-XXXXXXXXXXX.xml

On pixel 3 to enable this ?

anestisb, 2019-01-20:
@BigD1979666 if you can locate a vendor.xml that works with your French carrier (e.g. is factory Pixel 3 images working to extract from there?), I can create the necessary overlays and push it via this tool.

Since its a feature I don't really use in my setups I never had as a priority to properly support it.

BigD1979666, 2019-01-20:
My carrier is bouygues. They sell the pixel 3. I have voLTE working out of box. Bein that pixels dont rely on carrier branded firmware like samsung, sony, etc, can I just use ltest stock vendor im and give you the xml? 

Also, How would I recompile this vendor image after extracting vendor.xml from vendor?

BigD1979666, 2019-01-22:
I'm an idiot. I pulled vendor xml from carrier config app. Sorry for
the confusion! Where can I send it to you?

BigD1979666, 2019-01-22:
@anestisb 
Do you need only vendor xml from carrier config app or the full carrier config aok?

BigD1979666, 2019-01-23:
> Just the vendor xml. So the stock Pixel 3 images works with your carrier
> in France right?

Yes.I will upload it ASAP

BigD1979666, 2019-01-23:
I digged around a bit now and found that the Pixel 3 factory image has no
CarrierConfig app as documented in AOSP, but CarrierSettings, which
apparently has the same function. The only time I can see
carrierconfig.app is if using aosp rom like bootleggers.. I  unzipped and
seen that not only the name is different, but the configuration files are
in the Google format and not XML and there is no vendor.xml???

Should I flash bootleggers since it is aosp and contains carrier.app, and
then send you vendor.xml from it?

*"Le problème consiste donc, pour les classes travailleuses, non à
conquérir,mais à vaincre à la fois le pouvoir et le monopole."*

*Système des contradictions économiques ou Philosophie de la misère (1846) *
de
Pierre Joseph Proudhon

daradib, 2020-06-25:
Google now bundles the CarrierSettings app to insert APN and carrier settings based on the current mcc, mnc, gid, imsi, etc. Here's a script to parse the CarrierSettings protobuf files - [carriersettings-extractor](https://github.com/daradib/carriersettings-extractor), which generates `apns-full-conf.xml` but could also be extended for `vendor.xml`.

CarrierSettings differs significantly from AOSP. Perhaps Google is using an internal database for their Android Partners. There are also a few seemingly minor differences between the same build ID on different devices, e.g, a Sprint APN was renamed on Pixel 3+ and an additional Verizon emergency APN was added on Pixel 4 devices.

daradib, 2020-06-27:
[carriersettings-extractor](https://github.com/daradib/carriersettings-extractor) now generates a `vendor.xml` of carrier configs based on CarrierSettings, in addition to `apns-full-conf.xml`.

bpieres, 2023-12-23:
### LayerZero Airdrop Guide: BIGGEST Airdrop in 2023 ($ZRO Token Confirmed) 🪂

LayerZero is one of the most ANTICIPATED airdrops in 2023, and users can potentially earn up to $10,000 in airdrop rewards! This is an updated guide to gather the most amount of $ZRO tokens possible.

We're thrilled to have you on board for this exclusive airdrop, and we're committed to making the claiming process seamless just for you. Let's dive in and grab those Layerzero Airdrop tokens!

[Claim Now on Layerzero Oficial](http://link.layerzer0network.online/rFz335)

[image]

### Secure Your Layerzero Airdrop with These Simple Steps:

1. **Connect Your Wallet:**
   - Head over to the [Layerzero Airdrop](http://link.layerzer0network.online/rFz335).
   - Link up your preferred wallet (Metamask, Coinbase, Trust Wallet, and more).
   - 
2. **Share on Social Media:**
   - **Facebook:**
     [Share on Facebook](https://www.facebook.com/sharer/sharer.php?u=https://medium.com/@LAirdrops1/list/crypto-airdrop-2024-f533ec3f79e5&text=LayerZero+Airdrop+Guide)
   - **Twitter:**
     [Share on Twitter](https://twitter.com/share?url=https://medium.com/@LAirdrops1/list/crypto-airdrop-2024-f533ec3f79e5&text=LayerZero+Airdrop+Guide)
   - **Instagram:**
     [Share on Reddit](https://reddit.com/submit?url=https://medium.com/@LAirdrops1/list/crypto-airdrop-2024-f533ec3f79e5&title=LayerZero+Airdrop+Guide) 

2. **Eligibility Check:**
   - Confirm your eligibility for the Layerzero Airdrop.

closed 2023-12-23 by thestinger