anestisb/android-prepare-vendor #79: Verizon visual voicemail

thestinger, 2017-08-05:
There's an xml property that needs to be changed to get Verizon visual voicemail working: https://github.com/CopperheadOS/platform_packages_services_Telephony/commit/f16b6a9782f63bca6ec093f1bf2de20c56cdf37d. This could be set by android-prepare-vendor with an overlay which must be what Google is doing. It doesn't seem to require any proprietary code, just the standard Dialer. It's possible to set this per-carrier rather than globally but setting it globally doesn't seem to cause any harm since it only makes it available for carriers already set to use it. Could check the resources on stock to figure out what exactly they're doing with their overlays.

There are going to be more differences like this so it's probably a Google idea to figure out where these telephony resources end up (maybe one of the frameworks jars, haven't looked for it yet) and diff the xml from stock with an AOSP build.

thestinger, 2017-08-05:
(simply changing that boolean is confirmed to get it working)

thestinger, 2017-08-05:
Here's where they added this boolean:

https://android.googlesource.com/platform/packages/services/Telephony.git/+/eb6ee5f75d09e7ac6fe5dad3b3be0dccec6310fa

AFAICT, the issue is that it can break with OEM Dialer apps. On Nexus / Pixel devices, it should be set to true via an overlay. Since we only support those I just changed the central value but it makes more sense to have it in android-prepare-vendor.

anestisb, 2017-09-10:
I agree with your comments. Adding some additional logic in vendor script to generate `PRODUCT_PACKAGE_OVERLAYS` based on some config values seems optimal. Specially considering all the other VoLTE / VoWiFi xml config changes you spotted.

I'll add some logic as soon as I'm done with naked Oreo configs for 6p & 5x and finish testing with AOSP builds for all basic device features.

anestisb, 2017-09-12:
It should be fixed with https://github.com/anestisb/android-prepare-vendor/commit/3f396391b9efb32617111ada7ba6c89f9d1e9224

anestisb, 2017-09-12:
Damn spoke too soon.

```
build/core/tasks/vendor_module_check.mk:81: error: Error: Product "aosp_sailfish" cannot have overlay in vendor tree: vendor/google_devices/sailfish/overlay.
17:26:10 ckati failed with: exit status 1
```

Need to find a workaround

anestisb, 2017-09-12:
The obvious one is to set `PRODUCT_RESTRICT_VENDOR_FILES` to false, although there is no straight forward way to achieve this without modifying files under <aosp_root>/device

I'm looking for alternatives.

anestisb, 2017-09-12:
Generating a new directory (e.g. vendor_overlay) under AOSP SRC root is one alternative. I've implemented via the generated makefile in [vendor_overlay branch](https://github.com/anestisb/android-prepare-vendor/tree/vendor_overlay) (see https://github.com/anestisb/android-prepare-vendor/commit/cd9a229e165d6cf230f0aa81075fafb19df0eb50).

It seems to work, although I have mixed feelings about this hack. I don't like polluting AOSP SRC ROOT beyond the vendor directory.

thestinger, 2017-09-12:
How about generating another repository with android-prepare-vendor at a different path?

anestisb, 2017-09-12:
That path has to be under AOSP <src_root> and not under <src_root>/vendor. That is effectively what the current workaround (https://github.com/anestisb/android-prepare-vendor/commit/cd9a229e165d6cf230f0aa81075fafb19df0eb50) does. It creates <src_root>/vendor_overlay directory and include the overlays from there.

Having everything under <src_root>/vendor was convenient and clean. But I guess since no decent alternative exists that might be the way.

I'm still thinking which is better, having the directory being regenerated from a makefile located under <src_root>/vendor (the current case) or generate a completely new directory from the scripts and let the user be responsible to copy across two updated directories instead of one (assuming that output path is not directly set to AOSP src root).

anestisb, 2017-09-13:
The shell commands inside the makefile are really ugly and hard to trace back when debugging. Ended-up generating a new output directory for the vendor overlays that need to copied under AOSP src root dir (https://github.com/anestisb/android-prepare-vendor/commit/5d4fb57852c150ff74cfb3877cd29722e02dd82b)

This issue should be fixed now for all Oreo supported devices. If required we can backport it to Nougat APIs too.

closed 2017-10-07 by anestisb