anestisb/android-prepare-vendor #80: VoLTE / VoWiFi don't work on T-Mobile (or perhaps at all) for the Nexus 5X / 6P

thestinger, 2017-08-05:
VoLTE / VoWiFi works on T-Mobile with Pixels, but seemingly not with the Nexus 5X / 6P.

I tried including CNEService.apk but that wasn't enough to make it work.

anestisb, 2017-09-12:
Will pick this up as soon as Pixel is on a good state & tested properly.

anestisb, 2017-09-13:
N5x & N6p don't use the GoogleCarrierConfig like the Pixels. Instead they appear to build the AOSP CarrierConfig app with a replaced vendor.xml.

Have copied the latest version from Oreo factory images and created the overlays https://github.com/anestisb/android-prepare-vendor/commit/b8ded85b9ea8d00409f8a70a7846bf7ca3cae70a & https://github.com/anestisb/android-prepare-vendor/commit/a5ae78f9e3b1c8398c8b42885e08e14597481271

Have a look at them and let me know if they worked so we can port (all or subset of changes) to naked config overlays too.

anestisb, 2017-10-07:
It would be nice if we can get these retested again after the recent changes applied to 6p/5x configs. Having the core deoptimization issue resolved we can start now working the smaller broken / missing bits.

thestinger, 2018-01-05:
I don't know if the overlays fix this but neither of us can test this and we no longer care about it for CopperheadOS so I think it makes sense to close it and someone can open a new issue if they can confirm there's still a problem.

thestinger, 2018-01-05:
I expect it should work on normal carriers with the overlays anyway.

closed 2018-01-05 by thestinger