anestisb/android-prepare-vendor #81: Running error

garasmc, 2017-08-05:
hi, I was in use with a problem. 

Ubuntu 16

> sudo ./execute-all.sh -d angler -b MTC20F -o $(pwd) -i ../angler-mtc20f-factory-4355fe06.zip -k
[*] Setting output base to '/home/garasmc/source/android-prepare-vendor/angler/mtc20f'
[*] Extracting '/home/garasmc/source/angler-mtc20f-factory-4355fe06.zip'
[*] Unzipping 'image-angler-mtc20f.zip'
[*] Processing with 'API-23 config-naked' configuration
[*] '9' bytecode archive files will be repaired
[*] Repairing bytecode under /system partition using oat2dex method
[*] Preparing environment for 'arm' ABI
[*] Preparing environment for 'arm64' ABI
[*] Start processing system partition & de-optimize pre-compiled bytecode
[!] '/framework/framework-res.apk' not pre-optimized & without 'classes.dex' - copying without changes
[*] '/framework/framework.jar' is multi-dex - adjusting recursive archive adds
[*] System partition successfully extracted & repaired at '/home/garasmc/source/android-prepare-vendor/angler/mtc20f/factory_imgs_repaired_data'
[*] Generating blobs for vendor/huawei/angler
[*] Copying radio files '/home/garasmc/source/android-prepare-vendor/angler/mtc20f/vendor/huawei/angler'
[*] Copying product files & generating 'angler-vendor-blobs.mk' makefile
cp: cannot stat '/home/garasmc/source/android-prepare-vendor/angler/mtc20f/factory_imgs_repaired_data/system/priv-app/CallStatistics/CallStatistics.apk': No such file or directory
[-] Failed to copy 'system/priv-app/CallStatistics/CallStatistics.apk'
[-] Vendor generation failed

plz help me . 
Thanks

anestisb, 2017-09-08:
Please try with the `--debugfs` flag since the current default fuse-ext2 has been regularly reported not working properly.

anestisb, 2017-10-06:
DebugFS was activated by default when running under a Linux host since fuse-ext2 has been spotted to have many issues.

closed 2017-10-06 by anestisb