anestisb/android-prepare-vendor #87: Marlin (Pixel XL) API-26 support

anestisb, 2017-09-08:
- [ ] Create naked base configuration
- [ ] Create full configuration
- [ ] Verify compile-able builds
- [ ] Diff with production for additional blobs
- [ ] Test device features for AOSP build

anestisb, 2017-09-15:
Unfortunately I don't have a spare Pixel XL available at the moment to verify generated AOSP builds against an actual device. Please let me know if any outstanding issues are still open.

thestinger, 2017-09-15:
No issues with it. I think it's fairly safe to assume that if the configuration here is identical as the Pixel, there won't be issues, since it's nearly the same thing just with difference resources for large devices and a property difference or two tied to graphical asset size.

anestisb, 2017-09-15:
Excellent. Btw I've contact some carriers here in Greece that offer VoWiFI & VoLTE services and see if can I dig a bit further the reported fancy carrier issues. At any case the overlays processing is already in place to support finalised configuration investigations that need to override AOSP defaults.

closed 2017-09-15 by anestisb

thestinger, 2017-09-15:
I'm still using our take on it for now since I don't want to change it without testing the new way, but I'll eventually move to using the overlays from here and contributing to them.