anestisb/android-prepare-vendor #96: libminui.so might be able to come from AOSP

thestinger, 2017-09-11:
It looks like it's the library in bootable/recovery/minui, but nothing in AOSP is pulling it in as a dependency. I'm not sure what actually depends on it in the Nexus/Pixel vendor files that makes it get pulled in as a shared library dependency for them, but it's likely possible to set up a dummy module that depends on it and gets it pulled in.

thestinger, 2017-09-11:
Not 100% sure that they aren't modifying it but it's usually easy to tell since AOSP builds are deterministic, although I only have around the CopperheadOS oreo-r6-release tree at the moment which does change the build output.

anestisb, 2017-09-12:
Ok will check it.

anestisb, 2017-09-12:
Except compiler metadata AOSP libminui is identical to factory image. Removed them from blobs list and forced usage of AOSP module in https://github.com/anestisb/android-prepare-vendor/commit/bade1234af29c52033cbdae5856f41702d0e13e0

thestinger, 2017-09-12:
Confirmed that this works properly.

closed 2017-09-12 by thestinger