anestisb/android-prepare-vendor #100: sailfish API-26 CarrierServices crashes

anestisb, 2017-09-13:
Spotted when building AOSP with API-26 naked config

```
09-13 16:09:45.040  2557  2557 I CarrierServices: [2] a.a: Initializing Carrier Services Library.
09-13 16:09:45.044  2557  2557 D AndroidRuntime: Shutting down VM
09-13 16:09:45.045  2557  2557 E AndroidRuntime: FATAL EXCEPTION: main
09-13 16:09:45.045  2557  2557 E AndroidRuntime: Process: com.google.android.ims, PID: 2557
09-13 16:09:45.045  2557  2557 E AndroidRuntime: java.lang.RuntimeException: Unable to create application com.google.android.ims.CarrierServicesReleaseApp: java.lang.SecurityException: Failed to find provider com.google.android.gsf.gservices for user 0; expected to find a valid ContentProvider for this authority
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5794)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at android.app.ActivityThread.-wrap1(Unknown Source:0)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1661)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:105)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:164)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:6541)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)
09-13 16:09:45.045  2557  2557 E AndroidRuntime: Caused by: java.lang.SecurityException: Failed to find provider com.google.android.gsf.gservices for user 0; expected to find a valid ContentProvider for this authority
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at android.os.Parcel.readException(Parcel.java:1942)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at android.os.Parcel.readException(Parcel.java:1888)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at android.content.IContentService$Stub$Proxy.registerContentObserver(IContentService.java:768)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at android.content.ContentResolver.registerContentObserver(ContentResolver.java:1924)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at android.content.ContentResolver.registerContentObserver(ContentResolver.java:1913)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at com.google.android.b.a.a(SourceFile:7)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at com.google.android.b.a.b(SourceFile:115)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at com.google.android.b.a.a(SourceFile:86)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at com.google.android.a.a.g.a(SourceFile:4)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at com.google.android.a.a.b.a(SourceFile:3)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at com.google.android.a.a.a.a(SourceFile:12)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at com.google.android.ims.q.h.a(SourceFile:38)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at com.google.android.ims.q.h.a(SourceFile:2)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at com.google.android.ims.a.a(SourceFile:1)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at com.google.android.ims.CarrierServicesReleaseApp.onCreate(SourceFile:9)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1118)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5791)
09-13 16:09:45.045  2557  2557 E AndroidRuntime:        ... 8 more
09-13 16:09:50.028  2557  2574 I zygote64: Waiting for a blocking GC ProfileSaver
09-13 16:09:50.034  2557  2574 I zygote64: WaitForGcToComplete blocked for 6.210ms for cause ProfileSaver
```

anestisb, 2017-09-13:
Preserving the original Qualcomm signature for `vendor/app/QAS_DVC_MSP/QAS_DVC_MSP.apk` appears to resolve the problem.

```
include $(CLEAR_VARS)
LOCAL_MODULE := QAS_DVC_MSP
LOCAL_MODULE_TAGS := optional
LOCAL_BUILT_MODULE_STEM := package.apk
LOCAL_MODULE_OWNER := google
LOCAL_MODULE_PATH := $(PRODUCT_OUT)/$(TARGET_COPY_OUT_VENDOR)/app
LOCAL_SRC_FILES := vendor/app/QAS_DVC_MSP/QAS_DVC_MSP.apk
LOCAL_CERTIFICATE := PRESIGNED
LOCAL_MODULE_CLASS := APPS
LOCAL_MODULE_SUFFIX := $(COMMON_ANDROID_PACKAGE_SUFFIX)
LOCAL_DEX_PREOPT := false
include $(BUILD_PREBUILT)
```

I wonder if it's safer to restore the old logic where all bytecode under /vendor is left PRESIGNED or explicitly handle each case. Or maybe add some detection logic to keep non-Google signatures such as Qualcomm & Verizon (see [here](https://gist.github.com/anestisb/c83dd7a2c8dc5bce3cc8843d383937a5))

anestisb, 2017-09-13:
Furthermore the "com.qualcomm.ltebc_vzw" classes from the QAS_DVC_MSP.apk appear to strongly bind with the "vzw_msdc_api.apk". More specifically the first has a copy of the latter under its assets dir (`assets/vzw_msdc_api/vzw_msdc_api.apk`). It compares the MD5 sig of the asset against the one located under the `/vendor/app/vzw_msdc_api/vzw_msdc_api.apk`. If they don't match it tries to override destination which of course fails since /vendor is read-only.

Keeping the original PRESIGNED signature for `vzw_msdc_api.apk` satisfies the previous check and prevents a bunch of error/warning LTE messages from being displayed in logcat.

There is a lot of functionality implemented in the two packages so not sure what exactly is broken when they're resigned. However, since signature checks are hardcoded inside the code it seems that both APKs should be excluded from resigning. This is not a problem for android-prepare-vendor since all bytecode under /vendor is never optimised and stripped in factory images.

After I've looked into the N6p/N5x vendor bits I'll decide if it's better to explicitly exclude the APKs or add some generic certificate parsing logic in the scripts that will preserve the sigs for APKs not signed by Google (e.g. Qualcom, Verizon, etc.).

anestisb, 2017-09-13:
Created a temp branch to test the vendor PRESIGNED requirements. Can be found [here](https://github.com/anestisb/android-prepare-vendor/tree/vendor_presigned)

mydongistiny, 2017-12-18:
I had to change CarrierServices ims and imssettings to PRESIGNED to get them to stop crashing on API27. Wasn't sure if anyone else was having this problem so I didn't open a new ticket.

anestisb, 2017-12-19:
@mydongistiny thanks for reporting.

Is the signature error produced by the affected apps themselves or from a third app (a Google one?)? Did you use the naked or the full config for sailfish?

mydongistiny, 2017-12-19:
@anestisb The error is from whenever another app tries to use it like Messenger, SetupWizard etc. And I used the full config, but it was on marlin. I forgot this report for for sailfish sorry. Here's a bigger part of the logcat in case you wanted to see it. https://hastebin.com/ocemamoyaw.sql

anestisb, 2017-12-19:
Thanks. I'll try to reproduce and confirm the fix.

anestisb, 2018-01-06:
@mydongistiny I haven't been able to reproduce the issue. However, I've made a set of changes based on some static analysis checks I did against the mentioned archives. Can you please pull the latest update of the [vendor_presigned branch](https://github.com/anestisb/android-prepare-vendor/tree/vendor_presigned) and check if it resolves your problem?

mydongistiny, 2018-01-06:
It's not a problem anymore with CarrierServices gone I guess because everything works fine now on the regular branch.

anestisb, 2018-01-06:
Sounds good. Closing then.

closed 2018-01-06 by anestisb