anestisb/android-prepare-vendor #106: Cant auto download Nexus Images

minidude2012, 2017-10-08:
[amy-clara-rose@Predator-G3 android-prepare-vendor]$ sh ./execute-all.sh -d angler -b angler-opr5.170623.007 -o /fast-datavault/nexus-vendor-blobs -f
[*] Setting output base to '/home/amy-clara-rose/android-prepare-vendor//angler/angler-opr5.170623.007'

--{ Google Terms and Conditions
Downloading of the system image and use of the device software is subject to the
Google Terms of Service [1]. By continuing, you agree to the Google Terms of
Service [1] and Privacy Policy [2]. Your downloading of the system image and use
of the device software may also be subject to certain third-party terms of
service, which can be found in Settings > About phone > Legal information, or as
otherwise provided.

[1] https://www.google.com/intl/en/policies/terms/
[2] https://www.google.com/intl/en/policies/privacy/

[?] I have read and agree with the above terms and conditions - ACKNOWLEDGE [y|n]: y
[-] Image URL not found
[-] Images download failed
[amy-clara-rose@Predator-G3 android-prepare-vendor]$

anestisb, 2017-10-08:
Something appears to be broken with your env. You set the output path to `/fast-datavault/nexus-vendor-blobs`, although the logs print `/home/amy-clara-rose/android-prepare-vendor//angler/angler-opr5.170623.007`.

Please double check your paths and calling params.

minidude2012, 2017-10-08:
im not sure all i did was copy and edit what you put for volantis downloading zip

anestisb, 2017-10-09:
Execute the following (bash debug enabled) and paste the output here:

```
set -x ./execute-all.sh -d angler -b opr5.170623.007 -o /fast-datavault/nexus-vendor-blobs -f
```

minidude2012, 2017-10-09:
++ printf '\033]0;%s@%s:%s\007' amy-clara-rose Predator-G3 '~/android-prepare-vendor'

minidude2012, 2017-10-10:
+ set -e
+ set -u
+++ dirname ./execute-all.sh
++ cd .
++ pwd
+ readonly SCRIPTS_ROOT=/home/amy-clara-rose/android-prepare-vendor
+ SCRIPTS_ROOT=/home/amy-clara-rose/android-prepare-vendor
++ mktemp -d /tmp/android_prepare_vendor.XXXXXX
+ readonly TMP_WORK_DIR=/tmp/android_prepare_vendor.oq8PuE
+ TMP_WORK_DIR=/tmp/android_prepare_vendor.oq8PuE
+ SYS_TOOLS=("mkdir" "dirname" "wget" "mount" "shasum")
+ declare -a SYS_TOOLS
+ readonly REALPATH_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/realpath.sh
+ REALPATH_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/realpath.sh
+ readonly CONSTS_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/constants.sh
+ CONSTS_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/constants.sh
+ readonly DOWNLOAD_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/download-nexus-image.sh
+ DOWNLOAD_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/download-nexus-image.sh
+ readonly EXTRACT_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/extract-factory-images.sh
+ EXTRACT_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/extract-factory-images.sh
+ readonly GEN_BLOBS_LIST_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/gen-prop-blobs-list.sh
+ GEN_BLOBS_LIST_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/gen-prop-blobs-list.sh
+ readonly REPAIR_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/system-img-repair.sh
+ REPAIR_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/system-img-repair.sh
+ readonly VGEN_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/generate-vendor.sh
+ VGEN_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/generate-vendor.sh
+ trap 'abort 1' SIGINT SIGTERM
+ . /home/amy-clara-rose/android-prepare-vendor/scripts/realpath.sh
++ set -e
+ . /home/amy-clara-rose/android-prepare-vendor/scripts/constants.sh                                                                                                                                                                         
++ [[ /home/amy-clara-rose/android-prepare-vendor/scripts/constants.sh == \.\/\e\x\e\c\u\t\e\-\a\l\l\.\s\h ]]                                                                                                                                
++ SUPPORTED_DEVICES=("bullhead" "flounder" "angler" "sailfish" "marlin")                                                                                                                                                                    
++ declare -ra SUPPORTED_DEVICES                                                                                                                                                                                                             
++ readonly NID_URL=https://google.com                                                                                                                                                                                                       
++ NID_URL=https://google.com                                                                                                                                                                                                                
++ readonly GURL=https://developers.google.com/android/nexus/images                                                                                                                                                                          
++ GURL=https://developers.google.com/android/nexus/images                                                                                                                                                                                   
++ readonly TOSURL=https://developers.google.com/profile/acknowledgeNotification                                                                                                                                                             
++ TOSURL=https://developers.google.com/profile/acknowledgeNotification                                                                                                                                                                      
++ readonly 'L_OATDUMP_URL_API23=https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21490&authkey=ACA4f4Zvs3Tb_SY'                                                                                               
++ L_OATDUMP_URL_API23='https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21490&authkey=ACA4f4Zvs3Tb_SY'                                                                                                        
++ readonly 'D_OATDUMP_URL_API23=https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21493&authkey=AJ0rWu5Ci8tQNLY'                                                                                               
++ D_OATDUMP_URL_API23='https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21493&authkey=AJ0rWu5Ci8tQNLY'                                                                                                        
++ readonly 'L_OATDUMP_URL_API24=https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21492&authkey=AE4uqwH-THvvkSQ'                                                                                               
[... cut ...]
++++ readlink -- /fast-datavault/nexus-vendor-blobs
+++ path=
+++ '[' 1 -eq 0 ']'
+++ printf '%s\n' /fast-datavault/nexus-vendor-blobs
++ _canonicalize_path /fast-datavault/nexus-vendor-blobs
++ '[' -d /fast-datavault/nexus-vendor-blobs ']'
++ __canonicalize_file_path /fast-datavault/nexus-vendor-blobs
++ local dir file
+++ dirname -- /fast-datavault/nexus-vendor-blobs
++ dir=/fast-datavault
+++ basename -- /fast-datavault/nexus-vendor-blobs
++ file=nexus-vendor-blobs
++ cd /fast-datavault
+ OUTPUT_DIR=

anestisb, 2017-10-10:
From the snippet you've provided I cannot see any obvious problem. However, it does not appear to be complete. Can you please double check you copy and paste the entire text? If you struggle copying from your terminal you can redirect script's debug output to a file and attach it to a commend.

minidude2012, 2017-10-10:
that was all it gave me literally if you want you can teamviewer to my pc and diagnose yourself

anestisb, 2017-10-10:
Sorry can't do that. Can you try a different output directory (e.g. `-o $(pwd)`)?

mikalackis, 2017-10-10:
Sorry to interrupt the thread, but are Nexus (5X, 6P) devices now supported for Oreo?

anestisb, 2017-10-10:
@mikalackis yes. Check [this](https://github.com/anestisb/android-prepare-vendor/issues/84#issuecomment-334427657) thread.

minidude2012, 2017-10-10:
[amy-clara-rose@Predator-G3 android-prepare-vendor]$ bash -x ./execute-all.sh -d angler -b opr5.170623.007 -o $(pwd) -f                            
+ set -e
+ set -u
+++ dirname ./execute-all.sh
++ cd .
++ pwd
+ readonly SCRIPTS_ROOT=/home/amy-clara-rose/android-prepare-vendor
+ SCRIPTS_ROOT=/home/amy-clara-rose/android-prepare-vendor
++ mktemp -d /tmp/android_prepare_vendor.XXXXXX
+ readonly TMP_WORK_DIR=/tmp/android_prepare_vendor.ZWCJIR
+ TMP_WORK_DIR=/tmp/android_prepare_vendor.ZWCJIR
+ SYS_TOOLS=("mkdir" "dirname" "wget" "mount" "shasum")
+ declare -a SYS_TOOLS
+ readonly REALPATH_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/realpath.sh
+ REALPATH_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/realpath.sh
+ readonly CONSTS_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/constants.sh
+ CONSTS_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/constants.sh
+ readonly DOWNLOAD_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/download-nexus-image.sh
+ DOWNLOAD_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/download-nexus-image.sh
+ readonly EXTRACT_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/extract-factory-images.sh
+ EXTRACT_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/extract-factory-images.sh
+ readonly GEN_BLOBS_LIST_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/gen-prop-blobs-list.sh
+ GEN_BLOBS_LIST_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/gen-prop-blobs-list.sh
+ readonly REPAIR_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/system-img-repair.sh
+ REPAIR_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/system-img-repair.sh
+ readonly VGEN_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/generate-vendor.sh
+ VGEN_SCRIPT=/home/amy-clara-rose/android-prepare-vendor/scripts/generate-vendor.sh
+ trap 'abort 1' SIGINT SIGTERM
+ . /home/amy-clara-rose/android-prepare-vendor/scripts/realpath.sh
++ set -e
+ . /home/amy-clara-rose/android-prepare-vendor/scripts/constants.sh
++ [[ /home/amy-clara-rose/android-prepare-vendor/scripts/constants.sh == \.\/\e\x\e\c\u\t\e\-\a\l\l\.\s\h ]]
++ SUPPORTED_DEVICES=("bullhead" "flounder" "angler" "sailfish" "marlin")
++ declare -ra SUPPORTED_DEVICES
++ readonly NID_URL=https://google.com
++ NID_URL=https://google.com
++ readonly GURL=https://developers.google.com/android/nexus/images
++ GURL=https://developers.google.com/android/nexus/images
++ readonly TOSURL=https://developers.google.com/profile/acknowledgeNotification
++ TOSURL=https://developers.google.com/profile/acknowledgeNotification
++ readonly 'L_OATDUMP_URL_API23=https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21490&authkey=ACA4f4Zvs3Tb_SY'
++ L_OATDUMP_URL_API23='https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21490&authkey=ACA4f4Zvs3Tb_SY'
++ readonly 'D_OATDUMP_URL_API23=https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21493&authkey=AJ0rWu5Ci8tQNLY'
++ D_OATDUMP_URL_API23='https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21493&authkey=AJ0rWu5Ci8tQNLY'
++ readonly 'L_OATDUMP_URL_API24=https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21492&authkey=AE4uqwH-THvvkSQ'
++ L_OATDUMP_URL_API24='https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21492&authkey=AE4uqwH-THvvkSQ'
++ readonly 'D_OATDUMP_URL_API24=https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21491&authkey=AHvCaYwFBPYD4Fs'
++ D_OATDUMP_URL_API24='https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21491&authkey=AHvCaYwFBPYD4Fs'
++ readonly 'L_OATDUMP_URL_API25=https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21503&authkey=AKDpBAzhzum6d7w'
++ L_OATDUMP_URL_API25='https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21503&authkey=AKDpBAzhzum6d7w'
++ readonly 'D_OATDUMP_URL_API25=https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21504&authkey=AC5YFNSAZ31-W3o'
++ D_OATDUMP_URL_API25='https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21504&authkey=AC5YFNSAZ31-W3o'
++ readonly 'L_OATDUMP_URL_API26=https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21557&authkey=AG47qhXu164sYwc'
++ L_OATDUMP_URL_API26='https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21557&authkey=AG47qhXu164sYwc'
++ readonly 'D_OATDUMP_URL_API26=https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21561&authkey=ABu-oqJbQDwQ-ZQ'
++ D_OATDUMP_URL_API26='https://onedrive.live.com/download?cid=D1FAC8CC6BE2C2B0&resid=D1FAC8CC6BE2C2B0%21561&authkey=ABu-oqJbQDwQ-ZQ'
++ readonly L_OATDUMP_API23_SIG=688f1c1f97e3b9eb32702c60ca05abbf313bc98a25334aa3ed4a108095162244
++ L_OATDUMP_API23_SIG=688f1c1f97e3b9eb32702c60ca05abbf313bc98a25334aa3ed4a108095162244
++ readonly D_OATDUMP_API23_SIG=688f1c1f97e3b9eb32702c60ca05abbf313bc98a25334aa3ed4a108095162244
++ D_OATDUMP_API23_SIG=688f1c1f97e3b9eb32702c60ca05abbf313bc98a25334aa3ed4a108095162244
++ readonly L_OATDUMP_API24_SIG=1f99e7d0f2894cfe52fb7f2a24d5076f217977cbb1a46fafdf5ea38b0a11adce
++ L_OATDUMP_API24_SIG=1f99e7d0f2894cfe52fb7f2a24d5076f217977cbb1a46fafdf5ea38b0a11adce
++ readonly D_OATDUMP_API24_SIG=4a7f5614eb04d9bea85bfa05853523843f9cc80a64eab4c98efed2f70ed3d90e
[... cut ...]
+ local mount_point=/home/amy-clara-rose/android-prepare-vendor/angler/opr5.170623.007/factory_imgs_data/system
+ [[ -d /home/amy-clara-rose/android-prepare-vendor/angler/opr5.170623.007/factory_imgs_data/system ]]
+ unmount_raw_image /home/amy-clara-rose/android-prepare-vendor/angler/opr5.170623.007/factory_imgs_data/vendor
+ local mount_point=/home/amy-clara-rose/android-prepare-vendor/angler/opr5.170623.007/factory_imgs_data/vendor
+ [[ -d /home/amy-clara-rose/android-prepare-vendor/angler/opr5.170623.007/factory_imgs_data/vendor ]]
+ rm -rf /tmp/android_prepare_vendor.ZWCJIR
+ exit 1

anestisb, 2017-10-11:
Ok from your latest snippet I can see that the deps are successfully found and all the output dir preparation work is completed. So the problem appears to be in the download script which fails to find the corresponding image download link.

Can you please edit line 8 of the  `scripts/download-nexus-image.sh` file and enable the debug output for that script so we can try to trace the error.

minidude2012, 2017-10-13:
ill do that when next at pc sorry for late reply been busy

minidude2012, 2017-10-23:
no point now as my device stopped working thanks for your help anyway dude

closed 2017-10-23 by minidude2012