anestisb/android-prepare-vendor #108: walleye (Pixel 2) support

thestinger, 2017-10-23:
Google posted the factory images, but I don't have one yet and I plan on waiting a bit longer: https://developers.google.com/android/images#walleye.

anestisb, 2017-12-24:
Initial support available under the relative [branch](https://github.com/anestisb/android-prepare-vendor/tree/walleye).

Still doing configuration/static reviews. Plan to do device testing the next few days.

anestisb, 2017-12-25:
Did some quick device testing and no major issues surfaced. Basic support is considered completed. Haven't played much with the full config since the Open GApps support for Oreo is very unstable.

closed 2017-12-25 by anestisb

thestinger, 2018-01-06:
Yeah, I tested this and it works well. The one thing I can't figure out is verified boot... I get an error locking the bootloader and I'm not sure what's going wrong. It seems like it won't load the vbmeta image for AVB but it should all be set up properly. It doesn't work for either a test key signed build or after signing with release keys...

I'm trying to see if any Google engineers know how to get that working with AOSP. It's quite possible that it's simply broken in the firmware which would be really bad...

anestisb, 2018-01-06:
@thestinger did you manage to get a debug UART cable working for Pixel 2 devices?

thestinger, 2018-01-06:
No, but I think @ScottyBauer did. I'm not sure if that would be helpful for this. It prints the basic error to the bootloader screen but it's not particularly helpful. I guess the code might be open source somewhere but I just don't think I'm going to be able to figure out why it fails to load and authenticate.

thestinger, 2018-01-06:
It'd be helpful if you tried with AOSP to see if it's a local issue for me though. One thing I tried already is this patch for device/google/wahoo:

```
diff --git a/BoardConfig.mk b/BoardConfig.mk
index 4085bb5..f092e1f 100644
--- a/BoardConfig.mk
+++ b/BoardConfig.mk
@@ -59,6 +59,8 @@ TARGET_RECOVERY_WIPE := device/google/wahoo/recovery.wipe
 TARGET_RECOVERY_FSTAB := device/google/wahoo/fstab.hardware
 
 BOARD_AVB_ENABLE := true
+BOARD_AVB_ALGORITHM := SHA256_RSA2048
+BOARD_AVB_KEY_PATH := external/avb/test/data/testkey_rsa2048.pem
 
 BOARD_SYSTEMIMAGE_PARTITION_SIZE := 2684354560
 BOARD_SYSTEMIMAGE_JOURNAL_SIZE := 0
```

Since I checked their official builds and they use SHA256_RSA2048 not SHA256_RSA4096 which is the default for AOSP. That change makes it match the algorithms in their builds (`avbtool verify_image vbmeta.img` confirms that and also that it's properly signed with a bundled public key included).

anestisb, 2018-01-06:
@thestinger I'll try that on Monday since I've left the Pixel 2 devices back in the office.

thestinger, 2018-01-06:
Thanks, that'd be great.

ScottyBauer, 2018-01-06:
I can try flashing something today or tomorrow and looking at what's on the suzy-q output, if you want, strcat. Shoot me an irc message

thestinger, 2018-01-07:
I ended up figuring this out:

https://android-review.googlesource.com/c/platform/external/avb/+/582100

jamuir, 2018-03-26:
@ScottyBauer :  I'd like to get a uart cable working with the Pixel 2.  Do you have any hints or pointers you could share?

ScottyBauer, 2018-03-27:
Yeah contact me offline: sbauer@plzdonthack.me
I'll send you the details.